gphoto/libgphoto2 #2: External flash configuration on EOS 1200D

benoitguigal, 2015-08-24:
Hello ! 

I am using libgphoto2 on a Raspberry Pi to control my Canon EOS 1200D.  
I would like to configure an external flash (420 EX II) from libgphoto2 but there is no such configuration option, although it is possible to configure it from the camera.
I have copy pasted the list of configuration available below. 
Do you think it will be possible to configure the external flash ? 

Best 

```
/main/actions/syncdatetime                                                     
/main/actions/syncdatetime
/main/actions/uilock
/main/actions/autofocusdrive
/main/actions/manualfocusdrive
/main/actions/cancelautofocus
/main/actions/eoszoom
/main/actions/eoszoomposition
/main/actions/viewfinder
/main/actions/eosremoterelease
/main/settings/datetime
/main/settings/datetime
/main/settings/reviewtime
/main/settings/output
/main/settings/movierecordtarget
/main/settings/evfmode
/main/settings/ownername
/main/settings/artist
/main/settings/copyright
/main/settings/customfuncex
/main/settings/focusinfo
/main/settings/autopoweroff
/main/settings/depthoffield
/main/settings/capturetarget
/main/settings/capture
/main/status/serialnumber
/main/status/manufacturer
/main/status/cameramodel
/main/status/deviceversion
/main/status/vendorextension
/main/status/model
/main/status/ptpversion
/main/status/batterylevel
/main/status/lensname
/main/status/eosserialnumber
/main/status/shuttercounter
/main/status/availableshots
/main/imgsettings/imageformat
/main/imgsettings/imageformatsd
/main/imgsettings/iso
/main/imgsettings/whitebalance
/main/imgsettings/whitebalanceadjusta
/main/imgsettings/whitebalanceadjustb
/main/imgsettings/whitebalancexa
/main/imgsettings/whitebalancexb
/main/imgsettings/colorspace
/main/capturesettings/focusmode
/main/capturesettings/autoexposuremode
/main/capturesettings/drivemode
/main/capturesettings/picturestyle
/main/capturesettings/aperture
/main/capturesettings/shutterspeed
/main/capturesettings/meteringmode
/main/capturesettings/bracketmode
/main/capturesettings/aeb
/main/other/d402
/main/other/d407
/main/other/d406
/main/other/d303
/main/other/5001
```

makeplayhappy, 2020-08-11:
I'm attempting to trigger multiple captures where the first fires an external flash (not the popup one) via the hot shoe, and the rest fire without the flash. I've tested with Canon EOS 40D and 70D and can't find any configuration option that would allow this. Its available in the cameras menus (Menu > Flash control > Flash Firing) I thought their maybe the option already but I think the one in the source code is for lower class (2 / 3 ?) Canon cameras.

I've attached some output from the 40D and 70D, there are the command line responses from the gphoto2 program and some debug output from --list-config with the flash setting in the 40D enabled and disabled. 

If there is any help I can be with testing anything to allow this to happen let me know, I have attempted to check through the source but my C / USB debugging skills are not quite up to it! 

[raspberry-pi-canon-40D-gphoto.txt](https://github.com/gphoto/libgphoto2/files/5056903/raspberry-pi-canon-40D-gphoto.txt)
[raspberry-pi-canon-70D-gphoto.txt](https://github.com/gphoto/libgphoto2/files/5056904/raspberry-pi-canon-70D-gphoto.txt)

[gpdebug-40D-flash-on-02.log.out.txt](https://github.com/gphoto/libgphoto2/files/5056895/gpdebug-40D-flash-on-02.log.out.txt)
[gpdebug-40D-flash-on-01.log.out.txt](https://github.com/gphoto/libgphoto2/files/5056896/gpdebug-40D-flash-on-01.log.out.txt)
[gpdebug-40D-flash-off-02.log.out.txt](https://github.com/gphoto/libgphoto2/files/5056898/gpdebug-40D-flash-off-02.log.out.txt)
[gpdebug-40D-flash-off-01.log.out.txt](https://github.com/gphoto/libgphoto2/files/5056899/gpdebug-40D-flash-off-01.log.out.txt)

msmeissn, 2021-10-03:
(did not have time to look at this ... so far external flash control seems non obvious)

JacksonLegge, 2023-04-18:
Canon's EOS Utility on Windows/Mac allows this to be controlled, so at least it's not a limitation of the camera. Unfortunately I don't have the skills to implement that into anything else, so I'm probably no help with solving this issue.

msmeissn, 2023-04-19:
the EOS cameras report a lot of config variables, not all of them have been correctly decoded or exposed via the config api yet, as this is largely reverse engineered here currently.

I had a short look for options named with Strobo but nothing really easy to add sticks out in a short look for now. needs a bit research / usb sniffing