gphoto/libgphoto2 #4: Host->Camera ptpip transfers hang on Nikon due to ptp_ptpip_sendreq() wrong phase

horshack-dpreview, 2015-09-04:
Hi,

I wanted to bring to your attention a bug that causes Nikon cameras to stop responding after completing a PTPIP MTP request that involves a Host->Camera transfer, for example a "--set-config datetime=now". I'm new to github and not familiar enough with the libgphoto2 codebase to make the necessary change but here's what needs to be altered:

ptp_ptpip_sendreq() needs to use a "ptpip_cmd_dataphase" parameter of 2 for Host->Camera transfers. It presently always sets the dataphase to 1, irrespective of the command's data direction. Existing code:

ptp_ptpip_sendreq (PTPParams* params, PTPContainer* req)
{
       int     ret;
       int     len = 18+req->Nparam*4;
    unsigned char   *request = malloc(len);

    ptp_ptpip_check_event (params);

    htod32a(&request[ptpip_type],PTPIP_CMD_REQUEST);
    htod32a(&request[ptpip_len],len);
    htod32a(&request[ptpip_cmd_dataphase],1);   /* FIXME: dataphase handling */

The correct dataphase values for each data direction should be:
Camera->Host or No Data: 1
Host->Camera: 2

With the incorrect data phase, here's what the trace looks like for a SetDevicePropValue:DateTime request from gphoto:

0.307724 camera_set_config           (2): Setting property 'Camera Date and Time' / 0x5011
0.307834 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 22 bytes)
0000  16 00 00 00 06 00 00 00-01 00 00 00 16 10 35 00  ..............5.
0010  00 00 11 50 00 00      -                         ...P..          

0.307893 ptp_ptpip_senddata          (3): ptpip/senddata header: (hexdump of 20 bytes)
0000  14 00 00 00 09 00 00 00-35 00 00 00 21 00 00 00  ........5...!...
0010  00 00 00 00            -                         ....            

0.307987 ptp_ptpip_senddata          (3): ptpip/senddata data: (hexdump of 45 bytes)
0000  2d 00 00 00 0c 00 00 00-35 00 00 00 10 32 00 30  -.......5....2.0
0010  00 31 00 35 00 30 00 39-00 30 00 34 00 54 00 30  .1.5.0.9.0.4.T.0
0020  00 32 00 31 00 32 00 31-00 36 00 00 00           .2.1.2.1.6...   

0.312794 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 8 bytes)
0000  0e 00 00 00 07 00 00 00-                         ........        

0.312817 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (hexdump of 6 bytes)
0000  01 20 35 00 00 00      -                         . 5...          

0.314384 gp_camera_free              (2): Freeing camera...
0.314398 gp_camera_exit              (2): Exiting camera ('PTP/IP Camera')...
0.314482 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 18 bytes)
0000  12 00 00 00 06 00 00 00-01 00 00 00 c7 90 36 00  ..............6.
0010  00 00                  -                         ..              

30.350267 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (empty hexdump of empty buffer)
30.350309 ptp_ptpip_generic_read [ptpip.c:155](0): End of stream after reading 0 bytes of ptpipheader
30.350334 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (empty hexdump of empty buffer)
30.350345 ptp_ptpip_generic_read [ptpip.c:155](0): End of stream after reading 0 bytes of ptpipheader
30.350357 ptp_ptpip_sendreq           (3): ptpip/oprequest data: (hexdump of 18 bytes)
0000  12 00 00 00 06 00 00 00-01 00 00 00 03 10 37 00  ..............7.
0010  00 00                  -                         ..              

30.350409 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (empty hexdump of empty buffer)
30.350418 ptp_ptpip_generic_read [ptpip.c:155](0): End of stream after reading 0 bytes of ptpipheader
30.350425 ptp_ptpip_generic_read      (3): ptpip/generic_read data: (empty hexdump of empty buffer)
30.350431 ptp_ptpip_generic_read [ptpip.c:155](0): End of stream after reading 0 bytes of ptpipheader
30.350546 gp_port_close               (2): Closing port...

msmeissn, 2015-09-06:
Well spotted! 
I added a parameter to the sendreq functions that allow decisions on the data direction and commited it to master.
Tested out well with my D750.

msmeissn, 2015-12-26:
its done I think, and in 2.5.9

closed 2015-12-26 by msmeissn

horshack-dpreview, 2016-01-06:
Awesome, thanks Marcus!

-adam

[https://avatars0.githubusercontent.com/u/14128283?v=3&s=400]<https://github.com/gphoto/libgphoto2/issues/4#issuecomment-167355691>

Host->Camera ptpip transfers hang on Nikon due to ptp ...<https://github.com/gphoto/libgphoto2/issues/4#issuecomment-167355691>
github.com