gphoto/libgphoto2 #7: Unable to compile, configure file has gone

gonzalo, 2015-09-29:
Hi, I'm the owner of https://github.com/gonzalo/gphoto2-updater, a script to automatically compile the last version of gphoto/libgphoto over ubuntu/debian/raspian. 
I'm trying to update my code to github respository but I've observed that some files have changed. Configure file has gone and "autoreconf --install --symlink"  gets an "error: couldn't open directory 'm4m'".

So, do you have any suggestions to compile it  over Ubuntu?

msmeissn, 2015-09-30:
Did you really pull the current git?
or how do you get the sources?

m4m is not used there anymore, so it should not show up.

gonzalo, 2015-10-05:
Hi Marcus,

I was downloading the tgz's in order to keep my sources unaltered as much as I can. 

    https://github.com/gphoto/gphoto2/archive/gphoto2-2_5_8-release.tar.gz 
    https://github.com/gphoto/libgphoto2/archive/libgphoto2-2_5_8-release.tar.gz

Now I'm trying to use git pull and seems that compiles correctly. I will develop a new version of my script  using git and autoreconf. I'll keep you informed.

One question. Using GIT it's supossed that I'll allwayt get the last libghoto2 and gphoto2 version? Until now I have to update sources every time you realease a new version.

msmeissn, 2015-10-05:
Well, the last release tarballs are still in sourceforge. 2.5.9 will be done at github, but you will probably need autoreconf then.

If you check out master, you always get the current GIT status.

You can also use the release tags, depending on how up to date you want to be.

gonzalo, 2015-10-05:
Here you have the current version of script using github respositories.: https://github.com/gonzalo/gphoto2-updater/tree/GitHub-integration 
As soon as I can test it over my Raspberry PI I'll merge it to the master branch

gonzalo, 2015-10-06:
One last question Marcus, master branch will be an "stable" branch or a development branch?

msmeissn, 2015-10-06:
I currently only have 1 branch, master. its supposed to be stable. :)

gonzalo, 2015-10-11:
Finally I can confirm that my branch https://github.com/gonzalo/gphoto2-updater/tree/GitHub-integration works correctly in Debian, Ubuntu & Raspbian with last github release. I'll keep it as branch until 2.5.9 release.

Sija, 2015-10-18:
@gonzalo: I'd suggest using git tags for every stable release instead of pulling changes str8 from the `master` branch.

gonzalo, 2015-10-19:
That is what I've been doing during last releases (https://github.com/gonzalo/gphoto2-updater/releases). I'll keep two branchs, one with the official releases and the other one (https://github.com/gonzalo/gphoto2-updater/tree/GitHub-integration) associated to the master branch.

gonzalo, 2015-10-19:
I close this issue since is not a real bug. New releases of script will bie avaliable in the repository https://github.com/gonzalo/gphoto2-updater/

closed 2015-10-19 by gonzalo