gphoto/libgphoto2 #12: Problem after first capture image and download canon 1200D

petha, 2015-10-11:
Hi,

I seem to have the exact same problem as http://sourceforge.net/p/gphoto/support-requests/129/
Iam using the exact same hardware Raspberry PI 2 model B..

Sometimes I get "You need to specify a folder starting with /store_xxxxxxxxx/" and but most of the time it just seem to hang for a while and then reporting 

ERROR: Could not capture image. 
ERROR: Could not capture

You can find the debug file here: 
https://gist.github.com/petha/674dfe311c25ec20501b

msmeissn, 2015-10-11:
do you hear if the camera has triggered its shutter?

msmeissn, 2015-10-11:
i cannot reproduce on the eos 100d though :( perhaps adding a wait or similar might be possible

petha, 2015-10-11:
It does not seem to trigger the shutter, sometime it does but then the error seems to be a different one. 

Do you have a rough idea on where a wait could be added?

msmeissn, 2015-10-11:
can you try with this commit? i changed the wait logic a bit.

petha, 2015-10-11:
I think it resolves this issue! Thanks!

closed 2015-10-11 by petha

shtefanilie, 2019-07-15:
hi @msmeissn, I am experiencing the same issue, but with a 1100D. Any ideas why this is happening?