gphoto/libgphoto2 #13: PTP Device Busy on second capture and download for EOS 1200D

petha, 2015-10-11:
Consecutive calls to "gphoto2 --capture-image-and-download" fails after the first one succeeds.

Unfortunately this does not happen with --debug flag on
The shutter is not triggered

*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

msmeissn, 2015-10-12:
is this after the patch from yesterdays issue 12?

petha, 2015-10-12:
Yes this is after yesterdays issue 12.

msmeissn, 2015-10-13:
can you test the new patch?

if possible regular and with the lenscap on to see a non - focusing event

petha, 2015-10-13:
I've tried the new patch and from what i can tell the device busy error is no longer showing up.
But the behavior where everything works fine with debug flag seems to remain. 
It does not seem to make any difference if the lenscap is on but I have created a couple of log files both with cap on and off. 

After a clean reboot this is what I get:
pi@raspberrypi ~ $ env LC_ALL=C gphoto2 --capture-image-and-download --filename "photos/%Y%m%d%H%M%S.cr2"
New file is in location /capt0000.cr2 on the camera                            
Saving file as photos/20151013165943.cr2
Deleting file /capt0000.cr2 on the camera
pi@raspberrypi ~ $ env LC_ALL=C gphoto2 --capture-image-and-download --filename "photos/%Y%m%d%H%M%S.cr2"
ERROR: Could not capture image.                                                
ERROR: Could not capture.

You can download the log files from here: https://s3-eu-west-1.amazonaws.com/kolmarden-photos/logs.tar.gz
Most of the files did succeed but a few returned ptp io error, it will crash quite consistently without debug

Please let me know if I can help in any way, I really appreciate your help!

msmeissn, 2015-10-14:
I had to a bit of incremental work on this, i failed to initialize focus_start.

canb you try current git, or use commit 378abb40e25e9e8f96c172f8c4c0482db365db66 incrementally.

petha, 2015-10-14:
It works like a charm! Super!

closed 2015-10-14 by petha