gphoto/libgphoto2 #15: segmentation fault/corrupted double-linked list on powering down camera

alik55, 2015-10-16:
When camera is manually powered down shortly (1-2sec) after it gets connected and while still processing queued events (cam.processEvents) the program intermittently crashes with segfault or "corrupted double-linked list".

Main:
while (1)
{
	if ( !cam.isConnected() )
	{
		if ( !cam.connect() )
			sleep(2);
	}
	if ( cam.isConnected() )
		cam.processEvents(500);
}

cam.processEvents:
{
	if ( !bConnected )
		return;
	CameraEventType	evttype;
	CameraFilePath	*path;
	void	*evtdata = 0;
	if (GP_OK != gp_camera_wait_for_event (_camera,timeout_ms, &evttype, &evtdata, _context))
	{
		disconnect();
		return;
	}
	switch (evttype)
	{
	case GP_EVENT_FILE_ADDED: // did not ask for file, discard it
		path = (CameraFilePath*)evtdata;
		gp_camera_file_delete(_camera, path->folder, path->name, _context);
		break;
	default:
		break;
	}
	if ( evtdata )
		free (evtdata);
}

cam.disconnect:
{
	if ( _context && _camera ) 
             gp_camera_exit(_camera, _context) ;
	if ( _camera )
		gp_camera_unref(_camera);
	if ( _context )
		gp_context_unref(_context);
	_context = NULL;
	_camera = NULL;
}

Debug log:
waiting for events timeout 500 ms
Sending PTP_OC 0x911d / PTP_OC_CANON_EOS_KeepDeviceOn request...
Reading PTP_OC 0x911d response...
Sending PTP_OC 0x9116 / PTP_OC_CANON_EOS_GetEvent request...
Reading PTP_OC 0x9116 / PTP_OC_CANON_EOS_GetEvent data...
Reading PTP_OC 0x9116 response...
entry type 0003
waiting for events timeout 500 ms
Sending PTP_OC 0x911d / PTP_OC_CANON_EOS_KeepDeviceOn request...
Reading PTP_OC 0x911d response...
'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output Error (-1)
Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
Clearing halt on IN EP and retrying once.
Clear USB halt...
'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Input/Output Error (-1)
Reading 1024 = 0x400 bytes from port failed: Error reading from the port (-34)
Clearing halt on IN EP and retrying once.
[... 231 lines cut ...]
checking: transfer 3:0x172a620 status 0
checking: transfer 4:0x172a7a8 status 0
checking: transfer 5:0x1726e90 status 0
checking: transfer 6:0x1727048 status 0
checking: transfer 7:0x1727200 status 0
checking: transfer 8:0x17273b8 status 0
checking: transfer 9:0x1727570 status 0
0x171fd48 with status 3
0x172a310 with status 3
0x172a498 with status 3
0x172a620 with status 3
0x172a7a8 with status 3
0x1726e90 with status 3
0x1727048 with status 3
0x1727200 with status 3
0x17273b8 with status 3
0x1727570 with status 3
resetting filesystem
Clearing fscache LRU list...
fscache LRU list already empty
Internally deleting all folders from '/'...
Lookup folder '/'...
Found! / is 0x1721a38
Recurse delete folder 0x1721a38//
disconnected from camera

alik55, 2015-10-17:
It appears there are 2 threads - one from my program and one from libusb that gets created when a camera is conencted. If the camera is disconnected while there are USB transfers in progress from gp_camera_wait_for_event , the libsub thread tries to cancel transfers and my programs calls gp_camera_exit which also tries to cancel transfers from another thread. Comments before libusb/io.c say:
/* do the actual event handling. assumes that no other thread is concurrently
 * doing the same thing. */

msmeissn, 2015-10-18:
The other libusb thread should actually only poll udev events I would hope.

alik55, 2015-10-18:
I am not in a position to debug libusb or libgphoto so I solved my problem just by sleeping for second before calling gp_camera_exit after gp_camera_wait_for_event fails just to give libusb enough time to cleanup.

msmeissn, 2015-10-18:
i can try itto debug it, but I fear the problem is in the libusb area, as I am using the standard api calls in the documented way I think. :/

alik55, 2015-10-18:
I am not familiar at all with what APi libusb exposes and how it is used by libgphoto_port. If libusb exposes any mutexes, those could be used. After briefly looking at libusb code it looks that it was updated quite a bit in recent years and it seemed libgphoto was using some older interface from it (eg libusb_handle_events_timeout_completed is a new one). But I may be totally wrong here.

msmeissn, 2015-11-15:
could you possibly try 1.0.20 libusb? they had some async related fixes there.

alik55, 2015-11-15:
Thanks for letting me know. I will install it, but the bug was intermittent, so testing will require a large sample size, but in my release code I will want still keep my workaround of sleeping for a second just in case the problems is still around.

alik55, 2015-11-15:
Well, that did not take long. It still segfaults.

msmeissn, 2015-11-15:
not good :( i did some related fixes for the --reset option, but this is unrelated to this issue.

i also fail to reproduce so far here

msmeissn, 2015-12-21:
Ok, I think I found the reason and it is as of just now fixed in GIT.

(missing clearing malloced pointers on _close which would be reused on port reopen)

Please retest :)

closed 2015-12-21 by msmeissn