gphoto/libgphoto2 #18: sony slt dual mode wait-event

msmeissn, 2015-11-18:
--wait-event-and-download in a dual raw+jpg setting does not work, it leaves one image on camera,
which after some tries fills the memory.

use some logic like in sony_capture

falk0069, 2015-11-18:
Is this due to the new Sony functions that were just added?  Do you want me to test anything?

msmeissn, 2015-11-18:
no, its missing things that i have not yet implemented, unrelated to your code

gschwim, 2015-12-02:
I've seen images get stuck in the camera. With the A7S I have I'm usually able to flush the camera with the following:

    gphoto2 --set-config bulb=0 --wait-event-and-download=5s

This seems to work no matter what mode the camera is in. I usually run it 2x to be sure.

msmeissn, 2016-10-13:
this should be good now with current git and upcoming 2.5.11 release

closed 2016-10-13 by msmeissn