gphoto/libgphoto2 #19: Can't build libgphoto2 2.5.9.1 on OSX, libltdl error

gschwim, 2015-11-19:
On OSX w/ homebrew and xcode.

I acquired the source as follows:

  git clone https://github.com/gphoto/libgphoto2

I run autoreconf then ./configure and get the following error:

  checking that we can compile and link with libltdl... no
  configure: error: cannot compile and link against libltdl
  libgphoto2 requires libltdl (the libtool dl* library), 
  but cannot compile and link against it.
  Aborting.

I'm able to repeat this on two different macs. I'm able to compile fine on Ubuntu.

msmeissn, 2015-11-19:
On Wed, Nov 18, 2015 at 11:16:39PM -0800, gschwim wrote:
> On OSX w/ homebrew and xcode.
>
> I acquired the source as follows:
>
>   git clone https://github.com/gphoto/libgphoto2
>
> I run autoreconf then ./configure and get the following error:
>
>   checking that we [...]

Alternatively you can download the sourcetarball from http://sf.net/projects/gphoto/

Well, is there a libtool development or libltdl development package on MacOS?

I am not sure about the naming.

(The Travis CI builds find it nicely though, so there must be a package somewhere.)

Ciao, Marcus

gschwim, 2015-11-19:
The source tarball is 2.5.8. I'm after the sony bulb capabilities that appear to be in 2.5.9.1.

There is a libtool install for homebrew, but not sure that it is providing 100% of what is needed. Oddly, the default homebrew recipe for libgphoto2 builds just fine.

msmeissn, 2015-11-19:
i releaed 2.5.9 yesterday, so there are new tarballs since some hours. :) 

can you check that?

the libtool install for homebrew should be fine.

gschwim, 2015-11-19:
Ah, there the are, and they work. Awesome and thank you! 

Figured this should also work out of github. I wonder what's different.

msmeissn, 2015-11-19:
the git checkout is without all the funny automake,libtool etc bells and wrappers that "make dist" already adds. this makes building from git a bit harder than from a tarball.

gschwim, 2015-11-19:
Got it. Above my head on that, I'm just a lowly network engineer. :)

Thanks for the help. Closing...

closed 2015-11-19 by gschwim