gphoto/libgphoto2 #20: A7S ISO change down goes up when cam is at 25600 or higher, then gets stuck

gschwim, 2015-12-02:
When the camera is at ISO 25600 or higher and you try to set something lower, it will climb to 64000 then flap between 51200 and 64000. Also, 51200 specifically cannot be set, so this may be a contributor.

'gphoto2 --set-config iso=3200'

I think there is more going on here but haven't had a chance to test all of it.

msmeissn, 2015-12-03:
yes, the setting of these values is by doing +1 / -1 calls and we might go in the wrong direction for some reason.

simonepignotti, 2018-07-11:
This issue also happens on the A7R2, both on Linux and OSX.

How difficult would it be to set values directly instead of doing +1/-1 ? Or at least one could easily detect to be looping between two values and exit with an error.

I could take care of this issue if it is not too complicated and you give me some hints of where to start.

Thanks, this library is awesome!

elijahparker, 2018-07-11:
With Sony it's not possible to directly set values, which makes it
difficult and slow. I have a fix for this that I can post when I'm back in
the office, though. The problem seems to be that sometimes it fails to read
the current value and so goes the wrong direction initially.

simonepignotti, 2018-07-12:
Thank you @elijahparker 

What do you mean by it's not possible? Which limitation does Sony have wrt other manufacturers?

BTW the problem also appears when you set a configuration value which does not exist in the camera, e.g. if you try to set ISO to 600 the camera will loop over 400 and 800.

elijahparker, 2018-07-12:
The limitation is in Sony's PTP implementation in that it does not support
the standard "set property" command like pretty much all other cameras.
Instead, there's vendor-defined "up/down" commands. You'll notice even on
Sony's official tethering app the settings are changed by up/down buttons
(there's no drop-down list where you can jump to any).

Now their wifi API on the other hand is much more robust and well designed,
and has actual documentation available.

simonepignotti, 2018-07-12:
Unbelievable...

elijahparker, 2018-07-12:
Well, it doesn't end there, either, but I won't go on...  Here's the code
that I had modified that seems to work better for setting Sony parameters
(from config.c):
https://github.com/timelapseplus/VIEW/blob/master/lib/libgphoto2_ptp2_updates/config.c#L1393-L1504

You can replace that code in your source and see if it works better for
you.  Let me know if it does, since I'll be trying to merge this in for a
pull request soon.

simonepignotti, 2018-07-12:
Thank you, will try!

msmeissn, 2018-07-13:
THere is not much difference excepting the usleep(200*1000) -> usleep(100*1000); which seems to speed iot up.

msmeissn, 2018-07-15:
fwiw, since 2015 this code got changed and adjusted, so it is probably already fixed

musubk, 2018-12-07:
This is still an issue with my Sony a7s. I'm operating the camera remotely using a script which should vary the ISO from 100 to 3200 depending on the time of night. Occasionally I will notice the image is brighter than expected, and when I check it's at ISO 51200. No idea how it got there, but once there, I can issue gphoto commands to set it anywhere above ISO 25600, but attempting to set it to anything below 25600 will return 'value not in enumerated list'. From there, the only way to recover seems to be manually setting the ISO on the camera by hand to something below 25600. This is an inconvenience for my current setup located ~30 miles away, but it's going to be much more annoying with the second camera setup I'm looking to deploy a couple hundred miles above the arctic circle...

msmeissn, 2018-12-10:
can you attach debug logging of setting it?
also the ouztput of gphoto2 --summary   ?

musubk, 2018-12-11:
Here is the debug output when I get the error:

[************ ~]$ gphoto2 --set-config iso=12800

*** Error ***
Now value is not in enumeration.
[************ ~]$ gphoto2 --set-config iso=12800 --debug
0.002490 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.002546 main                        (2): gphoto2 2.5.17
0.002565 main                        (2): gphoto2 has been compiled with the following options:
0.002578 main                        (2):  + gcc (C compiler used)
0.002585 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.002600 main                        (2):  + exif (for displaying EXIF information)
0.002607 main                        (2):  + no cdk (for accessing configuration options)
0.002614 main                        (2):  + no aa (for displaying live previews)
0.002620 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.002627 main                        (2):  + readline (for easy navigation in the shell)
0.002634 main                        (2): libgphoto2 2.5.19
0.002644 main                        (2): libgphoto2 has been compiled with the following options:
0.002650 main                        (2):  + all camlibs
0.002657 main                        (2):  + gcc (C compiler used)
0.002665 main                        (2):  + ltdl (for portable loading of camlibs)
0.002672 main                        (2):  + EXIF (for special handling of EXIF files)
0.002687 main                        (2): libgphoto2_port 0.12.0
0.002699 main                        (2): libgphoto2_port has been compiled with the following options:
0.002713 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.002724 main                        (2):  + gcc (C compiler used)
0.002734 main                        (2):  + ltdl (for portable loading of iolibs)
0.002745 main                        (2):  + USB (libusb1, for USB cameras)
0.002755 main                        (2):  + serial (for serial cameras)
0.002765 main                        (2):  + no resmgr (serial port access and locking)
0.002776 main                        (2):  + no ttylock (serial port locking)
0.002786 main                        (2):  + no lockdev (serial port locking)
0.002798 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.002808 main                        (2): IOLIBS env var not set, using compile-time default instead
0.002819 main                        (2): invoked with following arguments:
0.002829 main                        (2):   --set-config
0.002839 main                        (2):   iso=12800
0.002849 main                        (2):   --debug
0.002917 load_settings               (2): Creating gphoto config directory ('/home/jason/.gphoto')
0.004831 load_settings               (2): Loading settings from file '/home/jason/.gphoto/settings'.
0.005086 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.005110 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/libgphoto2_port/0.12.0'...
0.007776 foreach_func                (2): Called for filename '/usr/lib/libgphoto2_port/0.12.0/disk'.
0.009533 gp_port_library_list        (2): found fstab fsname UUID=b68de95b-b287-4c49-a6bd-f5b8c7f94883
0.009600 gp_port_library_list        (2): found fstab fsname UUID=5C84-AAE0
0.009632 gp_port_library_list        (2): found fstab fsname UUID=e421970e-d71c-4b28-a5b9-6714876f4610
0.009865 gp_port_library_list        (2): found fstab fsname UUID=9db8cc40-2a38-4cbe-a8f7-0c5f4f9ae1ae
0.010075 gp_port_library_list        (2): found mtab fsname proc
0.010093 gp_port_library_list        (2): found mtab fsname sys
0.010108 gp_port_library_list        (2): found mtab fsname dev
0.010130 gp_port_library_list        (2): found mtab fsname run
0.010156 gp_port_library_list        (2): found mtab fsname efivarfs
0.010189 gp_port_library_list        (2): found mtab fsname /dev/sda3
0.010221 gp_port_library_list        (2): found mtab fsname securityfs
0.010237 gp_port_library_list        (2): found mtab fsname tmpfs
0.010258 gp_port_library_list        (2): found mtab fsname devpts
0.010268 gp_port_library_list        (2): found mtab fsname tmpfs
0.010300 gp_port_library_list        (2): found mtab fsname cgroup
0.010316 gp_port_library_list        (2): found mtab fsname pstore
0.010338 gp_port_library_list        (2): found mtab fsname cgroup
0.010353 gp_port_library_list        (2): found mtab fsname cgroup
0.010418 gp_port_library_list        (2): found mtab fsname cgroup
0.010433 gp_port_library_list        (2): found mtab fsname cgroup
0.010448 gp_port_library_list        (2): found mtab fsname cgroup
0.010457 gp_port_library_list        (2): found mtab fsname cgroup
0.010495 gp_port_library_list        (2): found mtab fsname cgroup
0.010510 gp_port_library_list        (2): found mtab fsname cgroup
0.010532 gp_port_library_list        (2): found mtab fsname cgroup
0.010544 gp_port_library_list        (2): found mtab fsname systemd-1
[... 1046 lines cut ...]
1.109363 _cb_irq                     (2): Transfer 0x5566b86e86d0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
1.109378 _cb_irq                     (2): 0x5566b86e8900 with status 3
[... cut ...]
1.115659 gp_filesystem_reset         (2): resetting filesystem
1.115693 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
1.115698 gp_filesystem_lru_clear     (2): fscache LRU list already empty
1.115703 delete_all_folders          (2): Internally deleting all folders from '/'...
1.115716 lookup_folder               (2): Lookup folder '/'...
1.115734 lookup_folder               (2): Found! / is 0x5566b867d3f0
1.115744 recurse_delete_folder       (2): Recurse delete folder 0x5566b867d3f0//

musubk, 2018-12-11:
And here's the gphoto2 --summary output:

[************ ~]$ gphoto2 --summary
Camera summary:
Manufacturer: Sony Corporation
Model: ILCE-7S
  Version: 3.0
  Serial Number: 00000000000000003282741003376648
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats:
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
        File Download, No File Deletion, No File Upload
        No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19] value: 3
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: Daylight (4)
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/0 (0)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,3,32772,32773,32776,32774,32775] value: Manual Focus (1)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [4,1,32770] value: Average (1)
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32772,32817,32818] value: Undefined (0)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,32769,2,3,4,1,32848,32849,32850,32851,32852,32833,7,32785,32789,32788,32786,32787,32790,32791,32792] value: M (1)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,1,2,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000,-3300,-3500,-3700,-4000,-4300,-4500,-4700,-5000] value: 0.0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32787,32788,32773,32772,32776,32777,33591,34103,33623,34135,33655,34167,33553,34065,33569,34081,33585,34097,33590,34102,33622,34134,33654,34166,33552,34064,33568,34080,33584,34096,32792,32808,32793,32809] value: Single Shot (1)
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,1,2,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21] value: 31
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 1
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 65551
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,8,9,10,4,5,6,7] value: 1
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
CC Filter(0xd210):(readwrite) (type=0x2) Range [121 - 135, step 1] value: 128
AB Filter(0xd21c):(readwrite) (type=0x2) Range [121 - 135, step 1] value: 128
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,64000,80000,102400,128000,160000,204800,256000,320000,409600,33554431,16777241,16777266,16777280,16777296,16777316,16777341,16777376,16777416,16777466,16777536,16777616,16777716,16777856,16778016,16778216,16778466,16778816,16779216,16779716,16780416,16781216,16782216,16783616,16785216,16787216,16790016,16793216,16802816,16828416] value: 200
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768] value: 32768
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
Property 0xd21f:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd217:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: -1
Property 0xd219:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd212:(readwrite) (type=0x2) Range [0 - 15, step 1] value: 0
Property 0xd221:(readwrite) (type=0x2) Enumeration [0,1,2] value: 1
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 18324480
Objects in memory(0xd215):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 0
Property 0xd220: error 2002 on query.
Property 0xd2c1:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c2:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c3:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c9:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Movie(0xd2c8):(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Still Image(0xd2c7):(readwrite) (type=0x4) Enumeration [1,2] value: 1

msmeissn, 2018-12-13:
hmm, it arrives at value 0x9c40 which is 40000 , but this does not even exists as valid ISO value in what we got from the camera

msmeissn, 2019-02-16:
i think we got more improvements, if you can try it