gphoto/libgphoto2 #24: Lags on v2.5.9

Sija, 2015-12-10:
Hi!

Version `2.5.9` introduced lag between call to `take_picture` and actual effect.
This didn't happen in previous versions.

Any idea what can be done with it?
We wouldn't like to lock ourselves on `2.5.8` as the last working version...

msmeissn, 2015-12-10:
can you get a debug logfile of this lag? --debug --debug-logfile=xx.log

msmeissn, 2016-02-18:
no reply .,.. if you could send debug, please reopen

ratwix, 2016-03-13:
Hello, I have the same issue with a 550D as well. Same lag with the last 5.9.1. I have to revert to 2.5.8.
I use libgphoto2 on C++. How can I build a log file ?

Thank.

msmeissn, 2016-03-13:
i changed the eos capturing to wait for the focusing event. are you using manual focus?

ratwix, 2016-03-13:
Hello, no I use automatic focus. The workflow looks : focus, when focus done wait 2-3 sec, shoot.

msmeissn, 2016-03-13:
can you reproduce this somehow with the commandline tool ?  have to try myself too

ratwix, 2016-03-13:
Unfortunatly my will get back my eos wednesday. I will test it asap.

ratwix, 2016-03-13:
Just for information, I have the preview mode active (mirror up), maybe it change something, I didn't try  without it.

msmeissn, 2019-03-09:
We made various improvements to EOS support in the last 2 years, please try a newer libgphoto2 version :)

closed 2019-03-09 by msmeissn