gphoto/libgphoto2 #25: Nikon CoolPix S2700 borken w/ 2.5.9

xondrejk, 2015-12-12:
I've been wondering why my wife's CoolPix S2700 stopped working properly (one could run gphoto for instance with `--list-folders`, but not with `--summary`) with libgphoto2-2.5.9, so I snooped around a bit. It'd seem to come down to: 929df71b04f892a9696cbb00970b5ea08d3b6f07. Namely:

    452                 if (camera->pl->params.device_flags & PTP_NIKON_1) {
    ...

In case of said CoolPix S2700 camera, we end up with `0x240000` & `0x200000` and hence add extra `0xfxxx` devicepropcodes which then in turn make the camera unhappy and the communications appear to hang (I haven't double checked, it could just be timing out on each extra reported property -> taking very long to eventually fail). I have however removed the above mentioned snipped and the resulting SW warked again with that Nikon. I am by no means well versed in matters camera related SW, so I hope this is of at least some use. Let me know if further information or tests should be of assistance.

I suppose other Nikon cameras can be affected as well. But basically grouping in that check seems a bit too broad.

msmeissn, 2015-12-13:
was my bad ... the Nikon S1 / S2 detection also triggered for Coolpix Ssomething

closed 2015-12-13 by msmeissn

xondrejk, 2015-12-13:
Thanks for having a look at that. I've tested your patch with said Camera and can confirm it have expected effect, i.e. works like a charm.