gphoto/libgphoto2 #26: Tethering doesn't work on the Sony A77M2

pedrocr, 2015-12-19:
I've tried using gphoto2 2.5.9 to do tethered capture with a A77M2. I'm running the ubuntu xenial packages on ubuntu trusty so it's possible there's something wonky with my install.

The camera gets detected apparently fine:

```
$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony Alpha-A77 M2 (Control)    usb:001,009     
pedrocr@hyde:~/Projects/RawspeedFormatSupport/full/test_tethering$ gphoto2 --port usb: --abilities
Abilities for camera             : Sony Alpha-A77 M2 (Control)                 
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Image
Configuration support            : yes
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes
```

However when I try --capture-image-and-download the camera seems to try and focus and then do nothing. When I do --capture-tethered nothing seems to happen even after I press the camera shutter. I've ran the following command as suggested by the program:

```
env LANG=C gphoto2 --debug --debug-logfile=a77m2-logfile.txt --port usb: --capture-tethered
```

and here's the output: [a77m2-logfile.txt](https://github.com/gphoto/libgphoto2/files/67604/a77m2-logfile.txt)

pedrocr, 2015-12-19:
I've now been able to take a photo by changing the camera to manual focus. It seems highly flaky though. It will sometime save the jpg, sometimes the raw (issue #18 probably) and will often get into a state where no more photos can be taken.

msmeissn, 2016-10-11:
we are doing some improvements on this in current git.

msmeissn, 2016-10-13:
its hopefully now good for usage. 2.5.11 will have this change (already in current git)

closed 2016-10-13 by msmeissn