gphoto/libgphoto2 #27: Moto X incorrectly detected

moimael, 2015-12-22:
Hi,

Gphoto2 do not seems to work with Motorola Moto X (2014) with android 6.

First it do not seems to be recognized as Moto X but as Moto G

```
Possibilités de l'appareil                      : Motorola Moto G (ID2)        
Gestion du port série                           : non
Gestion du port USB                             : oui
Choix d'acquisition                             :
                                                : Acquisition non gérée par ce pilote
Gestion de la configuration                     : non
Effacer les fichiers sélectionnés de l'appareil : oui
Supprimer tous les fichiers de l'appareil       : non
Gestion des vignettes de prévisualisation       : non
Gestion de l'envoi de fichiers                  : oui
```
and when trying to get all files from devices I get this error : 

```
*** Erreur ***
Identifiant de stockage PTP incorrect
*** Erreur (-1 : « Erreur indéfinie ») ***
```

It means:  PTP storage ID incorrect, Error(-1 : undefined error)

Thanks !

msmeissn, 2015-12-25:
can you get debug output of this? using 
gphoto2 -P --debug --debug-logfile=xx.log

msmeissn, 2020-02-29:
no reply . might be fixed

closed 2020-02-29 by msmeissn