gphoto/libgphoto2 #30: PTP Device Busy on second capture and download for EOS 1200D #2 + slow Jessie

Benjamin7785, 2015-12-29:
Hi.
I know this error has been seen before, but what exactly is the fix for it?. 
https://github.com/gphoto/libgphoto2/issues/13
Do I have to manually edit a file? sry, not an expert..... Where do I find the file?
I tried both the 2.5.8 and 2.5.9 (also 2.5.9.1) gphoto2 version. Because I thought the fix might have been implemented since then. 

Moreover, I have one Raspberry Pi with an Raspbian wheezy image and one Pi with an Raspbian Jessie image. What I see is that on the Whezzy image it takes about 200ms from the time the command "gphoto2 --capture-image-and-download" is executed to the time the camera reacts. That is a fast respond. BUT on the Jessie image the same command takes about 1500ms until the camera reacts. What is going on there? Anyidea?

Thanks in advance,
Ben

msmeissn, 2015-12-29:
did you try the libgphoto2 2.5.9 version?
gphoto2 --version

should show the versions.

If you run the cvommand with a debug trace it will show where the delays happens

gphoto2 --debug --debug-logfile=logfile.txt   --capture-image-and-download

Benjamin7785, 2015-12-29:
ok, did the debug command. The output is a 35MB file :-). Do you want me to put it here? 

Based on what I can see from the file there are 200ms delays when read/write happens to the camera. I will make the same debug command on the wheezy image and try to compare.

About my question regarding the file I couldn't find. Well, I used gphoto-updater.sh to update to 2.5.9. Means, I don't have those files on my computer apparently. 
Since, I'm new to linux based systems I have no idea how to clone, compile and install gphoto2 and libgphoto2 on my own. 
Is there a nice readme somewhere? :-)

May be I messed up my system. How do I remove gphoto2 and libgphoto cleanly?

msmeissn, 2015-12-29:
as you did, you can check it yourself, it is probably happening in the first part before the larger data parts gets tansferred. the initial column in the logfile is seconds.

there the area around big jumps to around 1.5 or so is interesting, you could cut off the rest afterwards.

if you used gphoto2 updater you should have the current one, it will give you a 2.5.9 version for both libgphoto2 and gphoto2.

There is not much of a documentation. but basically you download the tarballs, unpack them, then run 

./configure --prefix=/usr  
make
sudo make install

(might need more options to configure)

Benjamin7785, 2015-12-29:
I need your help... I am going crazy.....

I spend almost 3days installing/deinstalling gphoto2 (2.5.4, 2.5.8, 2.5.9, 2.5.9.1 using gphoto-updater) on vanilla versions of raspian. Both jessie and wheezy. I have no success in capture-image-and-download.

It randomly fails. sometimes after 3times working fine sometimes after just one try. The first try seems to work fine but than it fails eventually. 

capture-preview works fine.
Even --debug --capture-image-and-download works nicely. So I can't give you a debug file when it failed.

Even more strange: A Canon 760D works perfectly. I only see these problem on a canon eos1200d

What do I need to supply to you so that you have a chance to figure out what causes the problem?

I made the following observation: When execute the command --capture-image-and-download and the autofocus of the camera starts trying to find a focus the command will finish nicely. But when I execute the command and nothing happens on the camera after some time the Error "PTP device busy" comes up.
When I try to execute the command again after the error appeared, sometimes it works sometimes it fails. Other commands like --capture-preview execute just fine after the error.

Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

msmeissn, 2015-12-29:
the code waits for successful focusing, and errors out if that does not happen. it should however not get into Busy state.

can you capture logs of such a failed capture? it should be kind of short logfile then

i again tested the same code with my EOS 100D and I do not see the busyness problem.

Benjamin7785, 2015-12-29:
it seems like it doesn't reach the state of trying to focus when it fails. It probably get stuck before.

As I said, the code works fine for the 760D as well. 

Just curios, but maybe it is something related to the 4digit canon cameras only as they have cheaper chips implemented. :-) 

tomorrow I'll try again to give you a debug log file. However I don't think this will be of big success as I couldn't force the cam to fail when --debug was executed.

Just asking: will the --debug flag slow down the --capture-image-and-download process?

msmeissn, 2015-12-29:
yes, there is some focusing trouble that leads to follow up problems.

it will slow it down a bit, as it will write the logdata to disk

Benjamin7785, 2015-12-29:
...well, in my particular application I wouldn't even need the focusing. because I have the camera in "manual focus" :-)

msmeissn, 2015-12-29:
that you did not write earlier. I added code to git to handle manual focus without a 2 second delay, but I am not sure it will help more of your case

Benjamin7785, 2015-12-29:
okay, here is an interesting one.
Even with --debug it took unusually long before the focus started. However, in the end it finished just fine.

I try to get some more

[logfile_1.txt](https://github.com/gphoto/libgphoto2/files/74382/logfile_1.txt)

Benjamin7785, 2015-12-29:
Thanks for the new code. I'll give it a try tomorrow. I wasn't insisting on a working version by excluding autofocus because the problem then still remains for others.

Benjamin7785, 2015-12-30:
okay, here is another interesting one. as you can see from the logfile, the command eventually finished but with a massive 5.5s delay. From 2.2s to 7.8s.

[logfile_2.txt](https://github.com/gphoto/libgphoto2/files/74786/logfile_2.txt)

I'll keep posting strange events that occur under --debug conditions. most of the time it takes between 4.5 and 7s until the command has been fully executed. Is that a normal behavior? As far as I can tell there is no background activity going on and no other program is running.

Benjamin7785, 2015-12-30:
FYI: I have your new code running but I couldn't see any difference in behavior...

Benjamin7785, 2015-12-31:
hi, 
I tried using piggyphoto because my final project will be done in Python. I know piggyphoto uses libgphoto2 as well so I was wondering if it makes any difference.

The first call of c.capture_image() always fails on the1200d. But consecutive calls always succeed. 
On the eos760d even the first call of c.capture_image() succeeds.

Not sure if this is a pointer for you.

msmeissn, 2016-01-03:
There is a 5.6 seconds stop in the middle of a command in logfile_2.txt:

1.937177 ptp_usb_getresp             (2): Reading PTP_OC 0x9116 (PTP_OC_CANON_EOS_GetEvent) response...
1.937216 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
1.938235 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-25 00 00 00              ....... %...

2.138397 ptp_usb_sendreq             (2): Sending PTP_OC 0x9116 (PTP_OC_CANON_EOS_GetEvent) request...
7.872828 gp_port_write               (3): Writing 12 = 0xc bytes to port...
7.873220 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 16 91-26 00 00 00              ........&...

this is a camera side issue I think, not sure what the camera does at that time. would focusing take so long?

benvb, 2016-04-06:
Hi,

Also in my case i am trying to use a the canon 1200D together with a gphoto2. Which also results in the same error as Benjamin7785 is experiencing. 

Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

I get this error quite random, sometimes after taking 3 pictures successfully, sometimes after 1 and sometimes after 10.

The command i use is the following "gphoto2 --capture-image-and-download"
Also i tried all the "eosremoterelease" options, but with no results.
My application don't need auto focus, so all tests are done with manual focus.

The log file you can find [here](https://www.dropbox.com/s/6qs8fdia7cxa5a6/logfile.txt?dl=0)
The version i am using is 2.5.10.1

Thanks!
Ben

msmeissn, 2016-04-23:
i am still at loss how this is caused. I had a small fix added to current git, which _might_ help. can you try current git?

benvb, 2016-04-26:
I compiled the latest git, but still the same issue.

*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

If you want i send you our 1200D?

msmeissn, 2016-05-08:
i did some tests in the meantime. the device busy messages appear here when there is no focus achieved

the camera seems to always recover from it so far , the issue does not persists

benvb, 2016-05-10:
In my case i use manual focus, so you would think he is not waiting for a focus lock. 1 on 4 times there is no picture taken because of the device busy error. So when you get an error and no picture is taken i just send "gphoto2 --capture-image-and-download" again, this time the change is big it would take a picture.

So in your case it always takes an picture?

benvb, 2016-05-10:
I compiled the latest git again. After taking 5 pictures successfully i got this error again.

*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

benvb, 2016-05-11:
Today i did a test with a version of libgphoto2 from before the 1200D got supported version 2.5.1
With this version i never get an error back when using "gphoto2 --capture-image-and-download"

I don't know if this is helping you.

igos, 2016-05-20:
@benvb to which version are you refering to?

volkerjaenisch, 2017-01-15:
Have the same problem with 2.5.4/2.5.10/2.5.11 on RP3 with Raspbian Jessie on a 1200D.

elannaud, 2017-09-14:
after couple hours to find a solution, bellow my conclusion.
I think after a long time or not enough light, the autofocus cannot calculate the right setting... 
after a problem "Canon EOS Half-Press failed (0x2019: PTP Device Busy)" i run:

gphoto2  --set-config /main/actions/manualfocusdrive=0

It's work!

Also in timelap, put the camera lens Autofocus in manual for avoid the calculation by the camera..

jmunkki, 2017-11-23:
I'm trying to get a Canon EOS T6 (1300D) to capture reliably in manual focus & exposure mode regardless of wether there's anything in focus in the frame. I'm using the current release from python on OS X and the camera seems to become unresponsive after one capture, if nothing is in focus. The code works reliably as long as there's something to focus on.

DEBUG:gphoto2:(camera_canon_eos_capture) focusing - read event type 3
DEBUG:gphoto2:(camera_canon_eos_capture) focusing - read event type 3
DEBUG:gphoto2:(ptp_usb_sendreq) Sending PTP_OC 0x9128 (PTP_OC_CANON_EOS_RemoteReleaseOn) (0x2,0x0) request...
DEBUG:gphoto2:(ptp_usb_getresp) Reading PTP_OC 0x9128 (PTP_OC_CANON_EOS_RemoteReleaseOn) response...
-- long pause --
INFO:gphoto2:(ptp_usb_getresp [usb.c:466]) PTP_OC 0x9128 receiving resp failed: PTP Device Busy (0x2019)
INFO:gphoto2:(camera_canon_eos_capture [library.c:3424]) 'ptp_canon_eos_remotereleaseon (params, 2, 0)' failed: Canon EOS Full-Press failed (0x2019: PTP Device Busy)
INFO:gphoto2:(gp_context_error) Canon EOS Full-Press failed (0x2019: PTP Device Busy)
INFO:gphoto2:(gp_camera_capture [gphoto2-camera.c:1340]) 'camera->functions->capture (camera, type, path, context)' failed: -110

The issue may be timing-related, because if i run gphoto2 --shell --debug, the problem will occur, but if i run gphoto2 --shell, it works OK.

I'm not all that familiar with PTP, but in manual focus mode, the focus loop will exit after just one round, so I suppose that could leave some confusing events in the queue from the failed focusing? (And the result from focusing might take longer because there's nothing to focus on?)

I have also tested this functionality in DarkTable, which uses libgphoto2, and I can't make it fail there. Then again, that program is C code and they seem to be installing custom event handlers for PTP, so the timing & event handling is probably completely different.

jmunkki, 2017-11-26:
I managed to write python code to work around the problems. It's really just a simplified version of the C code with the half-press removed, because I was only interested in not touching the focus at all before capture. For me, this code does exactly what I want. The only weird thing is that it currently doesn't work unless I do the viewfinder toggle before the first capture (something probably isn't initialized right).

I have seen a lot of people ask about software-controlled focus and in writing this code, I incorporated that into it as well.

`import gphoto2 as gp
import logging
import time

def setup_logging():
    logging.basicConfig()
    gp.check_result(gp.use_python_logging(mapping={
        gp.GP_LOG_ERROR   : logging.INFO,
        gp.GP_LOG_DEBUG   : logging.DEBUG,
        gp.GP_LOG_VERBOSE : logging.DEBUG - 3,
        gp.GP_LOG_DATA    : logging.DEBUG - 6}))
    logging.getLogger('gphoto2').setLevel(logging.INFO)

setup_logging()
context = gp.gp_context_new()
camera = gp.check_result(gp.gp_camera_new())
gp.check_result(gp.gp_camera_init(camera, context))

def cameraAction(action, value):
    config = camera.get_config(context)
    target = config.get_child_by_name(action)
    target.set_value(value)
    camera.set_config(config, context)

# based on: https://www.cmcguinness.com/2015/11/using-python-and-gphoto2-to-control-the-focus-of-a-canon-t3i-eos-600d/
def doFocusStep(change, dt=0.2):
    time.sleep(dt)
    cameraAction('manualfocusdrive', change)
    time.sleep(dt)

def fAt(focus=None):
    if focus is not None:
        cameraAction('viewfinder', 1)
        for i in range(1,10):
            doFocusStep('Near 3')
        big = int(focus)
        small = int((focus - big) * 10)
        tiny = int(100 * focus - big * 100 - small * 10)
        for i in range(0,big):
            doFocusStep('Far 3')
        for i in range(0,small):
            doFocusStep('Far 2')
        for i in range(0,tiny):
            doFocusStep('Far 1', dt=0.1)
        cameraAction('viewfinder', 0)

def capAt(focus=None):
    fAt(focus)
    cameraAction('eosremoterelease', 'Press Full')
    cameraAction('eosremoterelease', 'Release Full')
    done = False
    while not done:
        n,file_path = camera.wait_for_event(1000, context)
        if n == 2:
            print file_path.folder, file_path.name
            camera_file = camera.file_get(file_path.folder, file_path.name, gp.GP_FILE_TYPE_NORMAL, context)
            gp.check_result(gp.gp_file_save(camera_file, file_path.name))
        elif n == 1:
            done = True
        elif n != 0:
            print n,file_path

cameraAction('viewfinder', 1)
cameraAction('viewfinder', 0)
capAt()
capAt(3.5)
`

Silex, 2018-06-12:
Seems related to #50

jmunkki, 2018-06-12:
We're using the python & gphoto based software that I wrote in our production now. The code is running on Ubuntu on PCs. Our production team has found that it's nearly unusable on Canon Rebel T5 (1200D) cameras due to a long delay in downloading the photos (about 10 seconds), but works really well on Rebel T3 and T6 cameras. We have a whole bunch of T5s that we can't use right now because of this.

Other than the issues with the T5 cameras, they are really happy with the new software, so many thanks to everyone who has worked on gphoto.