gphoto/libgphoto2 #31: Support for Pentax K-50

travnick, 2015-12-30:
I'm not able shot photo with Pentax K-50:

from gphoto --shell --debug:

```
gphoto2: {.../gphototest/piggyphoto} /> capture-image
393.856012 gp_camera_init              (2): Initializing camera...
393.856045 gp_port_set_error [gphoto2-port.c:1186](0): The supplied vendor or product id (0x0,0x0) is not valid.
393.856086 gp_libusb1_find_device_by_class_lib(2): Looking for USB device (class 0x6, subclass, 0x1, protocol 0x1)...
393.856099 gp_libusb1_find_device_by_class_lib(2): Found USB class device (class 0x6, subclass, 0x1, protocol 0x1)
393.856109 gp_libusb1_find_device_by_class_lib(2): inep to look for is 81
393.856116 gp_libusb1_find_device_by_class_lib(2): Detected defaults: config 1, interface 0, altsetting 0, idVendor ID 25fb, idProduct 0161, inep 81, outep 02, intep 83
393.856121 gp_camera_init              (2): Loading '/usr/lib/libgphoto2/2.5.8/ptp2'...
393.856443 gp_port_open                (2): Opening USB port...
393.856451 gp_libusb1_open             (2): ()
393.856515 gp_libusb1_open             (2): claiming interface 0
393.860132 camera_init                 (2): maxpacketsize 512
393.860175 gp_port_set_timeout         (2): Setting port timeout to 8000 milliseconds.
393.860248 ptp                         (2): PTP: Opening session
393.860269 ptp_usb_sendreq             (2): Sending PTP_OC 0x1002 / Open session request...
393.860303 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
393.860505 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 02 10-00 00 00 00 01 00 00 00  ................

393.860519 ptp_usb_getresp             (2): Reading PTP_OC 0x1002 response...
393.860529 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
393.860720 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-00 00 00 00              ....... ....    

393.860734 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
393.860749 ptp_usb_sendreq             (2): Sending PTP_OC 0x1001 / get device info request...
393.860759 gp_port_write               (3): Writing 12 = 0xc bytes to port...
393.860839 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 01 10-01 00 00 00              ............    

393.860877 ptp_usb_getdata             (2): Reading PTP_OC 0x1001 / get device info data...
393.860888 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
393.861118 gp_port_read                (3): Read    245 = 0xf5 out of 1024 bytes from port: (hexdump of 245 bytes)
0000  f5 00 00 00 02 00 01 10-01 00 00 00 64 00 06 00  ............d...
0010  00 00 64 00 24 6d 00 69-00 63 00 72 00 6f 00 73  ..d.$m.i.c.r.o.s
0020  00 6f 00 66 00 74 00 2e-00 63 00 6f 00 6d 00 2f  .o.f.t...c.o.m./
0030  00 44 00 65 00 76 00 69-00 63 00 65 00 53 00 65  .D.e.v.i.c.e.S.e
0040  00 72 00 76 00 69 00 63-00 65 00 73 00 3a 00 20  .r.v.i.c.e.s.:. 
0050  00 31 00 2e 00 30 00 3b-00 20 00 00 00 00 00 16  .1...0.;. ......
0060  00 00 00 01 10 02 10 03-10 04 10 05 10 06 10 07  ................
0070  10 08 10 09 10 0a 10 0b-10 0c 10 0d 10 0f 10 14  ................
0080  10 15 10 16 10 1b 10 01-98 02 98 03 98 05 98 04  ................
0090  00 00 00 03 40 04 40 05-40 02 40 05 00 00 00 01  ....@.@.@.@.....
00a0  50 11 50 06 d4 07 d4 03-d3 00 00 00 00 07 00 00  P.P.............
00b0  00 01 30 06 30 0a 30 0d-30 01 38 00 38 00 30 07  ..0.0.0.0.8.8.0.
00c0  50 00 45 00 4e 00 54 00-41 00 58 00 00 00 05 4b  P.E.N.T.A.X....K
00d0  00 2d 00 35 00 30 00 00-00 05 31 00 2e 00 30 00  .-.5.0....1...0.
00e0  31 00 00 00 08 36 00 31-00 36 00 39 00 38 00 37  1....6.1.6.9.8.7
00f0  00 33 00 00 00         -                         .3...           

393.861151 ptp_usb_getresp             (2): Reading PTP_OC 0x1001 response...
393.861161 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
393.861343 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-01 00 00 00              ....... ....    

393.861372 print_debug_deviceinfo      (2): Device info:
393.861381 print_debug_deviceinfo      (2): Manufacturer: PENTAX
393.861389 print_debug_deviceinfo      (2):   Model: K-50
393.861397 print_debug_deviceinfo      (2):   device version: 1.01
393.861405 print_debug_deviceinfo      (2):   serial number: '6169873'
393.861413 print_debug_deviceinfo      (2): Vendor extension ID: 0x00000006
393.861421 print_debug_deviceinfo      (2): Vendor extension version: 100
393.861429 print_debug_deviceinfo      (2): Vendor extension description: microsoft.com/DeviceServices: 1.0; 
393.861437 print_debug_deviceinfo      (2): Functional Mode: 0x0000
393.861445 print_debug_deviceinfo      (2): PTP Standard Version: 100
[... 76 lines cut ...]
0010  00 00 00 00 ff ff ff ff-                         ........        

393.863102 ptp_usb_getdata             (2): Reading PTP_OC 0x1007 / Get object handles data...
393.863111 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
393.863613 gp_port_read                (3): Read    20 = 0x14 out of 1024 bytes from port: (hexdump of 20 bytes)
0000  14 00 00 00 02 00 07 10-04 00 00 00 01 00 00 00  ................
0010  00 40 00 04            -                         .@..            

393.863630 ptp_usb_getresp             (2): Reading PTP_OC 0x1007 response...
393.863636 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
393.863714 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-04 00 00 00              ....... ....    

393.863724 ptp_list_folder             (2): adding old objectid 0x04004000 (nrofobs=1,j=0)
393.863738 gp_context_error            (0): Sorry, your camera does not support generic capture

*** Error ***              
Sorry, your camera does not support generic capture
393.863759 gp_camera_capture [gphoto2-camera.c:1032](0): 'camera->functions->capture (camera, type, path, context)' failed: -6
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***
```

I'm familiar with C++ programming so maybe I'm able to help you with bringing support of this camera to libgphoto2

msmeissn, 2016-02-20:
The pentax can not capture in PTP mode, but in the other (mass storage mode).

However ... We have some support for other Pentax cameras, but so far not the K50 :(

Blueshawk, 2017-01-22:
I'm following up on this as I'd also like to see the K-50 supported by Gphoto. In my case for use with Indilib/Ekos for astronomy. I'd like to suggest comparing and integrating code from the more active and up to date PKtriggercord. While also based on Pkremote, PKtriggercord has been actively updating and is currently functional on the k50 for capture,iso and exposure. Perusing the newer routines might reduce the work needed to bring that library up to speed with the newer cameras.  
I'd also put myself and "Pentax the Red!"(magic tavern reference) up as testers for this purpose. I would take a naive stab at hacking the two together myself but I can't get the thing to compile as yet.

Blueshawk, 2017-02-04:
I notice that the issue has been picked up, and know just how busy you folks are from the email digest (lurking). I've been busy working on other problems myself but have every intention of being available when the time comes. When you get time to look it over just let me know if there's anything I can do to help.

msmeissn, 2017-02-16:
(currently not working on this due to lack of time and lack of testhardware to verify ... mostly due to lack of time :( )

Blueshawk, 2017-02-16:
I understand Marcus, it's no problem. I'll be around if you get into it. I worked out a way to cut power to the Pentax instantly if it decides to go unstable as has happened a few times in the past. Is there any easy way to enable the k-50 for testing? If I could get it recognized I might be able to debug a bit on my own. I poked around in the source a bit and tried replacing the k30 references with k50 as I had read they were similar but failed to get it to register.

I've been watching the action by way of the email list and there are lots of things going on already, so I figured I'd have to be patient. Meanwhile, I'm working on other things myself in my spare time, (getting my arduino telescope to talk to indilib etc.) including a work around that (hopefully) all us astronomy-AP guys can use to integrate our non capture cameras (eos-m/ Pentax k-50 in my case) into Indi/ekos by way of a flash air or other wifi card and a usb trigger command to an external IR trigger. but this project would help a ton of Pentaxians get tethered too.

msmeissn, 2017-03-13:
Would you be able to build and test libgphoto2 git? 

i tried a port integrating pktriggercord and it was not so bad as I feared. 
But it will probably need some test turnarounds.

Blueshawk, 2017-03-13:
I'll get right on it!

Blueshawk, 2017-03-13:
Just to be sure, I made a fresh directory clone of current git and compiled from that but I don't see any new work at git master. Has master been updated, or am I showing my newbness in pulling from the wrong place? 
(so excited! and off work today too)

Blueshawk, 2017-03-13:
Here's the debug file I missed in previous post.
Starting without a settings file

Camera not found/enabled. 
[k50test1.txt](https://github.com/gphoto/libgphoto2/files/839337/k50test1.txt)

msmeissn, 2017-03-14:
i was not yet readsy there. ;)

can you checkout the pentax branch and check that out?

ig you have a checkout, just do "git checkout pentax"  and then make

What USB ID does it have in MSC mode, this probably the first thing to add

Blueshawk, 2017-03-14:
I had no notebook today as I forgot it this morning. Home now, and will check it out.

(DST sucks)

Blueshawk, 2017-03-14:
Git still throws me for a loop sometimes but I got it eventually. - Just for anyone following along, I did 
$ git checkout -b pentax to get on the branch 
then 
$ git pull https://github.com/gphoto/libgphoto2 pentax
...to update it. seemed to do the trick, lots of new code. :)

Still no joy on the camera ID in msc. 
Here's the usb info on the camera trimmed from lsusb -v. 
Bus 001 Device 099: ID 25fb:0160 
 idVendor           0x25fb 
 idProduct          0x0160 
The rest is here if needed: [k50info.txt](https://github.com/gphoto/libgphoto2/files/842956/k50info.txt)

And another debug just in case:
[k50debug.txt](https://github.com/gphoto/libgphoto2/files/842993/k50debug.txt)

msmeissn, 2017-03-15:
can you git pull again, remake, reinstall.

grep K50D camlibs/pentax/library.c  should now show it.

see if that helped?

Blueshawk, 2017-03-15:
Got it, built on my notebook at work and tested during a break. It found the camera, and even worked the shutter with --captture-image-and-download but only after declaring an error. lots of err. info is going to screen but for some reason isn't picked up by bash/xterminal to grab from scrollback.  I'll get some files up and corral that info when I get home. I noticed --config worked some too.  which should be in the capture/debug I'll get in a while.  good progress! 
note this is on my acer notebook running 16.04 xubuntu
[k50d-test1.txt](https://github.com/gphoto/libgphoto2/files/845825/k50d-test1.txt)
[debug.output to file.txt](https://github.com/gphoto/libgphoto2/files/845831/debug.output.to.file.txt)

[config.log.txt](https://github.com/gphoto/libgphoto2/files/845832/config.log.txt)

msmeissn, 2017-03-15:
i need to hook up the bulk support i think, its not there yet.

I changed the dbeugging in git to log to the gphoto2 logfile (my hookin code was incorrect)

msmeissn, 2017-03-15:
it would be nice to see what fails with capture-image  :)

Blueshawk, 2017-03-15:
Much quieter to screen now. Still gives an error but went ahead and worked the shutter. No capture file is apparent on the drive or sdcard.
I got this to screen when I did the test. 
-----
0x0000 | 00 00 00 02 00 00 00 00  01 00 02 10 00 00 00 00                      
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  
0x0020 | 00 00 00 0f 00 00 00 02  00 00 00 00 00 00 00 00  
0x0030 | 00 00 01 00 00 00 00 03  00 00 00 0a 00 00 00 50  
0x0040 | 00 00 00 0a 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0050 | 00 00 00 03 00 00 00 0a  00 00 00 00 00 00 00 03  
0x0060 | 00 00 00 00 00 00 00 00  00 00 03 20 00 00 00 64  
0x0070 | 00 00 0c 80 00 00 00 00  00 00 00 07 00 00 00 07  
0x0080 | 00 00 00 02 00 00 00 00  00 00 00 00 00 00 00 01  
0x0090 | 00 00 00 00 00 00 00 04  00 00 00 04 00 00 00 04  
0x00a0 | 00 00 00 00 00 00 00 01  00 00 00 00 00 00 00 01  
0x00b0 | 00 00 00 00 00 00 00 08  00 00 00 00 00 00 00 02  
0x00c0 | 00 00 00 01 00 00 00 00  00 00 00 00 00 00 00 02  
0x00d0 | 00 00 00 00 00 00 00 03  00 00 00 07 00 00 00 01  
0x00e0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 01  
0x00f0 | 00 00 00 01 00 00 00 00  00 00 13 88 00 00 00 04  
0x0100 | 00 00 00 00 00 00 00 01  00 00 00 00 00 00 00 03  
0x0110 | 00 00 00 0a 00 00 00 50  00 00 00 0a 00 00 00 00  
0x0120 | 00 00 00 0a 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0130 | 00 00 00 00 00 00 03 20  00 00 00 00 00 00 00 01  
0x0140 | 00 00 00 00 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0150 | 00 00 00 0a 00 00 00 02  00 00 00 00 00 00 00 50  
0x0160 | 00 00 00 0a 00 00 00 00  00 00 00 00 00 00 02 50  
0x0170 | 00 00 03 77 00 00 02 fd  00 00 00 00 00 00 00 00  
0x0180 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  
0x0190 | 00 00 00 87 00 00 00 00  00 00 00 00 00 00 00 ca  
0x01a0 | 00 00 0b f7 00 00 00 64  00 00 00 3c ff ff ff d8  
0x01b0 | 00 00 00 00 00 00 00 00  00 00 00 0f 00 00 00 3f  
0x01c0 | 00 00 00 00 00 00 00 00  
ERROR: Could not capture image.
ERROR: Could not capture.
-----

Here's the debug file from the second try. First try appended an old file and I realized it was getting big. 
[k50debug1.txt](https://github.com/gphoto/libgphoto2/files/846115/k50debug1.txt)

msmeissn, 2017-03-16:
I added some more debugging to git. if you can rebuild and rerun the test? 
I think the image type is not correctly detected, but it was not debug priunted.

I also added "bulb" as config option, so --bulb seconds   should produce some form of result, but it will probably encounter a failure similar to capture-image still.

Blueshawk, 2017-03-16:
I think you are right about image type. I changed it to .jpg only and it captured a file on camera. I often use raw+ which makes one of each, or just raw for post processing. --bulb=45 appears to have set the camera exposure to 30 seconds, but also had the effect of never finishing. I also saw similar with pktriggercord in the past however so that may be "next tier" problem.

Rebuilt and tested with capture-image: 
[k50d-test2.txt](https://github.com/gphoto/libgphoto2/files/847711/k50d-test2.txt)

Tested with --capture-image --bulb=10 ...no shutter , waited then ctrl-c out.
[k50d-bulbtest1.txt](https://github.com/gphoto/libgphoto2/files/847715/k50d-bulbtest1.txt)

msmeissn, 2017-03-16:
so, the format your camera is set to RAW_PLUS. I added code similar to handling RAW
(DNG?)

I also add a shutter release in bulb support. Can you check both again?

Blueshawk, 2017-03-16:
Yes. It saves RAW format as .DNG and the + setting makes it go ahead and develop a .JPG file of the same name. I usually shoot raw or raw+ for astronomy.
The bulb test was similar to before with the process stopping just after showing the bulb value and the hex dump. I let it run a bit before breaking out. The camera engaged the shutter when I did and it dropped out when I turned the camera power off. I suspect this may be the problem I was working on in pktriggercord last week. I had the idea that it might procedural problem and started learning the code but haven't isolated it. I was reading both Gphoto and Pktriggercord code when it occurred to me that this project might be a thing worth trying(still excited!). The k-50 has a menu item that sets bulb type to hold button or push on/off that may be offsetting the command from protocol somehow - just a hunch. I'd really like to get that working though as it's the number one thing we use in long exposure astrophotography(AP). It might also be worth noting that PKtriggercord usage was different and would dump photo code to terminal if you didn't give it a file name. Might be some weirdness involved there to check out.

 I switched the dial to bulb mode for this with no change.
[k50d-bulb2.txt](https://github.com/gphoto/libgphoto2/files/848328/k50d-bulb2.txt)

The capture-image test still reported capture error, worked the shutter, and no file report. I made sure it was in raw+ and dialed to user mode with M(manual) 
[k50d-test3.txt](https://github.com/gphoto/libgphoto2/files/848327/k50d-test3.txt)

msmeissn, 2017-03-16:
Ok , I forgot some break; statements. that should cure the --capture-image cases now.

will check bulb later

Blueshawk, 2017-03-16:
Failed to finish. Looks like it got stuck at the file type parsing again for some reason as log shows. 
did not report capture error to screen this time though.
[k50d-test4.txt](https://github.com/gphoto/libgphoto2/files/848653/k50d-test4.txt)

Blueshawk, 2017-03-16:
Appreciate all the attention. :) I've had my k50 at the workbench with me all day while I work on PLC's in a windows vm so I could just bang out tests.

msmeissn, 2017-03-16:
i am trying to move this a bit forward ;)

I now added a commit handling RAW_PLUS to the save function. (the single image capture modes should work already, RAW_PLUS needs a bit work)

Blueshawk, 2017-03-16:
Oh rats. I found a problem on my end. The dirty bit got set on the card at some point. I'll get that cleaned up and retest. I have it set to not mount but probably failed to eject it after checking at some point then repowered the camera.

Blueshawk, 2017-03-16:
Cleared card formatted with camera(easiest solution) and retested. Switched to .JPG to test. It still reports 
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0001.dng on the camera
But it isn't to be found. 
I tried --capture-image-and-download and it worked okay, then deleted the photo from camera. The file reads okay, shows a pic of junk on my desk.

ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0001.jpg on the camera
File capt0001.jpg exists. Overwrite? [y|n] n
Specify new filename? [y|n] y
Enter new filename: capt0002
Saving file as capt0002
Deleting file /capt0001.jpg on the camera

Last couple tests. sorry about the size/stacking in it. I'll do more when I get home.

[k50d-test5.txt](https://github.com/gphoto/libgphoto2/files/848930/k50d-test5.txt)

msmeissn, 2017-03-16:
yeah, capture-image-and-download is needed, as the file is only in temporary memory on the camera and in libgphoto2. 
I changed capture-image hopefully to also return an image already, so hopefully this error goes away.

Blueshawk, 2017-03-17:
Ah, that explains the missing file.  
Guess who left the only working pentax cable sitting on the workbench at work. :1st_place_medal: 
I saw some updates so blundering forward with another test this morning.
Got a jpg on the first try:
[k50d-test6-jpg.txt](https://github.com/gphoto/libgphoto2/files/850561/k50d-test6-jpg.txt)

Raw+ test downloaded a .DNG file but raw readers would not open it. No jpg was found. 
and looked like this:
hawk@hawk-Aspire-R3-131T:~/libgphoto2$ gphoto2 --debug --debug-logfile=k50d-test6-rawpl.txt --capture-image-and-download
0x0000 | 00 00 00 02 00 00 00 00  01 00 02 10 00 00 00 00                      
0x0010 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  
0x0020 | 00 00 00 0f 00 00 00 02  00 00 00 00 00 00 00 00  
0x0030 | 00 00 01 00 00 00 00 01  00 00 00 0d 00 00 00 32  
0x0040 | 00 00 00 0a 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0050 | 00 00 00 03 00 00 00 0a  00 00 00 00 00 00 00 03  
0x0060 | 00 00 00 00 00 00 00 00  00 00 03 20 00 00 00 64  
0x0070 | 00 00 0c 80 00 00 00 00  00 00 00 07 00 00 00 07  
0x0080 | 00 00 00 02 00 00 00 00  00 00 00 00 00 00 00 01  
0x0090 | 00 00 00 00 00 00 00 04  00 00 00 04 00 00 00 04  
0x00a0 | 00 00 00 00 00 00 00 01  00 00 00 00 00 00 00 01  
0x00b0 | 00 00 00 00 00 00 00 08  00 00 00 00 00 00 00 02  
0x00c0 | 00 00 00 01 00 00 00 00  00 00 00 00 00 00 00 02  
0x00d0 | 00 00 00 00 00 00 00 03  00 00 00 07 00 00 00 01  
0x00e0 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 01  
0x00f0 | 00 00 00 01 00 00 00 00  00 00 13 88 00 00 00 04  
0x0100 | 00 00 00 00 00 00 00 01  00 00 00 00 00 00 00 01  
0x0110 | 00 00 00 0d 00 00 00 32  00 00 00 0a 00 00 00 00  
0x0120 | 00 00 00 0a 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0130 | 00 00 00 00 00 00 03 20  00 00 00 00 00 00 00 00  
0x0140 | 00 00 00 00 00 00 00 00  00 00 00 0a 00 00 00 00  
0x0150 | 00 00 00 0a 00 00 00 02  00 00 00 00 00 00 00 50  
0x0160 | 00 00 00 0a 00 00 00 00  00 00 00 00 00 00 02 50  
0x0170 | 00 00 03 37 00 00 02 f0  00 00 00 00 00 00 00 00  
0x0180 | 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  
0x0190 | 00 00 00 87 00 00 00 00  00 00 00 00 00 00 00 ca  
0x01a0 | 00 00 0d ac 00 00 00 64  ff ff ff f6 ff ff ff d8  
0x01b0 | 00 00 00 00 00 00 00 00  00 00 00 0f 00 00 00 3f  
0x01c0 | 00 00 00 00 00 00 00 00  
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0001.dng on the camera
Saving file as capt0001.dng
Deleting file /capt0001.dng on the camera

[k50d-test6-rawpl.txt](https://github.com/gphoto/libgphoto2/files/850608/k50d-test6-rawpl.txt)

The raw only test resulted in a 0 length  .DNG file. after I said Y to overwrite. also after deleting the file and retrying.
[k50d-test6-raw.txt](https://github.com/gphoto/libgphoto2/files/850630/k50d-test6-raw.txt)

msmeissn, 2017-03-17:
k50d-test6-rawpl.txt

shows a JPEG file being downloaded, so capt0001.dng there might be .jpg instead... can you chek with "file capt0001.dng"  ? 

In the dual mode capture I need to detect the types better :/ 

k50d-test6-raw.txt only has a 0 byte files as you saw.

I added the status display code to --summary 
What you can try for instance:

gphoto2 --capture-image-and-download --summary

perhaps we can see something interesting there too

Blueshawk, 2017-03-17:
I thought that might be the case too, because geeqie(which uses jpg/raw grouping for thumbnails) loaded it when rawtherapee and ufraw would not. Sure enough, renaming to .jpg loads in viewers right away. this may be related somehow to the dual(DNG+JPG) file output of raw+ mode. 
Heres the output of summary: 
Camera summary:
Pentax K DSLR capture driver.
Using code from pktriggercord by Andras Salamon.
Collected Statatus Information:
current iso                     : 800
current shutter speed           : 1/13
camera max shutter speed        : 1/6000
current aperture                : 5.6
lens max aperture               : 40.0
lens min aperture               : 5.6
set shutter speed               : 1/13
set aperture                    : 5.6
fixed iso                       : 800
auto iso                        : 100-3200
jpeg quality                    : 4
jpeg resolution                 : 16M
jpeg image tone                 : Natural
jpeg saturation                 : 4
jpeg contrast                   : 4
jpeg sharpness                  : 4
jpeg hue                        : 4
zoom                            : 47.50 mm
focus                           : 0
color space                     : sRGB
image format                    : RAW+
raw format                      : DNG
light meter flags               : 32
ec                              : 0.00
custom ev steps                 : 1/3
custom sensitivity steps        : 0
exposure mode                   : 6 (NONE)
user mode flag                  : 2
ae metering mode                : Spot
af mode                         : AF.S
af point select                 : Auto-5
selected af point               : 0
focused af point                : 0
drive mode                      : SelfTimer-2
auto bracket mode               : off
auto bracket picture count      : 0
auto bracket picture counter    : 0
auto bracket ev                 : 0.30
shake reduction                 : off
white balance mode              : Auto
white balance adjust            : 0
flash mode                      : Manual
flash exposure compensation     : 0.00
manual mode ev                  : 1.30
lens                            : smc PENTAX-DA L 18-55mm F3.5-5.6 AL WR
battery                         : 7.95V 7.23V 0.00V 0.00V

Here's the debug from that, file got pretty big. 
[k50d-test7-rawpl.txt](https://github.com/gphoto/libgphoto2/files/851497/k50d-test7-rawpl.txt)

I also ran raw only again:
[summary.txt](https://github.com/gphoto/libgphoto2/files/851512/summary.txt)
[k50d-test7-raw.txt](https://github.com/gphoto/libgphoto2/files/851513/k50d-test7-raw.txt)

msmeissn, 2017-03-21:
the raw only was empty, as the code did not handle raw downloads yet (only jpeg downloads).

I fixed that in GIT I hope.

I have some ideas for the dual-mode download.

msmeissn, 2017-03-21:
(thats why the second trace for raw downlooad is so short... its 0 bytes, as there is no JPEG)

Blueshawk, 2017-03-21:
I see. Okay I'll repeat the raw tests then. The raw+ did the same thing as previous, as expected?
Never hurts to have a fresh debug: 
[k50d-RAWpl.txt](https://github.com/gphoto/libgphoto2/files/858854/k50d-RAWpl.txt)

The RAW only (DNG) shuttered okay, but failed to download a file. 
[k50d-RAW.txt](https://github.com/gphoto/libgphoto2/files/858830/k50d-RAW.txt)

msmeissn, 2017-04-04:
I added some dual mode (rawplus) capture code now. you should be seeing both images being downloaded.

I need to add some waiting after delete I think, it currently returns some images twice

Blueshawk, 2017-04-05:
I grabbed my camera soon as I noticed the message. :)

Here's a summary from the first test to show where I had things set:
[summary.txt](https://github.com/gphoto/libgphoto2/files/894810/summary.txt)

I had a little trouble and found that while it wasn't mounted, the msc drive being recognized seemed to be an issue. mounting then ejecting it in pcmanfm seemed to solve it and returned to expected operation.
From console:
" ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0000.jpg on the camera
ERROR: Could not get image.
ERROR: Buggy libcanon.so? "

The raw+ test gave me 4 files as you mentioned, the raw is now actually a raw image and loaded right up in darktable, accepting a raw only command to filter noise(as a test). The jpg is okay as well. I did notice a slight exposure difference between the two but it was not reported in exif. I thought I heard a double shutter as if in bracket mode as well(which gives 3 images, I.e.+1,-1,0) but isn't when I tested it locally. May be related to the timing/delay issue you mentioned.  More testing shows the shutter gets erratic in its timing, sometimes firing or working once before going ahead and doing the actual 2sec/remote delay shutter I usually use on the mount. It had been working okay with it but to be more sure I switched the camera to regular single mode and retested.
[k50-dngp.txt](https://github.com/gphoto/libgphoto2/files/894853/k50-dngp.txt)

This is output to console with the console hex dump omitted for clarity - repeating the command. 
hawk@hawk-Aspire-X1420:~/libgphoto2$ gphoto2 --capture-image-and-download --debug --debug-logfile=k50-dngp.txt 
ERROR: Could not capture image.
ERROR: Could not capture.
New file is in location /capt0000.jpg on the camera
ERROR: Could not get image.
ERROR: Buggy libcanon.so?
hawk@hawk-Aspire-X1420:~/libgphoto2$ gphoto2 --capture-image-and-download --debug --debug-logfile=k50-dngp.txt 
New file is in location /capt0000.dng on the camera
Saving file as capt0000.dng
Deleting file /capt0000.dng on the camera
New file is in location /capt0000.jpg on the camera
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
New file is in location /capt0001.dng on the camera
Saving file as capt0001.dng
Deleting file /capt0001.dng on the camera
New file is in location /capt0001.jpg on the camera
Saving file as capt0001.jpg
Deleting file /capt0001.jpg on the camera

Debug of RAW only:
[k50-dng.txt](https://github.com/gphoto/libgphoto2/files/894871/k50-dng.txt) 
I changed the debug level to "error" to help keep file size manageable, let me know if you need the really big (reporting 100+meg for some reason) version, though it may take a while to upload at home. probably wait till i'm at work to try, and zipping may help.

I managed to get the camera stuck twice breaking out of commands, and also managed to set the dirty bit again. Formatted with camera before the last couple tries. 

Off to go Jupiter hunting with the new asi178 I just got! :)

msmeissn, 2017-04-06:
I added some more waits to the capture function, so the errors should be gone now.

Blueshawk, 2017-04-06:
Like a good bike saddle, nothing much to report when it works. I got the right files and right amounts of files this time in all tests jpg, raw, and raw+. :dancing_women: 
[k50jpg.txt](https://github.com/gphoto/libgphoto2/files/903846/k50jpg.txt)
[k50d-RAW.txt](https://github.com/gphoto/libgphoto2/files/903834/k50d-RAW.txt)
[k50d-RAWpl.txt](https://github.com/gphoto/libgphoto2/files/903835/k50d-RAWpl.txt)
Might be time to look at some of the other config stuff? 

  I've been switching modes locally raw/etc. because the config takes the command but no change is made. Also that pesky bulb problem still eludes me - I stopped digging in the pktriggercord code once you started this since it seemed it might become redundant and your wisdom on this is going to be miles ahead of my meager efforts. the library called bulb correctly but it responded incorrectly as when I tried it with pktriggercord-cli. Before I stopped looking I thought the bulb setup might have been offset by a version change but I found a guy with a k30 who also reports having bulb press type 1-2 in the extra menus. I had noticed that the bulb setting I usually keep on 2 for press open-press closed changed to 1 after calling --bulb in pktirggercord. It seems to recognise the mode as bulb, and your code did get a shutter if I remember right, but never timed out or locked up. not sure which. I'll redo that test when I get home.
Would a todo list help? I could do a more complete checkout of --config as well if needed.

Blueshawk, 2017-04-07:
Running some tests at home, confirmed the proper downloads again on another computer. Nice!
Also noted the internal 2 second shutter delay works correctly and does not interfere.
I pulled something from my notes on pktriggercord that I thought might be of interest: 
" I think the return from status call may not be passing the string to status, but instead a number, presumably from the camera? ..so I did some more tests and ran through the dial collecting data."
(note the gui mode [user1/2 dial position changes on screen in software] switch to also does not work) 
dialed exposure modes (read from status)
7 (none) = B Bulb 
6 (none) = M Manual
5 (none) = Tav 
4 (none) = Av
3 (none) = Tv
2 (none) = Sv
0 (none) = P
1 (STANDARD) = movie  (movie camera icon) - odd,fired shutter during status
0 (MACRO) = green colored "auto" () shows submode
1 (LANDSCAPE) = "SCN" scene mode.() shows what scene is picked."
when I put the number into the bulb function parse it acted more correctly, but I still hadn't gotten it to shutter. 
I think this may be related to several problems with mode changes.

Blueshawk, 2017-04-07:
Bulb test with bulb mode1 and user1 set to bulb manually in camera
 gphoto2 --capture-image-and-download --bulb 5 --debug-loglevel=debug --debug-logfile=k50bulbjpg.txt
Bulb mode enabled (exposure time: 5s). 
[omitted hex code console output]                      
pentax/pslr.c:869:get_status(p->fd) failed: 129
New file is in location /capt0000.jpg on the camera
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
pentax/pslr.c:869:get_status(p->fd) failed: 129

camera shuttered but photo has 1/25 second exposure. (press and hold probably not the method to use)

[k50bulb.txt](https://github.com/gphoto/libgphoto2/files/904247/k50bulb.txt)

Next I changed to mode 2, dial user1 and gui set bulb
Camera opened shutter, errors came at time 5=seconds but had to break out and restart camera to close shutter.
pentax/pslr.c:869:get_status(p->fd) failed: 129
pentax/pslr.c:869:get_status(p->fd) failed: 129
^C
Cancelling...
^C
[k50bulb2.txt](https://github.com/gphoto/libgphoto2/files/904265/k50bulb2.txt)

 if you have other stuff i'll be glad to go in whatever order you like and as time allows of course. I figured it never hurts to send more data. :)

msmeissn, 2017-04-08:
The shootingmode values should now match what is on the camera, I was using the wrong enum.

so switch the camera mode dial to "B", plug it in, then run:

gphoto2 --set-config bulb=1 --wait-event=5s --set-config bulb=0 --wait-event-and-download=5s

I now merged the pentax branch back into GIT "master" branch for further work and the next release. "git checkout master" and then "git pull" as usual...

Blueshawk, 2017-04-08:
Good morning! This looks like a great thing to wake up to. :fireworks: 
Well it almost worked.  The shutter opened but never closed. After waiting long enough, I manually pushed the shutter button and the routine then downloaded files but kept on going and the camera became unresponsive. After pulling power to reset it, I tried setting the extras menu item to bulb=1 (press and hold) and it worked the shutter/released the first time but not the second which seems right. looks to me like the camera didn't like bulb=0 command. 
Because I had to break out the debug may be incomplete. I can try some more ways if needed.
[k50bulb.txt](https://github.com/gphoto/libgphoto2/files/907527/k50bulb.txt)

Blueshawk, 2017-04-08:
(head clearing) ..sorry bout that.
  When I did the command you sent I got a quick shutter open/close and then it waited, then downloaded the (very dark) picture. I checked and it was set to bulb mode 2 press/press so I changed it to mode 1 and tried again. It did similar to before with open shutter and time down but not close it. 
*** I tried something else which may be a clue. I set to mode 2 and changed the command to have the bulb 0 be another bulb 1 like this:
libgphoto2$ gphoto2 --debug --debug-logfile=k50bulb.txt --debug-loglevel=debug --set-config bulb=1 --wait-event=5s --set-config bulb=1 --wait-event-and-download=5s 
And it triggered the shutter both times! I'm thinking maybe it toggled back on its own? Hope this helps.

Both commands landed in the one file.
[k50bulb.txt](https://github.com/gphoto/libgphoto2/files/907626/k50bulb.txt)

In other tinkering, I got my first photo from the k50 in indilib/ekos last night! The exposure change doesn't work and the mode set causes issues as expected considering my previous report on it but it was neat to see it connect and give me results.

msmeissn, 2017-04-09:
You need to send --set-config bulb=0  at the end of bulb capture.

It seems not be doing bulb capture though, as the bulb=0 reports some error back from the camera.

It seems to access the filesystem at least.

gphoto2 --debug --debug-logfile=k50bulb.txt --debug-loglevel=debug --set-config bulb=1 --wait-event-and-download=5s --set-config bulb=0 --wait-event-and-download=5s

I suspect it takes a regular non-bulb image for some reason :/

Blueshawk, 2017-04-10:
hmm. Did my second try overwrite the first in the debug? I did it correctly the first time but did not change the file name on the second (experimental) try. I'll redo that test asap.

The camera did take a very short exposure image with the proper command. The bulb=1 may be changing back to 0 by itself in camera 

My improper use version was a troubleshooting experiment. It shows that the command will toggle shutter each time bulb=1 is called. I noticed in the pktriggercord code that there is a shutter call if the bulb mode is not found, which leads me to believe that routine is not parsing bulb because it's getting a number instead of the name.

Blueshawk, 2017-04-10:
Just to try to be clear I did it again and split things up better. :)

gphoto2 --debug --debug-logfile=k50bulb.txt --debug-loglevel=debug --set-config bulb=1 --wait-event-and-download=5s --set-config bulb=0 --wait-event-and-download=5s

Image exif shows 1/25s
This one produced the file:
[k50bulb - 1 press and hold.txt ](https://github.com/gphoto/libgphoto2/files/911102/k50bulb.-.1.press.and.hold.txt)

This one opens shutter and hangs.
[k50bulb-2 press start press stop.txt](https://github.com/gphoto/libgphoto2/files/911105/k50bulb-2.press.start.press.stop.txt)

msmeissn, 2017-04-10:
Thanks... I looked at it and the bulb command gets a status of 0x81 ... 

Good status would be 0x00 (ok) or 0x01 (busy, wait a bit) , but 0x81 seems to indicate something went wrong or was unexpected.

So it just executed the psl_shutter command, which then took a short picture.

The same thing works for my K10D, so the K50D seems to have a different thing going on.

Does bulb mode work on the camera itself?

Blueshawk, 2017-04-10:
Bulb mode works. I use an IR remote to fire bulb on the telescope in press press mode. Does the k10D have the two different bulb modes in the menu?

msmeissn, 2017-04-10:
I dont see any mention of bulb in the cameras menus. I can turn the dial to "B" but there seems no other modification possibility. Pressing the shutter button starts bulb, releasing it ends bulb capture.

Blueshawk, 2017-04-11:
Ideas, theory and hypothesis:

Idea: Since the first command set config bulb=1 also locks the shutter open okay, maybe doing something like calling shutter directly I.E. shutter/time/shutter while menu bulb mode=2 (+dial set to bulb) would make it close and finish up..

 Since the K10D is using the press and hold method (bulb=1/wait/bulb=0) then bulb=1 is shutter open in that model. My suspicion is that the later models bulb=1 sets the menu bulb input type. I'm pretty sure I saw bulb change from 2 to 1 in one of my early tests which made me suspect this. I saw it because I usually use mode 2 in normal(nighttime) use and found it had changed to 1 in the menu. 

Hypothesis: Is bulb=1 acting on the menu instead of a set shuttering routine as in the earlier models?
 I'm out of time tonight but i'll try to verify that as soon as I get a chance. 

If so, enabling the routine to --set-config bulb=2(which currently errors) for later models might prove useful in the long term, perhaps by being able to set the control mode at run time. 

On another note, this [article](http://support.us.ricoh-imaging.com/node/1648) shows the K-30 also has the menu items. I think that model may be when things changed, The trick is going to be finding out just how they differ.

Blueshawk, 2017-04-11:
I was unable to prove my theory about the menu change. It stayed the same when I set it to mode 2, ran the command then returned to the menu so that's out. I now think I must have seen it because the user mode reloaded from a previous save that happened to have it at mode 1.
I'm still hopeful about a repeated shutter command working for the mode=2 menu setting(press on press again off) in the K-50 but after trying several command combinations I couldn't find one using existing code that would retrigger the shutter without waiting for a return event.

msmeissn, 2017-04-12:
One thing is that even sending the initial pslr_bulb() command returns an error already, without any shutter presses.

It could also be that the bulb commands needs to be sent differently for the K50 :/ but it might be hard to find out without sniffing windows software.

Blueshawk, 2017-04-13:
Doing a bit of research. I've done usb sniffing before with PLC's and other electronic equipment so that is an option. I'll have to find some functional windows software to use for testing. I'm also looking at possibly using the firmware decryption that was done previously to get a new command list. Tricky business there though and while it's probably what we need, it's more of a long term answer and possibly a huge bucket-o-worms. I've done a fair amount of fishing in my time though. :D 

  I wonder if there's a way to just "not" call pslr_bulb in the newer cams and use something like, shutter/timer/shutter-wait/download instead. Since the bulb menu mode 2 option uses two button presses or two IR remote shutter calls , it may react nicely to two usb shutter calls in bulb mode as well.

Blueshawk, 2017-09-08:
Eureka!  I realize this is not strictly related to this issue but I wanted to share my discovery in hopes of helping more people. I was trying to work around the fiddly indilib drivers and hit upon a workaround for the K-50 bulb problem. Unlike some other cameras I've tried, the K-50 leaves its hardware trigger port enabled. I found out by taking apart a cellphone size plug and making a test plug with it. Then I ran: gphoto2 --wait-event-and-download and touched two of the wires together, and it worked!  The focus trigger also works. This means hardware triggers can be used to get around the problem. They usually work by waiting for an active call to toggle on their attached usb port, to the effect of "cts low - cts high" This means I might be able to  use a workaround in indilib/ekos similar to the one used for Nikon to get bulb to work on my remote telescope! I know it's not a great solution but at least now we have a direction to send folks on the issue of bulb in the k50d. 
Cheers!

Dabi01, 2017-09-11:
It's great to hear that development goes well!
I don't have my camera right now, but I can test as soon as I got it

Blueshawk, 2017-09-17:
We look forward to hearing about your experience @Dabi01 hope it works for you. :)
@msmeissn Following up on the idea of an external trigger, I made my own that uses usb/serial handshaking(RTS_flag) to trigger externally using the jack on the side of the k-50. I managed to get Indilib/ekos to trigger the external port but ran into a problem that lead me back here. The fact that the k-50 errors if bulb is called causes indi and the camera to crash when I use an external trigger. I bypassed the call to bulb() in indi-gphoto-ccd and got it working for ekos and then realized that we could bypass it in gphoto and help out other apps without all that hacking I did on mine. Any chance we could prevent that error for k50d? printing a message to the effect of "use an external trigger for bulb on this model" might help too.  This should allow external bulb triggering to be used with the k50 and other stricken models in setups like indilib that have external trigger code.  It may be especially useful for Astrophotographers, or anyone really as it would prevent the camera lockup.

Dabi01, 2017-09-20:
Unfortunately it'll be months to collect the money for it, but I'm looking forward to it so much.
I can't really understand that coding part, but I will use it for astrophotography, so it sounds good:)

Blueshawk, 2017-09-25:
Important followup on my most recent experiment: I was able to get bulb triggering in the k-50 using Gphoto2 --wait-event-and-download and then pulsing an external trigger cord by toggling the usb port RTS line in a separate terminal. After some hacking to make sure it could complete the routine by arbitrarily clearing the RTS_flag I was also able to get it capturing and showing previews and captures in ekos...but alas. It started doing the dreaded mirror flop while I was doing a focus run which included short bulb shots. The only way out of that is to pull power, and it does not sound healthy when it happens. The next morning I recreated the flapping mess of a lockup while connected to the pc and gtkam. and actually pressing the shutter button on the unit. I got the failure in just a few tries. This may be the end of the line for the k-50 bulb, Pretty much tanking an otherwise stellar camera for AP use. unless somebody has a clue as to what may trigger this nasty error. The only other time I saw this error was when using pktriggercord's gui interface way back when I first started trying to tether. 
  It's worth noting that earlier models have been reported to work with the bulb widget just fine, and a workaround is being worked on for later ones like the k70.
Kudos once again to @msmeissn for his help in bringing Pentax into the fold.

Blueshawk, 2018-05-21:
@msmeissn  I wasn't sure which way to contact you but I had to ask, did you get the SDK Ricoh just released? I noticed it includes libmtp which you helped with(credited in the code), but as I'm always watching for developments on the k-50 front, or indeed anything to help the Pentax situation in general, I was hoping there might be some nuggets in there we hadn't already encountered.

https://console.api.ricoh/sdks/camera/

I've been keeping up via the email list. Hope you are well. Cheers :)

msmeissn, 2018-05-21:
I looked at it, the library is binary only so it needs reverse engineering again to get at the nuggets. :/
Why cant they just post the protocol infos. :/

[3 more replies]