gphoto/libgphoto2 #32: Support for Fuji x-t10

overtune, 2016-01-02:
I was just wondering if there are any plans for supporting image capturing with Fuji x-t10?
File download seems to be supported, but if image capturing could be supported it would be great.

C++ programming is a bit out of my scope, but if you need any info/testing/debug dumps from an x-t10 I'll be glad to help.

Thanks!

msmeissn, 2016-01-03:
captuyring debug information would be good

gphoto2 --debug --debug-logfile=summary.log --summary > summary.txt

attach summary.txt and summary.log here

overtune, 2016-01-04:
Here is a zip with summary.txt and summary.log (couldn't attach the .log file it wasn't supported, so I zipped them).
Let me know if there is anything more I could do.
Thank you!

[summary.zip](https://github.com/gphoto/libgphoto2/files/77470/summary.zip)

msmeissn, 2016-01-04:
Thanks!

While I have added the ID and information to libgphoto2, the camera set of abilities seem limited to downloading images only, no remote control functionality is apparent in the logfile.

msmeissn, 2016-01-04:
can not do much more

Firstyear, 2018-01-30:
[summary.log](https://github.com/gphoto/libgphoto2/files/1679607/summary.log)
[summary.txt](https://github.com/gphoto/libgphoto2/files/1679609/summary.txt)

It's been a while since this issue was last looked at, and I think there has been some firmware updates. Supplying the relevant information to see if we can get live view/capture working on this camera.

Thanks!

msmeissn, 2018-01-31:
I looked at the firmware update page of the X-T10, and it does not mention remote control. It seems you are on 1.21 , which seems to be the latest.

msmeissn, 2018-07-21:
this seems only be provided for newer cameras, like X-T1 and X-T2 .../ not sure if potentially newer firmware could help, please try ... othrewise we cannot do much.

closed 2018-07-21 by msmeissn