gphoto/libgphoto2 #33: Building libgphoto2 on Visual Studio 2015

skn123, 2016-01-03:
Before someone shuts me off, I am aware of this disclaimer on the main page:
Platforms
---------

libgphoto2 should compile and run on pretty much all Unix-like platforms.
libgphoto2 has _not_ (yet?) been ported to any operating system from MicroSoft.

Having said that, I embarked on porting libgphoto2 for Visual Studio. It is a wonderful tool and my primary motivation was to provide access to SLR camera's using a port of libgphoto2 for Micromanager. 

I have been successful to a very large extent. However, I am facing issues with an issue that I am not well versed with; loading dynamic libraries using ltdl !! and I think if that is solved then we can indeed get a port of libgphoto2 on windows. Here is a screenshot of my build and the exact problem that I am facing:

a.) I am trying to detect my Lenovo K3-Note camera (a model supported by libgphoto2 and verified on a linux machine)
b.) libusb is detecting the device
c.) ptp2.dll is being loaded by ltdl (as seen in the screenshots)
d.) The camera is not being initialized after lt_dlsym (screenshots attached).

The only dependencies that I had to build were:
a.) libusb (built as a static library using VS2015)
b.) libtools (for ltdl ; Used Msys/MinGW - gcc 5.2.1 and exported the symbols using dumpbin and created a VS2015 import library using lib)
c.) pthread (built as a static library using VS2015)

The iolibs are stored in <path_of_executable>/iolibs 
The camlibs are stored in <path_of_executable>/camlibs

I have provided global paths for iolibs and camlibs as environment variables in Windows.

My fork (along with changes that I had to make) can be found
https://github.com/skn123/libgphoto2

[image]
[image]
[image]

Please comment on any aspect of the code so that I can share with the community a CMake version of libgphoto2!

msmeissn, 2016-01-03:
- Why Cmake .... can autoconf based set ups not made to work?

- the massive amount of new ifdefs mostly due to config.h is not good, this needs to be abstracted and done a bit better.

Portability things , originally intended to be in libgphoto2_port/gphoto2/
- usleep . we can redefine it as static inline in libgphoto2_port//gphoto2/gphoto2-port-portability.h probably

I think fixed all the dirent problems you saw, so the dirent stuff is not needed.
unistd.h ... i removed all unnecessary occurences and protected the rest with ifdefs

skn123, 2016-01-04:
Marcus, when the venerable KDE takes up CMake I am but a mere mortal :). 
https://lwn.net/Articles/188693/

CMake can give us cross platform functionality along with a plethora of other goodies that autoconf cannot. Having said that, there is absolutely no need to use CMake if folks are comfortable with autoconf. CMake is what we use across the board (Windows, Mac and Linux (Ubuntu and OpenSuse) ) and if we have CMake support in libgphoto2, it would be another way to build it without disrupting anyone's existing build and reaching out to a whole new audience.

I will incorporate all the changes you have suggested in my fork. As I mention, I have tried to be as faithful to the original code with the config's. Hence, there are a lot of "-D<....>" in the CMakeLists.txt which I will change once
"..the massive amount of new ifdefs mostly due to config.h is not good, this needs to be abstracted and done a bit better."
has been done in gphoto2.

However, once those are taken care of, building the code was a breeze! My problem still persists though. I think it is a ltdl problem and I don't know if that can be fixed unless I statically link the libraries rather than load it via ltdl. 

Have you looked at the Micromanager way of loading libraries? They too have support for various devices (libgphoto2 being one of them) via shared libraries.

msmeissn, 2016-01-04:
to avoid all the ifdefs around include config.h you could just create an empty config.h file btw, which would reduce the number of affected c source files

skn123, 2016-01-04:
I was doing that but as I had mentioned, I did not want to mess with the existing build. If it can be refactored then so be it. Is there some way you can help me with this?

skn123, 2016-01-13:
Marcus, One update from my end (and I must say this solves a different problem of mine)
a.) I have been able to get ptp2.dll to load using ltdl !!! Apparently, a global define -DDLL_EXPORT had to be defined and that did the trick.
b.) In library.c of ptp2.dll, I could get to 
```
ret = ptp_ptpip_connect (params, xpath);
```
and then it failed again. I think you have set this in ptpip.c
```
#ifdef HAVE_INET_ATON
	addr = strdup (address);
	if (!addr)
		return GP_ERROR_NO_MEMORY;
	s = strchr (addr,':');
	if (!s) {
		GP_LOG_E ("addr %s should contain a :", address);
		free (addr);
		return GP_ERROR_BAD_PARAMETERS;
	}
	*s = '\0';
	p = strchr (s+1,':');
	port = 15740;
	if (p) {
		*p = '\0';
		if (!sscanf (p+1,"%d",&port)) {
			fprintf(stderr,"failed to scan for port in %s\n", p+1);
			free (addr);
			return GP_ERROR_BAD_PARAMETERS;
		}
	}
	if (!inet_aton (s+1,  &saddr.sin_addr)) {
		fprintf(stderr,"failed to scan for addr in %s\n", s+1);
		free (addr);
		return GP_ERROR_BAD_PARAMETERS;
	}
	saddr.sin_port		= htons(port);
	saddr.sin_family	= AF_INET;
	free (addr);
	params->cmdfd = socket (PF_INET, SOCK_STREAM, 0);
	if (params->cmdfd == -1) {
		perror ("socket cmd");
		return GP_ERROR_BAD_PARAMETERS;
	}
	params->evtfd = socket (PF_INET, SOCK_STREAM, 0);
	if (params->evtfd == -1) {
		perror ("socket evt");
		close (params->cmdfd);
		return GP_ERROR_BAD_PARAMETERS;
	}
	if (-1 == connect (params->cmdfd, (struct sockaddr*)&saddr, sizeof(struct sockaddr_in))) {
		perror ("connect cmd");
		close (params->cmdfd);
		close (params->evtfd);
		return GP_ERROR_IO;
	}
	ret = ptp_ptpip_init_command_request (params);
	if (ret != PTP_RC_OK) {
		close (params->cmdfd);
		close (params->evtfd);
		return translate_ptp_result (ret);
	}
	ret = ptp_ptpip_init_command_ack (params);
	if (ret != PTP_RC_OK) {
		close (params->cmdfd);
		close (params->evtfd);
		return translate_ptp_result (ret);
	}
	/* seen on Ricoh Theta, camera is not immediately ready. try again two times. */
	tries = 2;
	do {
		if (-1 != connect (params->evtfd, (struct sockaddr*)&saddr, sizeof(struct sockaddr_in)))
			break;
		if ((errno == ECONNREFUSED) && (tries--)) {
			GP_LOG_D ("event connect failed, retrying after short wait");
			usleep(100*1000);
			continue;
		}
		GP_LOG_E ("could not connect event");
		close (params->cmdfd);
		close (params->evtfd);
		return GP_ERROR_IO;
	} while (1);
	ret = ptp_ptpip_init_event_request (params);
	if (ret != PTP_RC_OK)
		return translate_ptp_result (ret);
	ret = ptp_ptpip_init_event_ack (params);
	if (ret != PTP_RC_OK)
		return translate_ptp_result (ret);
	GP_LOG_D ("ptpip connected!");
	return GP_OK;
#else
	GP_LOG_E ("Windows currently not supported, neeeds a winsock port.");
	return GP_ERROR_NOT_SUPPORTED;
#endif
```
What is this piece of code doing? I could get a Windows equivalent of the "inet_aton" function
http://stackoverflow.com/questions/2420663/windows-equivalent-of-inet-aton
Is this what it is supposed to be doing?

zxgyy, 2016-11-16:
Hi, SKN123. 
  Have you fixed the problem of 'gp_camera_init return -6' , i follow you, and i can not solve this problem， it's very difficult. I could not get useful information from Google. my camera is Sony-A7R2, so what can i do, 
Would you like to give me some idea or suggestion,Thanks.

zxgyy, 2016-11-16:
When i change the camera from Sony-A7R2 to Nikon D90, it's still return the same result, -6.

doczoidberg, 2017-01-15:
Is it possible with this builld to run gphoto in an WPF application? If so it is also possible to show a livestream inside XAML when the camera supports this?

skn123, 2017-01-16:
I could not fix the issue. I could build the code as is with modifications. However, there are some commands that are very Linux specific. It tries to load a module and then when it cannot run those Linux commands it exits.