gphoto/libgphoto2 #34: copying images from a camera gives -1: 'Unspecified error', second attempt succeeds

kparal, 2016-01-06:
When I try to copy pictures from my camera, I get `error getting file: -1: unspecified error` in Nautilus for every single picture. The pictures *are* copied over, but it shows this error for every single one of them. Using gphoto2, it looks like this:
```
$ gphoto2 --debug --debug-logfile=gphoto2-debug.log --get-file 2 
                                                                               
*** Error ***              
PTP Invalid Object Handle
*** Error (-1: 'Unspecified error') ***       
```
Again, second attempt for the same image succeeds every time (even after disconnecting and reconnecting the device, that image can be copied without issues).

I have reported the issues initially against gvfs, please see the full bug report here:
https://bugzilla.gnome.org/show_bug.cgi?id=760162

There is gphoto2 debug log attached here:
https://bugzilla.gnome.org/attachment.cgi?id=318316

Copying works correctly with libgphoto2-2.5.7-4.fc23.x86_64, but is broken with libgphoto2-2.5.8-1.fc23.x86_64 (Fedora 23).

Thanks for looking into this.

msmeissn, 2016-01-09:
the setobjectarchive opcode failed (even though it should not). 

i fixed this only half way in 2.5.9, now it is fully fixed for 2.5.10

closed 2016-01-09 by msmeissn

kparal, 2016-01-09:
Awesome, thank you. If you could publish a 2.5.10 release soon, I'd then ask my distribution to package it and send us an update. Thanks.

kparal, 2016-01-18:
Hello, any chance to cut a new release soon? Thanks a lot.

msmeissn, 2016-01-18:
Given t he amount of testing I want to do before each release I cannot say how long this will take.

msmeissn, 2016-01-18:
(but I have planned to do a release soon.)

kparal, 2016-01-18:
:+1: ;) No rush, just asking, this bug does not break functionality too much, it's just annoying.