gphoto/libgphoto2 #36: SEGV when trying to access Pentax K100D

joysn71, 2016-01-24:
Hello,

i wanted to try `entangle` which is based on gphoto2 with my Pentax K100D. 

I use entangle and gphoto2 on Linux Mint Debiam Edition2, which is based on Debian Jessy:
```
chris@lm-joe ~ $ dpkg -l | grep gphoto
ii  gphoto2                                      2.5.4-1+b1                           amd64        digital camera command-line client
ii  libgphoto2-6:amd64                           2.5.4-1.1+b2                         amd64        gphoto2 digital camera library
ii  libgphoto2-port10:amd64                      2.5.4-1.1+b2                         amd64        gphoto2 digital camera port library
chris@lm-joe ~ $ uname -a
Linux lm-joe 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt7-1 (2015-03-01) x86_64 GNU/Linux
chris@lm-joe ~ $
```
The camera is detected by gphoto2:
```
chris@lm-joe ~ $ gphoto2 --auto-detect
Modell                         Port                                            
----------------------------------------------------------
Pentax K100D                   usbscsi:/dev/sg2
```
But when trying to access it gphoto crashes:
```
chris@lm-joe ~ $ gphoto2 --config
pentax/pslr.c:265:ipslr_status(p, statusbuf) failed: 4                         
Segmentation fault   
chris@lm-joe ~ $
```
Same, but called with strace to get some detail infos about the crash:
```
chris@lm-joe ~ $ strace gphoto2 --config 2> strace-out-crash.txt
Segmentation fault                                                             
chris@lm-joe ~ $
```
I also started gphoto2 as recommended when it crashes:
```
chris@lm-joe ~ $ env LANG=C gphoto2 --debug --debug-logfile=gphoto-debug.txt --config
pentax/pslr.c:265:ipslr_status(p, statusbuf) failed: 4                         
Segmentation fault
chris@lm-joe ~ $
```
See the following full logs for details:
[strace-out-crash.txt](https://github.com/gphoto/libgphoto2/files/102273/strace-out-crash.txt)
[gphoto-debug.txt](https://github.com/gphoto/libgphoto2/files/102274/gphoto-debug.txt)

msmeissn, 2016-01-24:
can youy get a backtrace with gdb? 

gdb gphoto2
r --config
... will crash now ... 
bt

joysn71, 2016-01-24:
Hello!

as requested the gdb bt output:

```
chris@lm-joe ~ $ gdb gphoto2             
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gphoto2...(no debugging symbols found)...done.
(gdb) r --config
Starting program: /usr/bin/gphoto2 --config
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff4398700 (LWP 9332)]
[Thread 0x7ffff4398700 (LWP 9332) exited]
[New Thread 0x7ffff4398700 (LWP 9333)]                                                                                                                                                                                                        
[Thread 0x7ffff4398700 (LWP 9333) exited]
pentax/pslr.c:265:ipslr_status(p, statusbuf) failed: 4

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff47c3aa4 in ?? () from /usr/lib/x86_64-linux-gnu/libgphoto2/2.5.4/pentax.so
(gdb) bt
#0  0x00007ffff47c3aa4 in ?? () from /usr/lib/x86_64-linux-gnu/libgphoto2/2.5.4/pentax.so
#1  0x00007ffff7b64473 in gp_camera_get_config () from /usr/lib/x86_64-linux-gnu/libgphoto2.so.6
#2  0x000000000040a146 in ?? ()
#3  0x0000000000413fc6 in ?? ()
#4  0x00007ffff67b89db in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#5  0x00007ffff67b8a14 in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#6  0x00007ffff67ba6f0 in poptGetNextOpt () from /lib/x86_64-linux-gnu/libpopt.so.0
#7  0x0000000000407b58 in ?? ()
#8  0x00007ffff612bb45 in __libc_start_main () from /lib/x86_64-linux-gnu/libc.so.6
#9  0x00000000004088f9 in ?? ()
(gdb)
```

msmeissn, 2016-01-24:
hmm. do you have the ability to also install debug symbols ... :/

joysn71, 2016-01-25:
sorry for the delay, never recompiled a package for myself on my box to get the debug symbols in.
Furthermore i noticed during reproduction of the issue, that gphoto2 not always crashes with a SEGV. Sometimes i may just hang. I attached the strace output of that scenario as well, maybe it is useful.

```
chris@lm-joe ~ $ env LANG=C gdb gphoto2
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gphoto2...done.
(gdb) r --config
Starting program: /usr/bin/gphoto2 --config
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff438f700 (LWP 32200)]
[Thread 0x7ffff438f700 (LWP 32200) exited]
[New Thread 0x7ffff438f700 (LWP 32201)]                                                                                                                                                                                                       
[Thread 0x7ffff438f700 (LWP 32201) exited]
pentax/pslr.c:265:ipslr_status(p, statusbuf) failed: 4

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff47bc34b in camera_get_config (camera=0x629080, window=0x7fffffffb418, context=0x62ade0) at pentax/library.c:370
370	pentax/library.c: No such file or directory.
(gdb) bt
#0  0x00007ffff47bc34b in camera_get_config (camera=0x629080, window=0x7fffffffb418, context=0x62ade0) at pentax/library.c:370
#1  0x00007ffff7b5f11e in gp_camera_get_config (camera=0x629080, window=0x7fffffffb418, context=0x62ade0) at gphoto2-camera.c:900
#2  0x0000000000407b0d in gp_cmd_config (camera=0x629080, context=0x62ade0) at gphoto2-cmd-config.c:555
#3  0x0000000000413c16 in cb_arg_run (ctx=0x62ae70, reason=POPT_CALLBACK_REASON_OPTION, opt=0x7fffffffbd60, arg=0x0, params=0x7fffffffb7b0) at main.c:1532
#4  0x0000000000414636 in cb_arg (ctx=0x62ae70, reason=POPT_CALLBACK_REASON_OPTION, opt=0x7fffffffbd60, arg=0x0, data=0x7fffffffb7b0) at main.c:1786
#5  0x00007ffff67b09db in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#6  0x00007ffff67b0a14 in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#7  0x00007ffff67b26f0 in poptGetNextOpt () from /lib/x86_64-linux-gnu/libpopt.so.0
#8  0x0000000000418107 in main (argc=2, argv=0x7fffffffe2a8, envp=0x7fffffffe2c0) at main.c:2429
(gdb)
```
[strace-out-hung.txt](https://github.com/gphoto/libgphoto2/files/103659/strace-out-hung.txt)

msmeissn, 2016-01-25:
Ah ... I think this is a bug that was fixed in libgphoto2 2.5.7 or 2.5.8.

can you compile the 2.5.9 libgphoto2 and see if that helps?

joysn71, 2016-01-25:
i got the 2.5.8 source via the Debian Jessie backports repository. After recompiling and replacing everything it looked like this:
```
chris@lm-joe ~ $ dpkg -l | grep gphoto
ii  gphoto2                                      2.5.8-1~bpo8+1                       amd64        digital camera command-line client
ii  libgphoto2-6:amd64                           2.5.8-1~bpo8+1                       amd64        gphoto2 digital camera library
ii  libgphoto2-dev:amd64                         2.5.8-1~bpo8+1                       amd64        gphoto2 digital camera library (development files)
ii  libgphoto2-port12:amd64                      2.5.8-1~bpo8+1                       amd64        gphoto2 digital camera port library
chris@lm-joe ~ $

chris@lm-joe ~ $ gphoto2 --version
gphoto2 2.5.8

Copyright (C) 2000-2015 Lutz Müller und andere

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

Diese Version von gphoto2 benutzt die folgenden Softwareversionen und Optionen:
gphoto2         2.5.8          gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.8          all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, no USB, serial without locking
chris@lm-joe ~ $
```
Then `gphoto --config` runs till the point where it opened a GUI and i was sent to "Camera and Driver Configuration". There it offered me two options: 1. Beenden and 2. Camera Settings. When i choose Camera Settings, it offers me 11 different menu entries. When i open 2. Model it opens a dialog asking me for a Model. When i enter something like Pentax K100D and press enter gphoto2 exits with an SEGV. I attach the output of gdb to this comment. I got it when i was attached to the running gphoto2 with gdb --pid.

[gphoto-2_5_8-debug.txt](https://github.com/gphoto/libgphoto2/files/103976/gphoto-2_5_8-debug.txt)

msmeissn, 2016-01-25:
canm you try the patch referenced in above commit. i think it crashes due to some uininitialized values, and that patch checks those.

joysn71, 2016-01-27:
i just copied the revision of library.c from 5942ea9 to my source tree and rebuild the libs. After this i can enter a value for "Model", and when i go to the next field "Image format" a small dialog opens asking for the "Image format", but then again i get a SEGV:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007fc9c14838ff in drawCDKItemlistField () from /usr/lib/libcdk.so.5
(gdb) bt
#0  0x00007fc9c14838ff in drawCDKItemlistField () from /usr/lib/libcdk.so.5
#1  0x00007fc9c14839ec in activateCDKItemlist () from /usr/lib/libcdk.so.5
#2  0x00007fc9c1d67b11 in show_radio (cmd_config=0x7ffd1d9a6ec0, 
    radio=0x7fc9c36c20d0) at gphoto2-cmd-config.c:320
#3  0x00007fc9c1d687ae in show_widget (cmd_config=0x7ffd1d9a6ec0, 
    widget=0x7fc9c36c20d0) at gphoto2-cmd-config.c:519
#4  0x00007fc9c1d672b3 in show_section (cmd_config=0x7ffd1d9a6ec0, 
    section=0x7fc9c36c13f0) at gphoto2-cmd-config.c:141
#5  0x00007fc9c1d686f0 in show_widget (cmd_config=0x7ffd1d9a6ec0, 
    widget=0x7fc9c36c13f0) at gphoto2-cmd-config.c:509
#6  0x00007fc9c1d688d8 in show_widget (cmd_config=0x7ffd1d9a6ec0, 
    widget=0x7fc9c36c1a60) at gphoto2-cmd-config.c:531
#7  0x00007fc9c1d672b3 in show_section (cmd_config=0x7ffd1d9a6ec0, 
    section=0x7fc9c36c13f0) at gphoto2-cmd-config.c:141
#8  0x00007fc9c1d686f0 in show_widget (cmd_config=0x7ffd1d9a6ec0, 
    widget=0x7fc9c36c13f0) at gphoto2-cmd-config.c:509
#9  0x00007fc9c1d672b3 in show_section (cmd_config=0x7ffd1d9a6ec0, 
    section=0x7fc9c36b0b30) at gphoto2-cmd-config.c:141
#10 0x00007fc9c1d686f0 in show_widget (cmd_config=0x7ffd1d9a6ec0, 
    widget=0x7fc9c36b0b30) at gphoto2-cmd-config.c:509
#11 0x00007fc9c1d68a25 in gp_cmd_config (camera=0x7fc9c369d050, 
    context=0x7fc9c369edb0) at gphoto2-cmd-config.c:573
#12 0x00007fc9c1d75a30 in cb_arg_run (ctx=0x7fc9c369ee40, 
---Type <return> to continue, or q <return> to quit---
    reason=POPT_CALLBACK_REASON_OPTION, opt=0x7ffd1d9a7820, arg=0x0, 
    params=0x7ffd1d9a7240) at main.c:1583
#13 0x00007fc9c1d76596 in cb_arg (ctx=0x7fc9c369ee40, 
    reason=POPT_CALLBACK_REASON_OPTION, opt=0x7ffd1d9a7820, arg=0x0, 
    data=0x7ffd1d9a7240) at main.c:1837
#14 0x00007fc9c050b9db in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#15 0x00007fc9c050ba14 in ?? () from /lib/x86_64-linux-gnu/libpopt.so.0
#16 0x00007fc9c050d6f0 in poptGetNextOpt ()
   from /lib/x86_64-linux-gnu/libpopt.so.0
#17 0x00007fc9c1d7a60c in main (argc=2, argv=0x7ffd1d9a9dc8, 
    envp=0x7ffd1d9a9de0) at main.c:2488
(gdb)
```

joysn71, 2016-01-27:
Another question: how do i know what values i can or should enter into those fields at --config in gphoto2?

```
hris@lm-joe ~ $ gphoto2 --auto-detect
Modell                         Port                                            
----------------------------------------------------------
Pentax K100D                   usbscsi:/dev/sg2
chris@lm-joe ~ $ gphoto2 --summary
pentax/pslr.c:272:ipslr_status(p, statusbuf) failed: 4                         
Kamerazusammenfassung:
Pentax K DSLR capture driver.
Based on pkremote by Pontus Lidman.

pentax/pslr.c:296:ipslr_status(p, statusbuf) failed: 4
chris@lm-joe ~ $
```

fabge, 2019-10-21:
Has any progress been made using the K100D?
When I connect it to my Raspberry Pi and use `--auto-detect` it recognizes the camera and SD-card:

```
Model                          Port                                            
----------------------------------------------------------
Mass Storage Camera            disk:/media/pi/NIKON D7200
Pentax K100D                   usbscsi:/dev/sg0
```

But when trying to capture an images it gives me following output:

```
pi@raspberrypi:~ $ gphoto2 --camera "Pentax K100D" --trigger-capture
    
*** Error ***              
This camera can not trigger capture.
ERROR: Could not trigger capture.
*** Error (-6: 'Unsupported operation') ***       
```

The debug file looks as follows:
[debug.txt](https://github.com/gphoto/libgphoto2/files/3751906/debug.txt)

msmeissn, 2019-10-22:
it used the mass storage camera.. 

Try
gphoto2 --camera "Pentax K100D" --port usbscsi:/dev/sg0 --trigger-capture

fabge, 2019-10-22:
Thank you for your answer! I tried it with your specified parameters and it seems to work, but not quite...

When I enter
`gphoto2 --camera "Pentax K100D" --port usbscsi:/dev/sg0 --trigger-capture --debug --debug-logfile="debug.txt"`
the command executes without warning.

If it is totally silent I hear a barely noticeable "click" in the camera, it doesn't fire though.
Is there something wrong with the camera or do I have to somehow force-take a picture? The camera works in manual mode.

msmeissn, 2019-10-27:
can you attach the debug.txt file?

It will be a bit hard to debug, sorry

Mirppc, 2020-06-27:
Where would be the best place to upload a debug text? K100DS is the one i am testing.

msmeissn, 2020-06-28:
you can attach one to this issue here.

Do you have the same issue?

Mirppc, 2020-06-28:
Correct, after ruling out PEBKAC, and then getting a new USB cable i was back to square one like i was in Opensuse 13.2.  Note the log is from PopOS 20.04.

[debug.txt](https://github.com/gphoto/libgphoto2/files/4843161/debug.txt)

gphoto2 -v
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.24         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking