gphoto/libgphoto2 #37: Nikon D810 setting the  "recordingmedia" config value

lrxgab005, 2016-01-28:
Hello,

we've been using libgphoto2 over the last 2 years with great success, however we have recently wanted to change the "recordingmedia" config value on our Nikon D810's and have been unable to do so. This works fine on our Nikon D800E's, which leads us believe that this problem is specific to the D810's.

Heres how we went about trying to set the config value with libgphoto2 2.5.9.

First identify the camera port and config value.
<pre lang="bash"><code>
OpenMVS-Shell:/local/bzffacel/tmp$ gphoto2 --auto-detect
Modell                         Port                                            
----------------------------------------------------------
Nikon DSC D810                 usb:002,035     
Nikon DSC D800E                usb:003,103     

OpenMVS-Shell:/local/bzffacel/tmp$ gphoto2 --port usb:002,035 --get-config=recordingmedia
Label: Aufnahmemedium                                                          
Type: RADIO
Current: Speicherkarte
Choice: 0 Speicherkarte
Choice: 1 Kamera SDRAM
Choice: 2 Unbekannter Wert 0002
</code></pre>

Then try and set the config and see if it has changed.

<pre lang="bash"><code>
OpenMVS-Shell:/local/bzffacel/tmp$ gphoto2 --port usb:003,100 --set-config-value=recordingmedia=Kamera\ SDRAM
OpenMVS-Shell:/local/bzffacel/tmp$ gphoto2 --port usb:002,035 --get-config=recordingmedia
Label: Aufnahmemedium                                                          
Type: RADIO
Current: Speicherkarte
Choice: 0 Speicherkarte
Choice: 1 Kamera SDRAM
Choice: 2 Unbekannter Wert 0002
</code></pre>

This same procedure works as expected with the D800E and changing other config values on the D810's hasn't been a problem thus far.

msmeissn, 2016-02-05:
what if you set it to Unbekannter Wert 0002 ?  occasionaly nikon shuffles values areound

msmeissn, 2016-02-05:
(and the naming might not match what is printed, as we decode the enumeration from integer to readable names)

lrxgab005, 2016-02-11:
Thanks for the response! This unfortunately didnt work for us. I tried setting the value to "Unbekannter Wert 0002" and also treid the --set-config-index function instead. Both didnt work...

msmeissn, 2016-02-18:
the value is not changing ... hmm.

can you get it into controlmode perhaps, 
gphoto2 --set-config controlmode=1  -set-config-value=recordingmedia=Kamera\ SDRAM

but I am not sure it will help.

can you perhaps attach gphoto2 --summary   output too

lrxgab005, 2016-02-19:
Alright. I tried again using the controlmode flag. It still didnt change the value according to the --get-config output. Heres the output from the summary. It does seem like its setting it , but not holding the value

gphoto2 --port usb:002,058 --set-config controlmode=1 --set-config-value=recordingmedia=SDRAM --summary | grep Record

Recording Media(0xd10b):(readwrite) (type=0x2) Range [0 - 2, step 1] value: SDRam (1)