gphoto/libgphoto2 #38: gtkam compile problem when HAVE_GP_PORT_INFO_GET_NAME not set

cdputnam, 2016-01-31:
I've been having a compile-time failure for gtkam-master/src/gtkam-chooser.c when HAVE_GP_PORT_INFO_GET_NAME is not set.

The problem stems from the fact that GPPortInfo is type-defined to a pointer to an incomplete type (e.g. "struct _GPPortInfo; typedef struct _GPPortInfo *GPPortInfo;"), but the functions on_port_added(), gtkam_chooser_set_camera(), and gtkam_chooser_set_port_mask() are treating GPPortInfo as a concrete type, e.g.:

name = g_strdup_printf ("%s (%s)", info.name, info.path);

instead of

name = g_strdup_printf ("%s (%s)", info->name, info->path);

I'm attaching a very hacky patch that fixes this set of bugs for me (though it's possible I need to figure out how to set HAVE_GP_PORT_INFO_GET_NAME in my config). The biggest problem in this patch is getting GPPortInfo type-defined to a concrete type, where I've directly included a local file.

[gtkam-chooser.patch.txt](https://github.com/gphoto/libgphoto2/files/111382/gtkam-chooser.patch.txt)

msmeissn, 2016-02-01:
HAVE_GP_PORT_INFO_GET_NAME is set by the configure script, after checking for presence.

it will be set for libgphoto2 2.5 and later,
it will not be set for libgphoto2 2.4 and older.

are you building gtkam against libgphoto2 2.4 ?

cdputnam, 2016-02-01:
Odd. No. I'm building against 2.5.7:

$rpm -q libgphoto2-devel
libgphoto2-devel-2.5.7-2.fc21.x86_64

and here is the relevant snippet from config.h

...
/* whether we use gnome */
/* #undef HAVE_GNOME */

/* Define to 1 if gp_port_info_get_name is present */
/* #undef HAVE_GP_PORT_INFO_GET_NAME */

/* Define if you have the iconv() function. */
#define HAVE_ICONV 1
...

But if I manually add the "#define HAVE_GP_PORT_INFO_GET_NAME 1", this also solves that build problem.

msmeissn, 2016-02-01:
did you ran configure ? can you attach config.log?

cdputnam, 2016-02-01:
I just realized that I unintentionally created a hostile build environment that circumvented some of the checks that configure was running. This bug was entirely my fault, so sorry for the noise.

In short, I ran configure without having the appropriate Fedora -devel packages present (and I was setting environment variables to the libraries). I'm guessing the version number that the configure script got was "0", which is less than 2.5 and that explains everything. (And when I started hitting compile problems, I installed the -devel packages, but the damage was already done.)

After running configure in a fresh shell, now everything compiles nicely.

Again sorry for the noise, and thanks for your help.

closed 2016-02-01 by cdputnam

msmeissn, 2016-02-01:
ok!