gphoto/libgphoto2 #40: Capturing not possible for Nikon P330

torstenhaase, 2016-02-05:
Hello, I bought a Nikon P330 do create time-lapse movies. It is listed in the list of supported cameras:
 > Nikon Coolpix P330 (PTP mode)	Image Capture, Configuration

Unfortunately it doesn't work :(
Please find attached the logs:

> pi@ZeitrafferBox`
~ $ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Nikon Corporation
Model: P330
  Version: V1.00
  Serial Number: 40003182
Vendor Extension ID: 0xa (1.0)
Vendor Extension Description: microsoft.com: 1.0; Nikon PTP Extensions

>Capture Formats: 
Display Formats: JPEG, Association/Directory, DPOF, MS Wave, Apple Quicktime, Unknown(b111), Undefined Type

>Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

>Storage Devices Summary:
>store_00010001:
	StorageDescription: None
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 3956801536 (3773 MB)
	Free Space (Bytes): 3721658368 (3549 MB)
	Free Space (Images): 1051

>Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 100, step 1] value: 100% (100)
Date & Time(0x5011):(readwrite) (type=0xffff) '20160205T174159'
Property 0xd280:(read only) (type=0x2) 3
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1

>pi@ZeitrafferBox ~ $ gphoto2 --capture-image
                                                                               
>*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***    

I am using the following gphoto2 version:
>pi@ZeitrafferBox ~ $ gphoto2 --version
gphoto2 2.5.6

>Copyright (c) 2000-2014 Lutz Mueller and others

>gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

>This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.6          gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.7          all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking`

Thanks for your help
Torsten

msmeissn, 2016-02-05:
can you also try attaching the debug logfile of 

gphoto2 --summary --debug --debug-logfile=logfile.txt 

?

torstenhaase, 2016-02-05:
Hi Marcus, please find attached the log file. Thanks for your help.

[logfile.txt](https://github.com/gphoto/libgphoto2/files/119670/logfile.txt)

msmeissn, 2016-02-05:
it has the capture complete event, but it has no opcodes listed that would do cpature... it might mean that they are not present or hidden :/

torstenhaase, 2016-02-06:
That means it won't work? Or is there any chance that it will work with a new version of gphoto2?
I checked the Nikon webpage, there is a firmware upgrade for the P330 but they don't write anything about changes in the PTP protocol: https://nikoneurope-de.custhelp.com/app/answers/detail/a_id/60695/~/coolpix-p330-firmware-update-1.1

msmeissn, 2016-02-06:
they might be hiding this little details. or not. Sadly, I have not much knowledge of their firmware updates.

I have the P330 down as:
       /* Jonas Stein <news@jonasstein.de> */
      {"Nikon:Coolpix P330 (PTP mode)", 0x04b0, 0x0227, PTP_CAP},

so the reporter seems to have been luckier

torstenhaase, 2016-02-07:
Hi, I contacted Jonas. He said that he was only able to download pictures. Remote control of the camera wasn't possible. So it look like it doesn't work :( I'll try with the new firmware and the newest version of gphoto2 but I don't think it will make any difference.

msmeissn, 2016-02-07:
I looked over my notes and it is probably possible over WLAN via ptpip ... . but thats harder to setup and use than just the usb link

torstenhaase, 2016-02-07:
As I though there was no change with gphoto2 v2.5.9 and firmware 1.1 I think you can close the thread.

msmeissn, 2016-04-23:
ok, thanks for the follow up!

closed 2016-04-23 by msmeissn