gphoto/libgphoto2 #42: Alternative to gp_camera_get_config to read only specific ptp properties

rlamarche, 2016-02-14:
Hi,

I work on an opensource project to allow remote control of dslr cameras through ptp.
I decided to use gphoto because I hate reinventing the wheel and gphoto2 seems to be perfect for my needs.

I would like to be able to read in real time some ptp properties like aperture, bargraph, iso etc, while showing the liveview frames (real time also).
My problem is that the only method I've found that allow to read the properties is `gp_camera_get_config`, and it reads **all** properties.

So if I call this method every 500ms (for example), I miss a lot of liveview frames.

The goal of my project is to use the remote liveview of the camera on an external screen inside an underwater housing, and so we don't want to miss liveview frames because it might be uncomfortable for film (and photography).

I already seen in the code base of libgphoto which kind of method I would have to add, but before doing this I would like to know if this feature already exists ? Or what would be the best way to implement it ?

Thank you.

Regards,

Romain.

msmeissn, 2016-02-15:
it currently does not exist. but it is definitely useful, so I will see if I can add such a thing

rlamarche, 2016-02-15:
I'm very happy that you find my idea useful.
If I can help don't hesitate. But to be honest I don't have a lot of knowledge about USB or the PTP protocol. That's why I'm using gphoto ;)

msmeissn, 2016-05-01:
is now in 2.5.10 :)

closed 2016-05-01 by msmeissn

rlamarche, 2016-05-02:
Thank you :+1: