gphoto/libgphoto2 #43: Error setting property "movie" on Nikon D5200

gauravraj, 2016-02-17:
I'm using Nikon D5200 to capture video. It provides toggle property "movie" to start/stop capturing video. I'm able to capture video using gphoto2 command line tool to capture video. But when I'm trying to toggle this property using libgphoto2 from my code, it always returns me error code -1. I'm using libgphoto2 2.5.9 and gphoto2 2.5.9. I tried other version too but same result. I've used gphoto's set_config_value_action() as my reference code to set the camera parameters. I'm able to set all other parameters like shutterspeed, iso, aperture, whitebalance except this particular.

msmeissn, 2016-02-17:
can you show the part of the code? 

the toggle is an integer to set

gauravraj, 2016-02-18:
**This is how I call the set function - **
```
if(cameraControllerApi->setConfigValue("movie", "1") != GP_OK) {
     // Error setting value.
}
```

**This is my set parameter function -**

```
int CameraControllerApi::setConfigValue(const char *name, const char *value) {
    CameraWidget *rootconfig,*child;
    int	ret;
    CameraWidgetType	type;
    ret = findWidgetByName (name, &child, &rootconfig);
    if (ret != GP_OK)
        return ret;

    ret = gp_widget_get_type (child, &type);
    if (ret != GP_OK) {
        gp_widget_free (rootconfig);
        return ret;
    }

    switch (type) {
    case GP_WIDGET_TEXT: {		/* char *		*/
        ret = gp_widget_set_value (child, value);
        if (ret != GP_OK)
            LOG4CXX_ERROR(Config::logger, "Failed to set the value of text widget "<<name<<" to "<<value);           
        break;
    }
    case GP_WIDGET_RANGE: {	/* float		*/
        float	f,t,b,s;

        ret = gp_widget_get_range (child, &b, &t, &s);
        if (ret != GP_OK)
            break;
        if (!sscanf (value, "%f", &f)) {           
            LOG4CXX_ERROR(Config::logger, "The passed value "<<value<<" is not a floating point value.");
            ret = GP_ERROR_BAD_PARAMETERS;
            break;
        }
        if ((f < b) || (f > t)) {
             LOG4CXX_ERROR(Config::logger, "The passed value "<<f<<" is not within the expected range "<<b<<" - "<<t);
            ret = GP_ERROR_BAD_PARAMETERS;
            break;
        }
        ret = gp_widget_set_value (child, &f);
        if (ret != GP_OK)
            LOG4CXX_ERROR(Config::logger, "Failed to set the value of range widget "<<name<<" to "<<f);            
        break;
    }
    case GP_WIDGET_TOGGLE: {	/* int		*/
        int	t;

        t = 2;
        if (	!strcasecmp (value, "off")	|| !strcasecmp (value, "no")	||
                !strcasecmp (value, "false")	|| !strcmp (value, "0"))
            t = 0;
        if (	!strcasecmp (value, "on")	|| !strcasecmp (value, "yes")	||
                !strcasecmp (value, "true")	|| !strcmp (value, "1"))
            t = 1;
        /*fprintf (stderr," value %s, t %d\n", value, t);*/
        if (t == 2) {            
            LOG4CXX_ERROR(Config::logger, "The passed value "<<value<<" is not a valid toggle value.");
            ret = GP_ERROR_BAD_PARAMETERS;
            break;
        }
        ret = gp_widget_set_value (child, &t);
        if (ret != GP_OK)
            LOG4CXX_ERROR(Config::logger, "Failed to set values "<<value<<"of toggle widget "<<name);            
        break;
    }
    case GP_WIDGET_DATE:  {		/* int			*/
        int	t = -1;
        struct tm xtm;

#ifdef HAVE_STRPTIME
        if (strptime (value, "%c", &xtm) || strptime (value, "%Ec", &xtm))
            t = mktime (&xtm);
#endif
        if (t == -1) {
            if (!sscanf (value, "%d", &t)) {
                LOG4CXX_ERROR(Config::logger, "The passed value "<<value<<" is neither a valid time nor an integer.");
                                ret = GP_ERROR_BAD_PARAMETERS;
                break;
            }
        }
        ret = gp_widget_set_value (child, &t);
        if (ret != GP_OK)
            LOG4CXX_ERROR(Config::logger, "Failed to set new time of date/time wiget "<<name<<" to "<<value);            
        break;
    }
    case GP_WIDGET_MENU:
    case GP_WIDGET_RADIO: { /* char *		*/
        int cnt, i;

        cnt = gp_widget_count_choices (child);
        if (cnt < GP_OK) {
            ret = cnt;
            break;
        }
        ret = GP_ERROR_BAD_PARAMETERS;
        for ( i=0; i<cnt; i++) {
            const char *choice;

            ret = gp_widget_get_choice (child, i, &choice);
            if (ret != GP_OK)
                continue;
            if (!strcmp (choice, value)) {
                ret = gp_widget_set_value (child, value);
                break;
            }
        }
        if (i != cnt)
            break;
        /* Lets just try setting the value directly, in case we have flexible setters,
             * like PTP shutterspeed. */
        ret = gp_widget_set_value (child, value);
        if (ret == GP_OK) break;
        LOG4CXX_ERROR(Config::logger, "Choice "<<value<<" not found within list of choices.");       
        break;
    }

        /* ignore: */
    case GP_WIDGET_WINDOW:
    case GP_WIDGET_SECTION:
    case GP_WIDGET_BUTTON:        
        LOG4CXX_ERROR(Config::logger, "The "<<name<<" widget is not configurable.");
        ret = GP_ERROR_BAD_PARAMETERS;
        break;
    }
    if (ret == GP_OK) {
        ret = gp_camera_set_config (camera, rootconfig, ctx);
        if (ret != GP_OK)
            LOG4CXX_ERROR(Config::logger, "Failed to set new configuration value "<<value<<" for configuration entry "<<name);           
    }
    gp_widget_free (rootconfig);
    return (ret);
}_
```

msmeissn, 2016-02-18:
which part does get -1, the gp_camera_set_config? or the gp_widget_set_value ? 

does it really get -1 or a different return value ret ?

msmeissn, 2016-02-18:
what is the LOG4CXX error line? the code looks generally OK as-is.

gauravraj, 2016-02-18:
I'm using log4cxx logging library. LOG4CXX_ERROR() is used to log errors. It is giving following error - 
Failed to set new configuration value 1 for configuration entry movie. 
gp_camera_set_config() returns -1 everytime. Sometimes it opens the shutter of the camera and then return -1.

gauravraj, 2016-02-18:
I have integrated gphoto2 to my Qt Application. Can this be a compiler issue?

msmeissn, 2016-02-18:
can you debug what value of t it is trying to set?

gauravraj, 2016-02-18:
Yes, I tried that. It enters the following if statement - 
```
if (    !strcasecmp (value, "on")   || !strcasecmp (value, "yes")   ||
                !strcasecmp (value, "true") || !strcmp (value, "1"))
            t = 1;
```

rlamarche, 2016-02-18:
Hi, I'm using libgphoto2 inside a qt application without problems. With a Nikon D800, I've succeed in capture movie.

rlamarche, 2016-02-18:
On a toggle widget, I use the following code : 

`
int CameraThread::setToggleWidget(QString widgetName, int toggleValue)
{
    CameraWidget* widget = m_widgets[widgetName];

    if (widget)
    {
        int ret = gp_widget_set_value(widget, &toggleValue);
        if (ret < GP_OK) {
            qWarning() << "Unable to toggle widget :" << widgetName;
        }
        return ret;
    } else {
        qWarning() << "Widget not found :" << widgetName;
        return -1;
    }

    return GP_OK;
}`

gauravraj, 2016-02-18:
gp_widget_set_value works for me. After setting value to widget, I try applying the widget to camera using  gp_camera_set_config() function. Its in this function that I'm getting -1 error value. But if I set this property using command line - 

> gphoto2 --set-config movie=1
It works perfectly.

msmeissn, 2016-02-18:
can I perhaps access your whole code somehow? I dont see the issue so far ;(

gauravraj, 2016-02-18:
My present project has lot of other files. It'll be difficult for you to debug. Let me make another project just to test the video switching and then update you. Give me some time.

gauravraj, 2016-02-19:
Hi, I found out the issue. It was not with the code but with the project. I was using pthread library in my project. This was creating some issue with libgphoto2 library(don't know what exactly). But when I removed the pthread library, it worked perfectly fine. I had linked it like this in my qt project -
```
unix|macx: LIBS += -L/usr/lib/x86_64-linux-gnu/ -lpthread
INCLUDEPATH += /usr/lib/x86_64-linux-gnu
DEPENDPATH += /usr/lib/x86_64-linux-gnu
```
Now I'm using Qt's QThread for my threading purposes instead of pthread. 

Thanks for the help.

closed 2016-02-19 by gauravraj

rlamarche, 2016-02-19:
Thank you for the info. I'm also using QT threads with libgphoto with no problems. I've dedicated a qthread to send gphoto commands, with a command queue.

gauravraj, 2016-02-19:
I have used pthread with my previous libgphoto2 projects but this one is creating issues. Anways thanks again.