gphoto/libgphoto2 #44: --new does not work with EOS 6D

arnew, 2016-03-17:
Hi!

I try to transfer only the images taken since the last transfer, but i always get a transfer of all images.
Example Workflow:
A: Take some Pictures
B: gphoto2 --new -P  => all images transferred
C: process and delete images on host
D: take more Pictures
E: gphoto2 --new -P  => all (old and new) images transferred

I am running gphoto2/libgphoto2 build from git.
I can provide traces if they are useful.

Greetings,
Arne

arnew, 2016-03-17:
here are the traces.

[eos6d_B.txt](https://github.com/gphoto/libgphoto2/files/178455/eos6d_B.txt)
[eos6d_E.txt](https://github.com/gphoto/libgphoto2/files/178456/eos6d_E.txt)

arnew, 2016-03-18:
ok, i have had a bit of a look at the source, and it seems that for this feature to work, the camera needs to be a canon model (check) and needs to implement PTP_OC_CANON_SetObjectArchive (not the case for EOS 6D) [see: https://github.com/gphoto/libgphoto2/blob/fe92c463e5348cd0cfc051fdc8899a2f599a83f8/camlibs/ptp2/library.c#L6063 ]

The camera does support PTP_OC_CANON_EOS_SetObjectAttributes, which sadly seems rather undocumented, but looks a bit promising....

msmeissn, 2016-03-18:
yeah. mght be straight forward to implement

arnew, 2016-03-18:
https://github.com/arnew/libgphoto2/commit/ebe4916f63c4352c98320ca199acc24233cb9c90
this worked with my 50D, i will do more tests with the 6D and M

msmeissn, 2016-03-18:
i merged your pull request, thanks. :)

closed 2016-03-19 by arnew