gphoto/libgphoto2 #50: multiple Canon 1200D's are giving randomly "Canon EOS Full-Press failed" error

benvb, 2016-04-12:
Hi,

I also commented on issue #30 because i think we are having the same problem.

So the error is as following:

Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

I get this error quite random, sometimes after taking 3 pictures successfully, sometimes after 1 and sometimes after 10.
We are having manny 1200D and they all give this problem.

The command i use is the following "gphoto2 --capture-image-and-download"
Also i tried all the "eosremoterelease" options, but with no results.
My application don't need auto focus, so all tests are done with manual focus.

The log file you can find [here](https://www.dropbox.com/s/6qs8fdia7cxa5a6/logfile.txt?dl=0)
The version i am using is 2.5.10.1

Thanks!
Ben

msmeissn, 2016-05-30:
if this happens on the first image capture for a gphoto2 call, I have found a timing problem and fixed this in git master. commit 8bdc423dca169752b892cfe8d7af26c650d65d8b . can you test?

benvb, 2016-05-31:
Today i was testing your fix. It indeed solved the problem of random errors in my case.
But there is still a lag of a second before the the picture is taken. With other cameras i don't experience this lag when executing the command "gphoto2 --capture-image".

Thanks for looking into this!

ohens, 2016-06-06:
Thanks, this fix helped me, too!
I'm also having the 1 second gap before a picture is taken which I didn't have with other cameras and also don't have with v2.5.8.

msmeissn, 2016-06-20:
I hope this fixes the 1 second gap, it at least does for me. If you could test your usecases, I hope i did not break anything :)

ohens, 2016-06-27:
Hi, thanks for your fix!
The 1 second gap is gone now, but there is still an issue left:
If the camera is not able to gain focus and refuses to take an image, something seems to break.
I'm using libgphoto2 directly, the error code I get after taking a picture without being able to gain focus is '-110'. If I try to take another picture afterwards I got the error '-53' before your latest fixes. Now I get the error '-60'. I need to restart my system to be able to use the camera again.

benvb, 2016-07-06:
It is indeed fixed, no other problems by so far. Thanks a lot!

Those errors of the comment above me i never experienced. In my case i'm using libgphoto through gphoto.