gphoto/libgphoto2 #51: capture-image-and-download doesn't work 2nd time

jamsterc, 2016-04-21:
Hi Marcus et al,

Unfortunately I am experiencing this issue with the latest gphoto2 v2.5.10. I am using a Canon EOS 1200D usb-connected to a Raspberry Pi. The camera is set in video mode, as capture-image-and-download wouldn't work even the first time with P, A or M modes.

Here is what happens:
*pi@raspberrypi ~ $ gphoto2 --capture-image-and-download
ERROR: Could not capture image.
ERROR: Could not capture.*

*pi@raspberrypi ~ $ gphoto2 --capture-preview
 Error (-110: 'I/O in progress') 

*For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-preview*
*
Please make sure there is sufficient quoting around the arguments.*

If I add the --debug flag it works, but not if I set a debug-logfile! 

I've attached the logfile and camera configuration.

(Btw, I removed those tethering files: 
/usr/share/dbus-1/services/org.gtk.Private.GPhoto2VolumeMonitor.service
/usr/share/gvfs/mounts/gphoto2.mount
/usr/share/gvfs/remote-volume-monitors/gphoto2.monitor
/usr/lib/gvfs/gvfs-gphoto2-volume-monitor)

I am at a loss now as to how to resolve it, and am even considering looking for an alternative product, so any help would be much appreciated.

Regards,
Jam
[logfile.txt](https://github.com/gphoto/libgphoto2/files/230519/logfile.txt)

[cam_config.txt](https://github.com/gphoto/libgphoto2/files/230522/cam_config.txt)

msmeissn, 2016-04-23:
can you try the patch i attached to your sf ticket.

(sorry i am still poking in the dark here. and dont have a 1200D availanble)

jamsterc, 2016-04-25:
I've tried your patch - please see my reply to the SF post.

fraxinas, 2016-04-25:
Hallo Marcus,
i've got a similar issue (maybe). In effect, it comes down to the same symptom: i can only capture one photo, the next time i run my captur function, it fails.
I'm not working with gphoto2 though. Instead, i'm writing a little photobooth application on top of libgphoto2, gtk3+ and gstreamer.
code is here: [https://git.schaffenburg.org/fraxinas/photobooth/src/master/photobooth.c](https://git.schaffenburg.org/fraxinas/photobooth/src/master/photobooth.c)
i'm capturing video preview in a seperate thread, on mouse click in the main window, the video capture thread is paused and i'm taking a snapshot to memory and display it (this works fine even though there are a bunch of error messages):
```
0:00:02.822321102 20352      0x1154450 DEBUG             photobooth photobooth.c:290:photo_booth_video_capture_thread_func:<PhotoBooth@0x11b6130> captured frame (28 frames total)
0:00:02.836396625 20352      0x118c490 DEBUG             photobooth photobooth.c:626:photo_booth_clicked:<PhotoBooth@0x11b6130> photo_booth_clicked state=1
0:00:02.836425429 20352      0x118c490 INFO              photobooth photobooth.c:674:photo_booth_snapshot:<PhotoBooth@0x11b6130> SNAPSHOT!
0:00:03.285514397 20352      0x118c490 DEBUG             photobooth photobooth.c:684:photo_booth_snapshot:<PhotoBooth@0x11b6130> photo_booth_preview! halt video_bin...
0:00:03.287090692 20352      0x118c490 DEBUG             photobooth photobooth.c:699:photo_booth_snapshot:<PhotoBooth@0x11b6130> linking <photobin> ! <gdkpixbufoverlay0> ret=1
0:00:03.291285010 20352      0x1154450 DEBUG             photobooth photobooth.c:290:photo_booth_video_capture_thread_func:<PhotoBooth@0x11b6130> captured frame (29 frames total)
0:00:03.304000196 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
0:00:03.404541904 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
0:00:03.505099341 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
0:00:03.605688076 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
0:00:03.706336082 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
0:00:05.925357232 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, camera_nikon_capture [library.c:2967]:Parentobject of newobject 0x2b000001 is 0x0 now?
0:00:06.035873804 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x100b receiving resp failed: PTP Invalid Object Handle (0x2009)
0:00:06.035910724 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, camera_nikon_capture [library.c:2998]:deleteobject(2b000001) failed: 2009
0:00:06.037166771 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, ptp_usb_getresp [usb.c:428]:PTP_OC 0x90c3 receiving resp failed: PTP Invalid Object Handle (0x2009)
0:00:06.037183950 20352      0x118c490 DEBUG             photobooth photobooth.c:148:_gphoto_gst_errordumper: GPhoto 0, camera_nikon_capture [library.c:3001]:deleteobjectinsdram(2b000001) failed too: 2009
0:00:06.038686357 20352      0x118c490 DEBUG             photobooth photobooth.c:600:photo_booth_take_photo: gp_camera_capture gpret=0 Pathname on the camera: //capt0000.jpg
0:00:06.038704091 20352      0x118c490 DEBUG             photobooth photobooth.c:605:photo_booth_take_photo: gp_file_new gpret=0
0:00:06.039104579 20352      0x118c490 DEBUG             photobooth photobooth.c:608:photo_booth_take_photo: gp_camera_file_get gpret=0
0:00:06.039119748 20352      0x118c490 DEBUG             photobooth photobooth.c:617:photo_booth_take_photo: gp_camera_file_delete gpret=0
0:00:06.039172316 20352      0x118c490 INFO              photobooth photobooth.c:717:photo_booth_snapshot:<PhotoBooth@0x11b6130> photo_booth_snapshot now waiting for user input... PB_STATE_ASKING
```
the next mouse click unpauses the video capture thread and i see video again
when i now click again, it attempts to take another photo, but this time it fails:
```
0:00:23.873011660 20352      0x118c490 INFO              photobooth photobooth.c:674:photo_booth_snapshot:<PhotoBooth@0x11b6130> SNAPSHOT!
0:00:24.320044535 20352      0x118c490 DEBUG             photobooth photobooth.c:684:photo_booth_snapshot:<PhotoBooth@0x11b6130> photo_booth_preview! halt video_bin...
0:00:24.320080522 20352      0x118c490 DEBUG             photobooth photobooth.c:692:photo_booth_snapshot:<PhotoBooth@0x11b6130> photo_booth_preview! unblock photo_bin...
0:00:24.321497491 20352      0x118c490 DEBUG             photobooth photobooth.c:699:photo_booth_snapshot:<PhotoBooth@0x11b6130> linking <photobin> ! <gdkpixbufoverlay0> ret=1
[... cut ...]
0:00:25.772080572 20352      0x118c490 INFO              photobooth photobooth.c:721:photo_booth_snapshot:<PhotoBooth@0x11b6130> photo_booth_take_photo_push_to_appsrc failed!
```
i actually modified the sample-capture.c example so that there are two capture_to_memory calls with a gp_camera_capture_preview trying to mimic what my application does, but couldn't reproduce the problem that way. no idea what i'm doing wrong. i'd be grateful for any hints. Camera model is a Nikon D7200 btw.

Cheers,
Fraxinas

msmeissn, 2016-05-29:
i fixed the device busy during preview capture (in mpovie mode) for the eos1200d with commit 50f64234f2ead2e7995335444cfe7e2cd7166733

msmeissn, 2016-05-29:
and perhaps the other issue with eos1200d is fixed with commit 8bdc423dca169752b892cfe8d7af26c650d65d8b , so if you could test current git?