gphoto/libgphoto2 #52: videos not shown on GoPro Hero 4 Session camera

ossman, 2016-04-28:
Downstream bug: https://bugzilla.redhat.com/show_bug.cgi?id=1331161

In short, no video files are shown from a GoPro Hero 4 Session. Works fine using libmtp though.

ossman, 2016-04-28:
I'm trying to debug this myself, but the code is not terribly easy to follow with all the caching going on. I think this might be a symptom though:

```
30.859791 folder_list_func [library.c:5318](0): Duplicate foldername 'DCIM' in folder '/store_00010001'. Ignoring nth entry.
```

as later references are to `/store_00010001/DCIM_00000001`, which I assume is a generated name to avoid conflict. Unfortunately it doesn't look like `find_child()` can handle these generated names as it directly compares it to the objects reported file name. Hence any fail to list that folder fails because it cannot map the name back to a object id.

If found the oddity in `camera_init()` where it first enumerates all root objects on every storage, but then proceeds to go over every storage separately and enumerate the root objects again. It did however not fix anything if I removed the initial enumeration.

msmeissn, 2016-04-28:
i had changed this stuff back in current git and 2..5.10 ... can you try that

msmeissn, 2016-04-28:
commit ce4d4bcd5480fd82f335b81e54ba91603d66f833

ossman, 2016-04-29:
2.5.10 seems to work just fine. Many thanks. :)

msmeissn, 2016-04-29:
thanks for confirming! :)

closed 2016-04-29 by msmeissn