gphoto/libgphoto2 #54: Segmentation violation near exit of gphoto2 under libusb_handle_events_timeout_completed

RalphCorderoy, 2016-05-07:
Arch Linux, gphoto2 2.5.9-1.  Ran `gphoto2 -l -L`, it produced the output I expected, and then SEGV'd.  Core dump backtrace from coredumpctl(1).

    Stack trace of thread 2484:
    #0  0x00007f8607876eca n/a (libusb-1.0.so.0)
    #1  0x00007f86078712b9 n/a (libusb-1.0.so.0)
    #2  0x00007f86078721c3 libusb_handle_events_timeout_completed (libusb-1.0.so.0)
    #3  0x00007f86078722af libusb_handle_events (libusb-1.0.so.0)
    #4  0x00007f8607a82ba8 n/a (usb1.so)
    #5  0x00007f8607a82d05 n/a (usb1.so)
    #6  0x00007f8609448478 gp_port_close (libgphoto2_port.so.12)
    #7  0x00007f8609658be7 gp_camera_exit (libgphoto2.so.6)
    #8  0x00007f8609658d7c gp_camera_free (libgphoto2.so.6)
    #9  0x00007f8609658f65 gp_camera_unref (libgphoto2.so.6)
    #10 0x000000000040e738 n/a (gphoto2)
    #11 0x00000000004074be n/a (gphoto2)
    #12 0x00007f860880b710 __libc_start_main (libc.so.6)
    #13 0x0000000000407a09 n/a (gphoto2)
                
    Stack trace of thread 2487:
    #0  0x00007f86088c9c3d poll (libc.so.6)
    #1  0x00007f8607877cbc n/a (libusb-1.0.so.0)
    #2  0x00007f860922e424 start_thread (libpthread.so.0)
    #3  0x00007f86088d2cbd __clone (libc.so.6)

I had a similar failure a couple of weeks ago, but that time was on deleting with -DR.  Again, it was after the work was done, and a following "list" showed the deleting had finished.

Hard to reproduce, perhaps it is timing dependent.

msmeissn, 2016-05-07:
This is a bug I fixed with libgphoto2 2.5.10 I think. (the freeing of async usb transfers was doubled)

RalphCorderoy, 2016-05-07:
Thanks for the quick response.  It didn't occur to me that Arch Linux might be lagging.  I've flagged its gphoto2 package as being out of date so the fix will hopefully ripple down to all users in about a week, post "testing".

msmeissn, 2016-05-28:
closing as solved

closed 2016-05-28 by msmeissn