gphoto/libgphoto2 #55: Segmentation fault

roland20162016, 2016-05-09:
Hi

**When I use the 'gphoto2 -a' command**.Receive Segmentation fault!

What can I do?
------------------------------------------------------------------------------------
root@OpenWrt:~# gphoto2 -a
Segmentation fault
------------------------------------------------------------------------------------
root@OpenWrt:~# gphoto2 -v
gphoto2 2.4.11

Copyright (c) 2000-2011 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.4.11         ccache_cc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.4.12         ccache_cc, ltdl, EXIF
libgphoto2_port 0.8.0          ccache_cc, ltdl, USB (libusb1), serial without locking

---------------------------------------------------------------------------------------------------------
**Status
System
Hostname	OpenWrt
Model	VoCore
Firmware Version	OpenWrt Barrier Breaker 14.07 / LuCI 0.12 Branch (0.12+git-15.143.53098-670e046)
Kernel Version	3.10.49**

roland20162016, 2016-05-09:
It's the debug information!

root@OpenWrt:~# gphoto2 --debug  -a
0.000042 main(2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.001242 main(2): gphoto2 2.4.11
0.001824 main(2): gphoto2 has been compiled with the following options:
0.002349 main(2):  + ccache_cc (C compiler used)
0.002888 main(2):  + popt (mandatory, for handling command-line parameters)
0.003427 main(2):  + exif (for displaying EXIF information)
0.003961 main(2):  + no cdk (for accessing configuration options)
0.004496 main(2):  + no aa (for displaying live previews)
0.005032 main(2):  + jpeg (for displaying live previews in JPEG format)
0.005564 main(2):  + readline (for easy navigation in the shell)
0.006152 main(2): libgphoto2 2.4.12
0.006761 main(2): libgphoto2 has been compiled with the following options:
0.007294 main(2):  + ccache_cc (C compiler used)
0.007875 main(2):  + ltdl (for portable loading of camlibs)
0.008417 main(2):  + EXIF (for special handling of EXIF files)
0.008980 main(2): libgphoto2_port 0.8.0
0.009574 main(2): libgphoto2_port has been compiled with the following options:
0.010097 main(2):  + ccache_cc (C compiler used)
0.010644 main(2):  + ltdl (for portable loading of camlibs)
0.010770 main(2):  + USB (libusb, for USB cameras)
0.010890 main(2):  + serial (for serial cameras)
0.011013 main(2):  + no resmgr (serial port access and locking)
0.011138 main(2):  + no ttylock (serial port locking)
0.011264 main(2):  + no lockdev (serial port locking)
0.011398 main(2): CAMLIBS env var not set, using compile-time default instead
0.011519 main(2): IOLIBS env var not set, using compile-time default instead
0.012045 setting/gphoto2-setting.c(2): Creating $HOME/.gphoto
0.012622 setting/gphoto2-setting.c(2): Can't open settings file for reading
0.012802 setting/gphoto2-setting.c(2): Loading settings from file "/root/.gphoto/settings"
0.012992 setting/gphoto2-setting.c(2): Can't open settings for reading
0.013180 setting/gphoto2-setting.c(2): Creating $HOME/.gphoto
0.013421 setting/gphoto2-setting.c(2): Can't open settings file for reading
0.013562 setting/gphoto2-setting.c(2): Loading settings from file "/root/.gphoto/settings"
0.013741 setting/gphoto2-setting.c(2): Can't open settings for reading
0.014212 main(2): The user has not specified both a model and a port. Try to figure them out.
0.014481 gphoto2-port-info-list(2): Using ltdl to load io-drivers from '/usr/lib/libgphoto2_port/0.8.0'...
0.015857 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/disk'.
0.019825 gphoto2-port/disk(2): found mtab fsname rootfs
0.020241 gphoto2-port/disk(2): found mtab fsname /dev/root
0.027184 gphoto2-port/disk(2): found mtab fsname proc
0.030989 gphoto2-port/disk(2): found mtab fsname sysfs
0.031535 gphoto2-port/disk(2): found mtab fsname tmpfs
0.032211 gphoto2-port/disk(2): found mtab fsname /dev/mtdblock5
0.032908 gphoto2-port/disk(2): found mtab fsname overlayfs:/overlay
0.033531 gphoto2-port/disk(2): found mtab fsname tmpfs
0.034187 gphoto2-port/disk(2): found mtab fsname devpts
0.034788 gphoto2-port/disk(2): found mtab fsname debugfs
0.036273 gphoto2-port-info-list(2): Could not load port driver list: 'Unspecified error'.
0.036985 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/ptpip'.
0.040342 gphoto2-port-info-list(2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/lib/libgphoto2_port/0.8.0/ptpip'.
0.051091 gphoto2-port-info-list(2): Loaded '' ('^ptpip') from '/usr/lib/libgphoto2_port/0.8.0/ptpip'.
0.052222 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/serial'.
0.057589 gphoto2-port-info-list(2): Loaded 'Serial Port 0' ('serial:/dev/ttyS0') from '/usr/lib/libgphoto2_port/0.8.0/serial'.
0.058540 gphoto2-port-info-list(2): Loaded 'Serial Port 1' ('serial:/dev/ttyS1') from '/usr/lib/libgphoto2_port/0.8.0/serial'.
0.059268 gphoto2-port-info-list(2): Loaded '' ('^serial') from '/usr/lib/libgphoto2_port/0.8.0/serial'.
0.059973 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/usb1'.
0.075345 gphoto2-port-info-list(2): Loaded '' ('^usb:') from '/usr/lib/libgphoto2_port/0.8.0/usb1'.
0.090898 gphoto2-port-info-list(2): Loaded 'Universal Serial Bus' ('usb:') from '/usr/lib/libgphoto2_port/0.8.0/usb1'.
0.091631 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/usbdiskdirect'.
0.095258 gphoto2-port-info-list(2): Called for filename '/usr/lib/libgphoto2_port/0.8.0/usbscsi'.
0.098975 gphoto2-port-info-list(2): Counting entries (8 available)...
0.099704 gphoto2-port-info-list(2): 4 regular entries available.
0.104019 gphoto2-abilities-list(2): Using ltdl to load camera libraries from '/usr/lib/libgphoto2/2.4.12'...
0.115902 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/adc65'.
0.116686 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/agfa_cl20'.
0.117238 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/aox'.
0.117769 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/ax203'.
[... 53 lines cut ...]
0.177247 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/stv0674'.
0.177771 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/stv0680'.
0.178300 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/sx330z'.
0.178827 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/topfield'.
0.179351 gphoto2-abilities-list(2): Found '/usr/lib/libgphoto2/2.4.12/toshiba_pdrm11'.
[... cut ...]
2.072852 gphoto2-port(2): Setting timeout to 500 milliseconds...
2.080827 gphoto2-port(2): Setting settings...
2.081339 gphoto2-port-info-list(2): Getting info of entry 3 (8 available)...
2.089358 gphoto2-port(2): Setting timeout to 5000 milliseconds...
2.090154 gphoto2-port(2): Setting settings...
2.090690 libusb1(2): gp_port_usb_update(old int=0, conf=-1, alt=-1), (new int=0, conf=-1, alt=-1)
2.091028 gphoto2-abilities-list.c(1): Auto-detecting USB cameras...
Segmentation fault
root@OpenWrt:~#

msmeissn, 2016-05-09:
libgphoto2 2.4.13 fixed this startup crash I think.

2.4.12 was released january 2012, so it is time for a newer libgphoto2 ...

closed 2016-05-09 by msmeissn

roland20162016, 2016-05-09:
Thans!