gphoto/libgphoto2 #56: Libgphoto2-2.5.10 Installation Problem

Haliph, 2016-05-13:
While attempting to install libgphoto2 version 2.5.10, I noticed that after make install, my library version was still 2.5.8. I have gphoto2 version 2.5.10 so the newly added features such as gp_camera_get_single_config produce a "symbol lookup error". I'm not sure why libgphoto2 is not updating/installing properly. Here's my process

sudo ./configure --prefix=/usr/local
sudo ./make
sudo ./make check
sudo ./make install

Thanks.

msmeissn, 2016-05-16:
It will be the system library installed in /usr/ that is still being used

--prefix=/usr 

will overwrite the system libgphoto2 library.

Haliph, 2016-05-18:
Unfortunately, now when I run sudo make, I get stuck in a loop:
        mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/usr/share/locale\" -DCAMLIBS=\"/usr/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2 -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -MT soundvision/soundvision.lo -MD -MP -MF soundvision/.deps/soundvision.Tpo -c soundvision/soundvision.c  -fPIC -DPIC -o soundvision/.libs/soundvision.o
depbase=`echo soundvision/agfa_cl18.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
        /bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..  -D_GPHOTO2_INTERNAL_CODE -DLOCALEDIR=\"/usr/share/locale\" -DCAMLIBS=\"/usr/lib/libgphoto2/2.5.10\" -I.. -I.. -I../libgphoto2_port -I../libgphoto2 -I../libgphoto2   -g -O2 -Wall -Wmissing-declarations -Wmissing-prototypes -MT soundvision/agfa_cl18.lo -MD -MP -MF $depbase.Tpo -c -o soundvision/agfa_cl18.lo soundvision/agfa_cl18.c &&\
        mv -f $depbase.Tpo $depbase.Plo

It then repeats over and over. As well, it does the same thing for sudo make check.

msmeissn, 2016-05-19:
if there is a loop, there likely is inconsistent time on the device, the device running in the future or past.