gphoto/libgphoto2 #57: Capturing not possible for Sony DSC-T900

genfersee, 2016-05-16:
Dear all,

I have configured my Sony DSC-T900 in PTP/MTP mode.
I am able to download a picture within camera memory stick.

My issue is that I am not able to capture a picture.
I get (T900 camera seems to be detected as model W130!):

MacPro:~ Yohann$ env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --abilities
Abilities for camera             : Sony DSC-W130 (PTP mode)                    
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Capture not supported by the driver
Configuration support            : no
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes

and:

MacPro:~ Yohann$ gphoto2 --capture-image
                                                                               
env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image
                                                                               
*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (No such file or directory). Make sure no other program (MacOS PTPCamera service) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-53: 'Could not claim the USB device') ***   

Thanks you so much in advance for your help :):)

Kind regards,
Yohann

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/266691/my-logfile.txt)

msmeissn, 2016-05-16:
Can you switch the camera to PC Control Mode?

not all sonys might be controllable

genfersee, 2016-05-16:
Dear Marcus, 

Thanks for your quick answer!
I am working on a Mac with Macport.
I have the following modes in the camera:
auto, PTP/MTP, PicBridge, auto

msmeissn, 2016-05-18:
I think it is not possible.

The Sony with capture support have a "PC Remote" in that list too.

So I think we are out of luck on this camera for remote capture support.

closed 2016-05-18 by msmeissn