gphoto/libgphoto2 #58: Capturing with Canon EOS M3

TobiasBaeumlin, 2016-05-17:
It seems that capturing images with the Canon EOS M3 is not possible though it is listed with 
gphoto2 --list-cameras:

gphoto2 --list-cameras | grep -i "canon eos m3"
	"Canon EOS M3"

Neither the option --capture-image nor --caption-tethered works.

[capture-eos-m3.txt](https://github.com/gphoto/libgphoto2/files/268142/capture-eos-m3.txt)
[capture-tethered.txt](https://github.com/gphoto/libgphoto2/files/268148/capture-tethered.txt)

This is rather unfortunate since I specially bought this model based on this.
Is there any way around this limitation or is full support for this model planned in the near future?

msmeissn, 2016-05-18:
we currently do not know how the EOS M series capture works or if it is possible at all.

information on that is missing.

coderliss, 2017-02-27:
There is a app called "camera connect" from canon, It can capture photo remotely. I captured it's ptp/ip packet and trying to support EOS M3 on libgphoto. It works but with problem.
After first capture, you must reboot your camera manually before you take the second picture.

I don't know is because bugs in my code or something. I'm appreciate if someone could help me.
here is my test code, patch for camlibs/ptp2/library.c
[eosm3.txt](https://github.com/gphoto/libgphoto2/files/802541/eosm3.txt)

msmeissn, 2017-02-27:
actually can you just try upgrading to current GIT ... It might work without modification on the M3.

The code change you did sees small bits of difference in what we do in config.c::camera_prepare_canon_eos_capture

coderliss, 2017-02-27:
I'm already working on current GIT(https://github.com/gphoto/libgphoto2.git)

It's there anyone knows ptp_canon_eos_setremotemode 0x15 or 0x01 code means?
And ptp_canon_eos_setdevicepropvalue PTP_DPC_CANON_EOS_EVFOutputDevice 0x00000008 means?
There is too many magic numbers.

here is debug log of patched and unpatched version.

[unpatched.txt](https://github.com/gphoto/libgphoto2/files/803557/unpatched.txt)
[patched.txt](https://github.com/gphoto/libgphoto2/files/803587/patched.txt)

xsnrg, 2017-03-11:
In looking through things trying to find information for my M5 that I am also trying to get working, I stumbled across this page:  [sdk-information](https://www.usa.canon.com/internet/portal/us/home/explore/solutions-services/digital-camera-sdk-information)

 I am hoping that it just means the SDK doesn't support capture yet, but the camera does.  It seems logical anyway as the camera connect app is capable of capture.

Blueshawk, 2017-04-08:
I've been watching Eos-M threads in hopes of some of the other model's fixes trickling back to the Eos-M(1) 
I just retested it with current git and the m1 sadly still reports capture unsupported as before. Emails to Canon about a firmware fix were met with a stern. "Whatever comes from Japan" bit of finger pointing and no other help. The later models appear to be using wireless which would seem to make them more willing to include remote capture. Canon has dropped the ball and set it on fire with this one, especially as the website continues to make claims to the contrary.

msmeissn, 2017-04-16:
So capture-preview, capture-image and trigger-image now work for my EOS M10 in current git.

The configuration handling is not fully working

thinkgareth, 2018-05-17:
Some good news... The next generation of the M range is now available - The M50.

I've just read that it is the first of the M range to support wired tethering with the desktop Canon EOS Utility app(!).

There's also a "Electronic shutter only" mode - but thats tucked away in a "fully auto" SCN mode - such a shame!

I'll try and get my hands on one and send over the debug output... Would you like me to try and test any functions to see if the existing M support is compatible? Anything else I can do when i'm next to the model to help?

msmeissn, 2018-06-14:
yes please. I have confirmed a M100 working, someone else also confirmed M50.

tellblom, 2018-10-03:
I can confirm that the M50 Camera reports

Model                          Port
----------------------------------------------------------
USB PTP Class Camera           usb:001,004

But when running gphoto2 --capture-image-and-download the camera takes the picture but Duisplays an error message saying " Err 70 An error prevented shooting. Turn the camera off and on again or re-install the battery."

Th eeroor on teh console is:
gphoto2 --capture-image-and-download

*** Error ***
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***
PTP I/O Error

What can I do to help you getting the M50 to work?

msmeissn, 2018-10-03:
which libgphoto2 version is in use?

can you provide drebug output of abvoe? 

gphoto2 --capture-image-and-download --debug --debug-logfile=xx.log  
attacjh xx.log here

tellblom, 2018-10-03:
Hi, 
here comes the versions and debug log

Using gphoto2 version 2.5.11, libgphoto2 2.5.12 and libgphoto2_port 0.12.0

[xx.log](https://github.com/gphoto/libgphoto2/files/2443357/xx.log)

msmeissn, 2018-10-04:
I added EOS M capture support in libgphoto2 2.5.14, so please upgrade to a newer version first

tellblom, 2018-10-04:
Ok, I have to to find a package for octopi with the upgraded version then.
Thank you

msmeissn, 2019-02-11:
considering this is done .

closed 2019-02-11 by msmeissn

schousda, 2019-12-12:
> There's also a "Electronic shutter only" mode - but thats tucked away in a "fully auto" SCN mode - such a shame!
Does the electronic shutter of the EOS M3 work or not?
Could you please help me with that question.
Thanks.

lagrandeours, 2021-02-25:
Hello,
I tried with the eos M2, and it's work fine for capture and settings, i did not test live view yet. 
But for M3 it seems not working due to Wi-Fi settings there is no option to shut down the Wi-Fi, from the camera, so it sucks with freezen view... very strange

wokawoka, 2021-03-21:
Any update on the issue @lagrandeours ?
I would like to use an eos M3 as a webcam on linux and I'm very interesting to understand if the ligphoto2 support works or not.
Thank you!