gphoto/libgphoto2 #60: Canon 80D Support

dabbill, 2016-05-20:
Not sure what all information is needed for requesting a new camera to be added. Please let me know anything I can provide to help out with this process. 

I am able to compile and test code if needed, or any debugging you need done. 

Thanks.

msmeissn, 2016-05-20:
Can you just try it with the libgphoto2 version you have, the generic code should already make it work.

we would need to USB ids for adding the device to our database (run lsusb)

dabbill, 2016-05-21:
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon EOS 80D
  Version: 3-1.0.1
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button
        Canon Wifi support

Storage Devices Summary:
store_00020001:
        StorageDescription: SD
        VolumeLabel: None
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 64457015296 (61471 MB)
        Free Space (Bytes): 62844960768 (59933 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 80D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 56% (56)

Debug output file. 
https://dl.dropboxusercontent.com/u/47070387/Canon%2080D/Canon-80D.txt

msmeissn, 2016-05-21:
thanks! can you try

gphoto2 --capture-image-and-download

and 

gphoto2 --capture-preview

dabbill, 2016-05-21:
gphoto2 --capture-image-and-download
                                                                               
*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

gphoto2 --capture-preview
                                                                               
*** Error ***              
PTP Operation Not Supported

*** Error ***              
An error occurred in the io-library ('Unsupported operation'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-preview

Please make sure there is sufficient quoting around the arguments.

gphoto2 --capture-preview
Saving file as capture_preview.jpg

I had to keep plugging and unplugging the camera to even get it to work once on the --capture-preview.

dabbill, 2016-05-24:
I built the new version of the lib. Camera is detected in kamera now. Let me know if there is any more testing / debugging i can do to help out. 

gphoto2 --capture-image-and-download
                                                                               
*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.

gphoto2 --capture-image-and-download
                                                                               
*** Error ***              
PTP Operation Not Supported

*** Error ***              
An error occurred in the io-library ('Unsupported operation'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download

Please make sure there is sufficient quoting around the arguments.

https://dl.dropboxusercontent.com/u/47070387/Canon%2080D/my-logfile.txt

kentavv, 2016-06-02:
When I attach an 80D to a Windows computer, an icon of a computer appears on the 80D's display and the files on the camera are accessible. When I attach an 80D to a Linux computer, the 80D's display flashes almost certainly the same icon. Occasionally the icon remains, but generally the icon flashes too fast to be visible in all but a dark room. 

I write all of this, because if I issue a gphoto2 command to the 80D immediately after plugging the camera into the Linux computer, before the icon flashes on the 80D's display, the command is executed. For instance, --capture-image-and-download will trigger the shutter. The command eventually times out, but it's improvement over waiting a bit after plugging the 80D, when gphoto2 commands return the previously reported error (also included below.) I hope this report is somehow helpful. I also am willing to help with any debugging. The gphoto2 and libgphoto2 that I am using are up to date from GitHub. Thank you 

*** Error ***              
PTP Operation Not Supported

*** Error ***              
An error occurred in the io-library ('Unsupported operation'): No error description available
*** Error (-6: 'Unsupported operation') ***

dabbill, 2016-06-02:
Kentavv, not sure if you use Darktable or not, but I have added Canon 80D support for the full size RAW files. 

https://github.com/dabbill/darktable

I have a pull request in to get the code merged in to darktable.

kentavv, 2016-06-02:
Thank you Dabbill, that will be very helpful.

As a follow-up, in case it'll help, attached are two --debug outputs (hopefully), both of --capture-image-and-download. The file waited.txt was generated after waiting a few seconds after attaching the 80D by USB, when no command will work. The file immediate.txt was generated immediately after plugging in the 80D, when it's possible to get gphoto2 commands to trigger the shutter.

[waited.txt](https://github.com/gphoto/libgphoto2/files/296526/waited.txt)
[immediate.txt](https://github.com/gphoto/libgphoto2/files/296527/immediate.txt)

kentavv, 2016-08-08:
I'm need three additional Canon 70D or 80D cameras... and just wanted to check if support for the 80D has been added. I tried the latest code in the repository and see the same error. But perhaps it's the computer? Again, just wanted check before making the purchase. I could hold off buying the cameras for a month, but then would have to go with what works. Hoping for the best. Thank you

dabbill, 2016-08-09:
Kentavv, Canon 80D works if you build from git. I think it was added right after 2.5.10 was released.

kentavv, 2016-08-09:
Huh, any idea what I might be doing wrong? I have the most recent libgphoto2 and gphoto2 from git, and am seeing the same sort of errors. I can get or trigger the camera if I'm lucky by immediately running gphoto2 after plugging the camera in, but can never download files.

Thank you

[kent@localhost gphoto2]$ env LANG=C /usr/local/bin/gphoto2 --debug --debug-logfile=my-logfile.txt --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 80D                  usb:002,010     
[kent@localhost gphoto2]$ env LANG=C /usr/local/bin/gphoto2 --debug --debug-logfile=my-logfile.txt --list-config
                                                                               
*** Error ***              
PTP Operation Not Supported

*** Error ***              
An error occurred in the io-library ('Unsupported operation'): No error description available
*** Error (-6: 'Unsupported operation') ***    

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/408725/my-logfile.txt)

dabbill, 2016-08-09:
What linux distro are you using? Its possible you have your repo version of libgphoto2 installed and being used. 

I am using Fedora 24. I had to install my repo package, then remove all libgphoto2 files from /usr/lib64/. Then install the compiled version from git. Add file in /etc/ld.so.conf.d/ to include /usr/local/lib to make sure it was using the git version of libgphoto2

kentavv, 2016-08-09:
That's promising, I also have Fedora 24. I removed gphoto2* and libgphoto2* add /usr/local/lib and the error is unfortunately the same. The new libraries are the ones being linked in.

As a sanity check, these are the correct repositories?

git clone https://github.com/gphoto/gphoto2.git
git clone https://github.com/gphoto/libgphoto2.git

Thank you

[root@localhost gphoto2]# ldd /usr/local/bin/gphoto2 
	linux-vdso.so.1 (0x00007ffd77559000)
	libgphoto2.so.6 => /usr/local/lib/libgphoto2.so.6 (0x00007fcc2919d000)
	libgphoto2_port.so.12 => /usr/local/lib/libgphoto2_port.so.12 (0x00007fcc28f91000)
	libltdl.so.7 => /lib64/libltdl.so.7 (0x00007fcc28d70000)
	libaa.so.1 => /lib64/libaa.so.1 (0x00007fcc28b50000)
	libjpeg.so.62 => /lib64/libjpeg.so.62 (0x00007fcc288e5000)
...

dabbill, 2016-08-09:
Well, it was working for me on Fedora 24, but it appears to be broken again. I can say tho that it worked perfectly on Arch Linux. Just installed the -git versions of gphoto2 / libgphoto2 and never had a problem with Darktable or gphoto2 on CLI.

dabbill, 2016-08-09:
gphoto2 --auto-detect sees that its a canon 80d, but not sure why nothing else is working.

kentavv, 2016-08-09:
It's comforting that someone is seeing the same problem. I can gphoto to get a little further by immediately running gphoto after plugging the camera into the computer, but that works once if I'm lucky. I assumed the problem was a timing issue and someone in the know would be able to bump up a value. 

Do you know when gphoto last worked, so I could try checking out an old version, or a distribution that worked for sure? You mentioned Arch Linux.

Thank you

dabbill, 2016-08-09:
Worked great about 2 - 3 weeks ago on Arch Linux. It was working with Darktable on Fedora 24 last week. I didnt change anything to do with libgphoto2 other then rebooting the computer and now its not working. So not a clue what changed.

kentavv, 2016-08-09:
I tried updating libusb without improvement. After plugging in the 80D, I see the following...
1. The release button causes the LCD screen on top to say "busy"
2. I can run gphoto2 --list-folders and see a result (one time only)
3. The large flip LCD screen on the camera shows a computer icon
4. Running gphoto2 --list-folders results in the same PTP error
5. Now the release button causes the camera to trigger (and the computer icon disappears)
(The order of 4 and 5 can be swapped and the results are the same)

dabbill, 2016-08-09:
Yea, I am seeing the same thing that you are.

msmeissn, 2016-08-09:
What can happen is that another process, like gvfs or so grabs the camera and you can not run a gphoto2 session in parallel.      is the gvfs gphoto slave running?

kentavv, 2016-08-09:
It seems one gets a single chance to interact with the 80D from gphoto2 and then has to reattach the camera, but can do any number of operations. For instance, I can string together a bunch of commands, and they all run, including the capture, but only with the first invocation of gphoto2. All subsequent invocations result in the PTP error, until the camera is reattached.

E.g., the following executes fine once.

env LANG=C /usr/local/bin/gphoto2 --debug --debug-logfile=my-logfile.txt --list-folders --summary --list-folders --summary --trigger-capture --list-files

kentavv, 2016-08-09:
That's good to know there are other processes. I removed gvfs and killed all remaining gvfs processes and my previous message about one chance to run as many commands as you like is unfortunately still true.

kentavv, 2016-08-09:
I dropped to single user mode, which hopefully would have anything more than what's necessary running, and the results are the same. One chance to run gphoto2. If there was another process attached to the camera, would the computer icon remain on the LCD? Is there a way to diagnose if there is another competing process? Thank you

kentavv, 2016-08-09:
I  started to look at libusb to see how in software one could accomplish the same as reattaching the camera. Maybe that's what 'gphoto2 --reset' does? Unfortunately, the results are the same. A message appears in dmesg, but calling 'gphoto2 --reset' before or after the desired gphoto2 call, appears to change nothing regarding the camera.

[ 1568.292238] usb 2-3: reset high-speed USB device number 5 using ehci-pci

kentavv, 2016-08-09:
The "you get one shot" may not be by design. After gphoto2 interacts with the 80D, and subsequent calls fail, with the USB cable still attached, I can take any number of shot using the shutter release button, but if I press the "play" button to review shots, "Busy... please wait" appears on the flip out LCD and remains stuck until the USB cable is detached or the shutter release is pressed. There are a few other oddities like this after gphoto2 successfully talks to the camera. Sort of like, the camera is still partially attached to the computer. Sometimes the shutter release does not work after pressing the "play" button. It looks like it is working, but the shutter never releases, and then after detaching the USB cable, the camera jumps into the photo review mode, like the effect of pressing the "play" button was delayed until the camera was released from the computer. 

Hope this information is in some way helpful. Please let me know if there is anything I can do to help.

Thank you

xaphod, 2016-08-09:
Hi i'm the developer of thewifibooth.com iOS app. Not sure if it helps anyone, but I found that the eventing of the 80D has changed: modern Canons usually send event 0xC181 when an object has been created (photo taken). The 80D does not, it sends an 0xC1A7 instead. Inside the 0xC1A7 event, the object handle, format, and compressed sizes are still at their usual offsets (8, 16, and 28 bytes respectively).

kentavv, 2016-08-09:
Granted, I have no idea what I'm doing, but that sounds promising. I changed c181 to c1a7 in camlibs/ptp2/ptp.h, but nothing changed. Again, I don't know what else is involved. I did notice that when the camera is turned off, after being triggered by gphoto2, the flip out LCD often shows "Recording... Remaining images: xxx" where xxx has been 1 or 2. This is with a SD card installed. The recording never finishes, at least after five minutes it had not finished. Cycling the power on the camera resulted in the same message. Only after pulling the USB cable was the camera allowed to turn off. 

It seems so close, demanding commands like the following work... once, after plugging the camera in. (Although the camera never autofocused, which I think the 70D did.)

env LANG=C /usr/local/bin/gphoto2 --debug --debug-logfile=my-logfile.txt --capture-movie --stdout | vlc -

stloeffler, 2016-10-26:
Capturing single images seems to work for me, however detecting the focus info does not (see [exposure2.txt](https://github.com/gphoto/libgphoto2/files/553878/exposure2.txt)). On line 5323 (time code 0.710276), there is an OLCInfoChanged event, which, if I understand it correctly, has a mask value of 0x0900, but the event does not seem to get processed due to a packet size error. So gphoto2 waits until the timeout before taking the picture. Changing
```if (len != size-8) {``` to ```if (len != size-4) {``` in camlibs/ptp2/ptp-pack.c @ 2234 seems to fix the problem for me (i.e., a focusinfo event is reported and the exposure is started immediately), but I have no idea if this breaks compatibility with any other camera or introduces other problems.

msmeissn, 2016-10-26:
hmm, i have to review this. the OLCINfo format might have changed. :/

dabbill, 2016-11-29:
Just upgraded to libgphoto2 2.5.11-1 on Arch. Still cannot get 80D to work correctly in Darktable. 

Debug output from Darktable
[camera_control] loaded 8 port drivers.
[camera_control] 1 cameras connected
[camera_control] gphoto2 error: PTP Operation Not Supported
[camera_control] failed to initialize camera Canon EOS 80D on port usb:001,018
[camera_control] failed to initialize device Canon EOS 80D on port usb:001,018, probably causes are: locked by another application, no access to udev etc.

~ » gphoto2 --abilities
Abilities for camera             : Canon EOS 80D                               
Serial port support              : no
USB support                      : yes
Capture choices                  :
                                 : Image
                                 : Preview
                                 : Trigger Capture
Configuration support            : yes
Delete selected files on camera  : yes
Delete all files on camera       : no
File preview (thumbnail) support : yes
File upload support              : yes

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/617958/my-logfile.txt)

msmeissn, 2016-11-30:
same issue as in https://github.com/gphoto/libgphoto2/issues/92

msmeissn, 2016-12-04:
can you test this patch perhaps?

closed 2016-12-04 by msmeissn

dabbill, 2016-12-05:
This patch appears to be working now. One thing I do notice is when using Darktable the camera disconnects and reconnects randomly.

mjg, 2017-01-03:
I just applied this patch to libgphoto2-2.5.11-1.fc25.x86_64 on Fedora 25, and it turned support for my EOS 80D from mostly non-working (listing files, but no download, no tethering with gphoto2;, no darktable camera support) to working (fully as far as tested). Thanks!