gphoto/libgphoto2 #63: PTP/IP - different port for control and events

spectras, 2016-06-14:
Hello,

I am investigating the use of gphoto, which looks very promising, with a Fujifilm X-T1 camera. Over wifi.

So far, by specifying the port manually, I manage to have it open the control connection and successfully register itself. However, the camera expects the events connection on a different port, so everything stops at this point.

In detail:

 * Laptop opens control connection to camera (port 55740)
 * Laptop sends Init_Command_Request
 * Camera sends Init_Command_Ack
 * *Camera now expects an incoming events connection on port 55742*
 * Laptop opens event connection to camera (port 55740).
 * Laptop sends Init_Event_Request, which the camera does not understand.
 * Everything hangs.

I rtfs, and it seems there is no way at the moment to specify a different port of the event connection. Thus the feature request: allow it somehow.

msmeissn, 2016-06-15:
I could improve the --port ptpip: syntax. currently it can do

--port ptpip:IP[:port]  

but a secodnary :eventport could also be added

msmeissn, 2016-06-15:
you can try above commit?

with --port ptpip:CAMERAIP:55740:55742

spectras, 2016-06-15:
I checked it out, but it might be a few days before I can try it. Thanks for implementing this so quick, I'll post here again as soon as I can give it a try.

spectras, 2016-06-19:
I gave it a try. It works, but unfortunately the X-T1 is more stubborn than I thought. It seems it requires additional exchanges to take place before it accepts to open an event connection, probably to switch it into remote mode (it has other remote features such as GPS data exchange with a phone). That would probably require a whole hooks/plugin system to add camera-specific setup. It's getting out of the scope of this issue.

Thanks again for being so fast at answering.

malc0mn, 2020-08-26:
I made a working PTP/IP implementation for a Fuji X-T1 camera in Go: https://github.com/malc0mn/ptp-ip

Feel free to check out the source to see how to connect to the camera and issue commands...