gphoto/libgphoto2 #65: list-files causes unexpected disconnect - cannot download pictures

nifx, 2016-06-21:
Environment:
Ubuntu 16.04. (64-bit)
gphoto2         2.5.9          gcc, popt(m), exif, no cdk, aa, jpeg, readline
libgphoto2      2.5.9         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0    gcc, ltdl, no USB, serial without locking

Problem:
The --list-files command works for the first time after fresh connecting the device via USB. After callinng --list-files, the camera is disconnected. Additionally, it is not possible to download any pictures (or videos), because this also disconnects camera from computer.

===========================================
calling -> dmesg | tail 
[ 3156.030678] usb 3-3: Manufacturer: Sony
[ 3156.030680] usb 3-3: SerialNumber: C9E3203B6F59
[ 3192.272700] usb 3-3: USB disconnect, device number 4
[ 3196.283891] usb 3-3: new full-speed USB device number 5 using xhci_hcd
[ 3196.413008] usb 3-3: New USB device found, idVendor=054c, idProduct=0994
[ 3196.413016] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3196.413019] usb 3-3: Product: USB Charger
[ 3196.413022] usb 3-3: Manufacturer: Sony
[ 3196.414582] input: Sony USB Charger as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:054C:0994.0004/input/input17
[ 3196.414846] hid-generic 0003:054C:0994.0004: input,hidraw1: USB HID v1.11 Device [Sony USB Charger] on usb-0000:00:14.0-3/input0

===========================================
calling -> gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Sony Alpha-A6000 (MTP)         usb:003,029

===========================================
calling -> gphoto2 --get-file 1 (or --list-files)
results in the following dmesg output (no file could be downloaded):

[ 3192.272700] usb 3-3: USB disconnect, device number 4
[ 3196.283891] usb 3-3: new full-speed USB device number 5 using xhci_hcd
[ 3196.413008] usb 3-3: New USB device found, idVendor=054c, idProduct=0994
[ 3196.413016] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3196.413019] usb 3-3: Product: USB Charger
[ 3196.413022] usb 3-3: Manufacturer: Sony
[ 3196.414582] input: Sony USB Charger as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/0003:054C:0994.0004/input/input17
[ 3196.414846] hid-generic 0003:054C:0994.0004: input,hidraw1: USB HID v1.11 Device [Sony USB Charger] on usb-0000:00:14.0-3/input0

Afterwards, no device available!

msmeissn, 2016-06-21:
What might happen is that the camera only allows connecting once, and after the first session resets itself ot be just a charging device.

nifx, 2016-06-21:
Thanks for your quick reply. The camera worked before running Ubuntu 15.10. Unfortunately, I do not know which versions of gphoto2 and library were installed.

nifx, 2016-06-22:
Attached, you'll find the debug output of:
gphoto2 --debug --debug-logfile=foo.txt --get-all-files

Only one photo (jpg+raw) currently stored on camera.

[foo.txt](https://github.com/gphoto/libgphoto2/files/327491/foo.txt)

closed 2021-10-30 by nifx