gphoto/libgphoto2 #66: gphoto fails on second --capture-preview

kappeler, 2016-06-24:
Hi!

I have an issue with gphoto2 on Ubuntu gphoto2 2.5.9

`gphoto2 --debug --capture-preview `
i get the preview
If i run 
`gphoto2 --debug --capture-preview `
Again I get the error message:
`8.417772 ptp_usb_getresp [usb.c:428] (0): PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)
8.437947 ptp_usb_sendreq             (2): Sending PTP_OC 0x90c8 (PTP_OC_NIKON_DeviceReady) request...
8.437976 gp_port_write               (3): Writing 12 = 0xc bytes to port...
8.438058 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 c8 90-9d 01 00 00              ............    
`

How can I fix this? My goal is to use --capture-preview in a live-view to set the focus.

Thank you!

FBoss, 2017-01-04:
I've got the same issue with my Nikon D7000. 
As a workaround you can set the "viewfinder" setting to 1 before calling the capture_preview function.

msmeissn, 2017-01-15:
hmm.  will see if I can reproduce on my D90 ... after i find the charger :/

closed 2017-01-16 by msmeissn

msmeissn, 2017-01-16:
fixed in current git

kappeler, 2017-01-17:
Thank you @msmeissn !
I do not work on this anymore... But I forwarded this to the guy who is in charge.