gphoto/libgphoto2 #67: Nikon Coolpix L840 not working

phisigma, 2016-06-27:
For my camera Nikon Coolpix L840 gphoto2 2.5.10 doesn’t do anything correct, in contrast to version 2.5.9 where I could at least download all files from my camera.

```
./gphoto2 --list-files
001,001: Datei oder Verzeichnis nicht gefunden                                 
Es gibt keine Datei im Verzeichnis »/«.
Es gibt 2 Dateien im Verzeichnis »/store_00010001«.
#1     <none>                     rd     0 KB application/x-unknown
#2     README.txt                 rd     2 KB text/plain
```

But the camera is listed as a supported model (since version 2.5.10).

Hints:
I use Ubuntu 16.04 64bit and have parallel installed the version from repository. To avoid clashing with my package management, I compiled and installed gphoto2 via:
```
cd libgphoto2-2.5.10
./configure --prefix=/media/ssd-main/appz/gphoto2-2.5.10
make
make install

cd ../gphoto2-2.5.10
PKG_CONFIG_PATH=/media/ssd-main/appz/gphoto2-2.5.10/lib/pkgconfig ./configure --prefix=/media/ssd-main/appz/gphoto2-2.5.10
make
make install
```

Everything seemed fine :-/

msmeissn, 2016-08-03:
You did not build with USB support, so you are only seeing a "virtual testcamera"

Make sure you have libusb development headers installed and found during configure.

configure will report this at the end, also gphoto2 --version will show it after installation.

closed 2016-08-03 by msmeissn

phisigma, 2016-08-28:
For the sake of completeness: I installed both `libusb-1.0-0-dev` and `libusb-dev`, this solved the issue. Thanks, Marcus.

So all supported features of Coolpix L840 are working, but unfortunately this camera does not support much: Photos and videos can be recorded, but without further configuration like zoom, focus etc. (This is just an info for everybody who is interested, it’s not a missing feature or a failure of libgphoto2.)