gphoto/libgphoto2 #68: Please support "Olympus Stylus XZ-10" for image capturing

markuszoeller, 2016-07-03:
Olympus offers tethering software, but only for windows + macos (I use Linux). Using "wine" to work with the windows software doesn't successfully start the installation. Would be great to to have this in your library. Attached is my logfile. Thanks a lot in advance!

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/344974/my-logfile.txt)

msmeissn, 2016-08-02:
does the software really tether over usb? hmm

markuszoeller, 2016-08-15:
It doesn't state otherwise on its page http://support.olympus-imaging.com/ov3download/ 
I'll use another camera system now, no need to keep this ticket open.

closed 2016-08-15 by markuszoeller

msmeissn, 2016-08-21:
The trace did not show any capture related or unknown opcodes. :/ without a usb trace or reverse engineering drivers or documentation hard to implement.