gphoto/libgphoto2 #70: Support for Sony a6300

sdjerdj, 2016-07-25:
Is there an official support for the Sony Alpha 6300 camera? I could not find it on the list.
If no, what is needed for official support?

msmeissn, 2016-08-03:
It might just work as-is if switched to PC Control mode, even without explicitly listing the usb ids.

Please report the USB ids in this mode and if it works.

sdjerdj, 2016-08-03:
Most of the functionality works as is in generic mode, I played with it it a few weeks ago, but I had some things not working as I was expecting. Could have been just me :)
To avoid confusion, do you mind giving me the exact command you want me to run ?

msmeissn, 2016-08-05:
PC Control mode is set on camera.
on the computer side, try:

gphoto2 --list-all-config

gphoto2 --capture-image-and-download

you can try getting and setting some variables, 
gphoto2 --get-config shutterspeed
gphoto2 --set-config shutterspeed=value 

If something does not work as expected, specify ;)

sdjerdj, 2016-08-14:
Please see the attachment for the details.

[Sony_a6300_info.txt](https://github.com/gphoto/libgphoto2/files/417194/Sony_a6300_info.txt)

I haven't had a lot of time to play with gphoto2 and this camera, but once I recall what the issue was I'll hare the details

Regards

msmeissn, 2016-08-21:
so capture works to some degree. setting shutterspeed works through all values, thats sadly currently the way the sonys work.

can you also report the usb ids? 

lsusb

sdjerdj, 2016-08-27:
Please see below:

Remote control mode:
$ lsusb |grep Sony
Bus 003 Device 003: ID 054c:079c Sony Corp. 
$ 

MTP mode:
$ lsusb |grep Sony
Bus 003 Device 005: ID 054c:077a Sony Corp. 
$

Mass storage mode:
$ lsusb |grep Sony
Bus 003 Device 006: ID 054c:0aba Sony Corp. 
$

Auto mode:
$ lsusb |grep Sony
Bus 003 Device 007: ID 054c:0ab9 Sony Corp. 
$

closed 2016-10-11 by msmeissn