gphoto/libgphoto2 #71: Canon EOS 750D capture-preview failure

marcelmatula, 2016-07-27:
Hi
I have an issue with gphoto2 2.5.10 on Ubuntu
capture-image works fine but capture-preview not.

I see in log:
2.030556 ptp_usb_getresp [usb.c:438] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
2.032688 camera_capture_preview [library.c:2569](0): get_viewfinder_image failed after all tries with ret: 0xa102

However it sometimes works but only when i run: gphoto2 --debug --capture-preview
It seems that debug option slowdown capturing which helps finish operation successfully.

See attached log.
I really need capture-preview feature - so let me know if you need any help in order to fix the issue.

Thanks,

Marcel
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/386756/my-logfile.txt)

msmeissn, 2016-08-03:
can you try the attached commit (or current git)?

marcelmatula, 2016-08-03:
Hi Marcus,

I tested latest development version 2.5.10.1 and it works perfectly for both capture-preview and capture-movie. Great job.

Thanks,
Marcel

closed 2016-08-03 by marcelmatula