gphoto/libgphoto2 #73: Inquiry func gp_widget_changed(widget) resets widget->changed flag to 0

Sija, 2016-08-03:
Hi,

Is there any reason why inquiry function `gp_widget_changed(widget)` modifies passed `widget`? Typical use-case involves checking if given `widget` was changed and there's no indication that such call will modify the given `widget`.

```c
gp_widget_changed(widget) # => true
gp_widget_changed(widget) # => false but expected true
```

msmeissn, 2016-08-05:
this seems due to historic  reasons, was added during initial development of the library :(

Sija, 2016-08-06:
I hope it's the time in history when we can move forward and str8 things up ;)

Sija, 2016-11-20:
hi @msmeissn, could it be fixed for the next release?

msmeissn, 2016-11-20:
(I did this now, but I wonder why you need it this way.)

closed 2016-11-20 by msmeissn

Sija, 2016-11-20:
@msmeissn Thanks!