gphoto/libgphoto2 #74: Canon G12 - can't capture images

Pokerfjes, 2016-08-07:
I can't capture images from my G12 with gphoto2 using a raspberry zero. Summary even says it right out - "No image capture". Am I out of luck? :(
I have to say though, I'm using an USB hub. I power up my G12 (otherwise the device isn't recognized), but I notice that the lens doesn't come out. I'm in view mode somehow. 
I even got the libgphoto2-dev package just to make sure I wasn't missing any drivers.

See below for the gphoto2 --summary command.

Thanks,
Ørjan

pi@canonzero:~ $ gphoto2 --summary
Camera summary:
Manufacturer: Canon Inc.
Model: Canon PowerShot G12
  Version: 1-9.0.1.0
  Serial Number: 0B6333A09F8C41708DB3DDF4082EDD89
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, Defined Type, CRW, Unknown(b103), Unkn                                                        own(b105), Unknown(b104), Unknown(bf01)

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
        StorageDescription: None
        VolumeLabel: None
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 31902400512 (30424 MB)
        Free Space (Bytes): 31295471616 (29845 MB)
        Free Space (Images): -1

Device Property Summary:
Event Emulate Mode(0xd045):(readwrite) (type=0x4) Enumeration [1,2,3,4,5,6,7] value: 2
Property 0xd04a:(readwrite) (type=0x2) Enumeration [0,1,2,3] value: 0
Size of Output Data from Camera(0xd02e):(read only) (type=0x6) 524288
Size of Input Data to Camera(0xd02f):(read only) (type=0x6) 524288
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3] value: 3% (3)
Battery Type(0xd002):(read only) (type=0x4) Enumeration [0,1,2,3,4,5] value: AC (1)
Battery Mode(0xd003):(read only) (type=0x6) Enumeration [0,1,2,3] value: Normal (1)
UNIX Time(0xd034):(readwrite) (type=0x6) 1470613401
Type of Slideshow(0xd047):(read only) (type=0x4) 2
DPOF Version(0xd046):(read only) (type=0x4) 257
Remote API Version(0xd030):(read only) (type=0x6) 256
Model ID(0xd049):(read only) (type=0x6) 43122688
Camera Model(0xd032):(read only) (type=0xffff) 'Canon PowerShot G12'
Camera Owner(0xd033):(readwrite) (type=0x4002) a[0]
Firmware Version(0xd031):(read only) (type=0x6) 16777216
Property 0xd050:(read only) (type=0x2) 0
Property 0xd051:(read only) (type=0x4004) a[2] 14337,45317
Property 0xd402:(read only) (type=0xffff) 'Canon PowerShot G12'
Property 0xd406:(readwrite) (type=0xffff) 'Windows'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1

Pokerfjes, 2016-08-07:
Sorry, forgot the output from the capture on command that was the first I tried.

gphoto2 --set-config capture=on

*** Error ***
Failed to set new configuration value on for configuration entry capture.
*** Error (-6: 'Unsupported operation') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config capture=on

Please make sure there is sufficient quoting around the arguments.

Pokerfjes, 2016-08-07:
pi@canonzero:~ $ gphoto2 --version
gphoto2 2.5.4

Copyright (c) 2000-2014 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.4          gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.4          all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.10.0         gcc, ltdl, no USB, serial without locking

Pokerfjes, 2016-08-07:
Testing with my EOS 1000D, works like a charm. Looks like the difference is that with the 1000D I have "Canon EOS capture" available, but with the G12 I have "No vendor specific capture" available.

pi@canonzero:~ $ gphoto2 --capture-image-and-download
New file is in location /capt0000.jpg on the camera
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
Deleting 'capt0000.jpg' from folder '/'...
pi@canonzero:~ $ gphoto2 --summary
Camera summary:
Manufacturer: Canon Inc.
Model: Canon EOS 1000D
  Version: 3-1.0.3
  Serial Number: 80c5d1e5dd484905985f3048bb9afea5
Vendor Extension ID: 0xb (2.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type

Device Capabilities:
        File Download, File Deletion, File Upload
        No Image Capture, No Open Capture, Canon EOS Capture

Storage Devices Summary:
store_00020001:
        StorageDescription: SD
        VolumeLabel: None
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Digital Camera Layout (DCIM)
        Access Capability: Read-Write
        Maximum Capability: 8160542720 (7782 MB)
        Free Space (Bytes): 4261609472 (4064 MB)
        Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 1000D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'

msmeissn, 2016-08-08:
Yes, Canon stopped supporting remote capture over USB in their Powershot series some years ago.
(Modern Powershots can do capture over WLAN, but still not over USB.)

The EOS have capture support.

For the Powershot series, there is a third party addon called "CHDK" ( http://chdk.wikia.com/wiki/CHDK  ) , that has some basic capture support. Newer libgphoto2 versions have support for that capture method, although it is not so good as the previous powershot capture support.

Pokerfjes, 2016-08-08:
Thanks for the quick response Marcus. I'm searching for a cheap way to use reuse cheap but fairly high quality 2nd-hand cameras as time lapse cameras. The Powershot G-series has been the compact standard everyone else measured up against, so that's the reason I tried that. 
When  you say "not so good" as the previous capture support - what are the drawbacks? I might not even notice :).
The only issue with the EOS series is their size, but with a larger sensor they may be my best bet still.

msmeissn, 2016-08-08:
It can not configure all parameters (but most of them)

I have not much personal experience with CHDK yet.

msmeissn, 2020-02-29:
can you try the G12 with a newer libgphoto2 release?

We have meanwhile found out that Canon uses the EOS firmware also on its PowerShot series, and allow capture to some degree.