gphoto/libgphoto2 #75: Is there a way to change Canon EOS EVF resolution?

xaphod, 2016-08-12:
Hi GPhoto2 team,
First off: you're all amazing. I love you all. (I've been lurking for years)

Question: i'm using libGPhoto2 code in another project. One limitation i've hit with Canon EOS cameras like the Canon 80D during live-view (EVF), is that framerate when pulling the photos over PTP/IP is too slow, because the individual frames from `camera_capture_preview` are 200kB each! That makes it pretty impossible to get framerates high enough for the view to look "live", ie. video-like.

Is there any property that one can set to influence this? Or a setting on the camera? Ie. have each EVF frame be much smaller resolution? Even when setting the camera on S1 (smallest JPGs), the EVF frame resoution is 960x640 px which is huge.

rlamarche, 2016-08-12:
Hi, I'm getting 60 fps with a Nikon D800 connected to the Raspberry PI (640x360px)
Which API are you using, how are you displaying the frames, and what is the current resolution of live view frames ?

xaphod, 2016-08-12:
Frames are 960x640. It's PTP/IP. Since the wifi of the cam is max 1MB/s, and frames of 960x640 are 0.2MB, then i'm seeing a max of about 5 fps.

rlamarche, 2016-08-12:
Ok I did not understood that you were using wifi.
Why not connecting a usb cable ? You need wireless ?

xaphod, 2016-08-12:
Yes it is wifi.
I figured it out: in gphoto2, PTP_DPC_CANON_EOS_EVFOutputDevice is commented as "2 means PC, 1 means TFT". If you wireshark the official canon app from an iOS device, you can see it sending 0x0009 instead of 2. I tried using that: sure enough, setting 0x0009 instead of 0x0002 gives smaller resolution frames: 480x320 instead of 960x640.  Happy me 😄

rlamarche, 2016-08-12:
Good, congratulations ;-) ! Does canon provide à detailed sdk ? It might be useful.

xaphod, 2016-08-12:
thx. Canon's SDK is not open-source, and doesn't support wireless (only wired) connections. So it's doubly useless to me :( Plus you have to "apply" to get it... not useful.

rlamarche, 2016-08-12:
ok I was talking about a document that explains all operation codes, parameters and property codes.
Nikon provide it.

msmeissn, 2016-08-12:
canon does not provide this in their SDK; no.

msmeissn, 2016-08-13:
This should be reachable over the "output" config variable, although as unknown values or so.

closed 2017-02-03 by xaphod