gphoto/libgphoto2 #77: Canon 5d normal mode not recognized by file managers anymore (FC23)

pianopilot, 2016-08-16:
Hello!

I recently upgraded from FC22 to FC23, and my Canon 5d ("mark nothing", the original one) is not shown in file managers (Thunar and Nautilus) anymore.

The gphoto2 command line functions seem OK, except that file retrievals show:

``
*** Error ***              
canon_usb_set_file_attributes: canon_usb_dialogue failed
``

However, I'm not sure that's related.

Darktable is the only GUI utility I have that can interact with the camera at all now. It still can do pretty much everything (retrieval, deletion, tethered shooting). Gthumb and ShowFoto, for example, do not show the camera anymore.

Could this be a problem with gvfs interaction?

msmeissn, 2016-08-21:
gvfs is the backend for nautilus at least, so it should show there.

its weird the other frontends do not show it, as the commandline supports it.

No idea why its not working. :/

pianopilot, 2016-09-17:
Sorry for the slow response. 

Sorry if I was unclear. My point was that it looks to me like nothing that uses gvfs seems to show the camera now. Only Darktable, which may use some other backend (not sure), shows it.

Could there be something in the interaction between libgphoto2 and gvfs that changed, and broke the functionality for this camera?

pianopilot, 2016-09-17:
Quick update: Konqueror (which seems to use something called kioslave) also has no problem communicating with the camera. So this definitely looks to be gvfs-related.

msmeissn, 2016-09-18:
hmm,. no, our gvfs interaction did not change... still weird.

it might be triggering on this error though :/