gphoto/libgphoto2 #78: Unable to get widget liveviewafmode (0xD05D) on Nikon D800

rlamarche, 2016-08-21:
Hi,

I'm unable to get the widget liveviewafmode (0xD05D) on Nikon D800.
The point that I don't understand is that the 0xD05D code is listed with `gphoto2 --list-all-config` and I can set values through the widget name "d05d" but "liveviewafmode" is not accessible (so I don't get the labels).

Here my "--list-all-config" :

[d800.txt](https://github.com/gphoto/libgphoto2/files/429030/d800.txt)

msmeissn, 2016-08-21:
can you also attach 
gphoto2 --summary

we might have a mismatch on the data type of the value

msmeissn, 2016-08-21:
can you also run

gphoto2 --debug --debug-logfile=xx.log --list-all-config  and attach xx.log? it will probably contain errors on d05d

rlamarche, 2016-08-21:
Ok

[summary.txt](https://github.com/gphoto/libgphoto2/files/429035/summary.txt)

[debug.txt](https://github.com/gphoto/libgphoto2/files/429036/debug.txt)

rlamarche, 2016-08-21:
The most weird thing is that gp_camera_get_single_config is returning GP_OK but my widget is set to 0x0.

msmeissn, 2016-08-21:
can you test this commit 96896d957381cc58afbe28d1193b27bfae07d819

rlamarche, 2016-08-21:
[image]

closed 2016-08-21 by rlamarche