gphoto/libgphoto2 #80: Android port

FallenRiteMonk, 2016-08-24:
Is it somehow possible import this library into an android application?

rlamarche, 2016-08-24:
A long time ago I tried a fork and made it working on Android device, so the answer is "yes" ;)
I did use this fork : http://gitorious.org/agphoto2/agphoto2 and had compiled libusb for android.
The main problem was that I needed the root permission to open the usb device because the ask for permission on usb port in JAVA was not easy to make in C/C++. So I was doing chmod 666 on /dev/usb0 just before running my app ...

Here the source code of my (largely) unfinished project : https://github.com/rlamarche/photomaton-mobile
I was able to open the camera with libgphoto2 but unable to get liveview image. I'm sure that it is possible because dslrdashboard is doing it with pure JAVA code (without libgphoto2).

laheller, 2018-12-18:
@FallenRiteMonk  @rlamarche 
This repository is for Android also contains **libgphoto2.so**, but it's quite old:
https://github.com/JohnChu/Snoopy

wangguanxiao, 2020-04-21:
> @FallenRiteMonk @rlamarche
> This repository is for Android also contains **libgphoto2.so**, but it's quite old:
> https://github.com/JohnChu/Snoopy

I think,  the file named 'libgphoto2.so' is the most  important . I tried to build this file ,but failed .

laheller, 2020-04-22:
> > @FallenRiteMonk @rlamarche
> > This repository is for Android also contains **libgphoto2.so**, but it's quite old:
> > https://github.com/JohnChu/Snoopy
> 
> I think, the file named 'libgphoto2.so' is the most important . I tried to build this file ,but failed .

You can try the new library available at:
https://github.com/rupiapps/ptplibrary

The author states that it is compatible with most of Canon, Nikon and Sony cameras.