gphoto/libgphoto2 #81: Canon EOS700D hangs after some days

obi141, 2016-08-29:
Hi everybody!
I'm using gphoto2 in combination with a Canon EOS700D on a Raspberry Pi 3. Everything works perfectly for near a week and then suddenly gphoto2 can't reach the camera anymore ("No camera found" error).
I take a picture every 10 minutes and save them on a folder on the Raspberry Pi.
The display of the camera is black and even if I push the menue button on the cam there is no reaction.
Even if I disconnect the power of the cam doesn't helps. In the moment when I take a picture manually everything turns normal but only for a couple of days.
I have this issue on gphoto2.5.9 and even on 2.5.10. Does anybody already have encontered this issue?
I made this test with two EOS700D and I get always the same issue.

Thanks in advice,
obi141

obi141, 2016-08-30:
Some additional infos: I've found a difference between a earlyer bought EOS700D and the actual which I have in use. The actual lens is a EF-S 18-55 IS STM on the older cam is even a EF-S 18-55 but not STM.
Is it possible that this "new" lens makes the issue?

closed 2016-09-05 by obi141

msmeissn, 2016-09-06:
The lens should not matter here.

Did you find the problem?

obi141, 2016-09-06:
Hi Marcus,
you're right, I've checked another system and I've found one where I'm 
using the same lens
but gphoto2 and libgphoto2 with version 2.5.8 and there I have no issues.
At the moment I made a downgrade to 2.5.8 on my test system and now I'm 
waiting
if the issue appears again. Unfortunately it happens after some days so 
I have to wait.

I already found one difference between version 2.5.8 and the 2 higher 
versions (2.5.9. + 2.5.10).
On 2.5.8 the display stays off and doesn't turn on if I press for 
example the menu-button on the camera.
It turns only on when I take a picture via gphoto or if I disconnect the 
cam from USB.
Via gphoto everything works well.

On the newer versions ( > version 2.5.8 ) the display stays on and I can 
navigate trought the menu and change settings while the camera is connected.
At the moment when the camera begins to hang the display stays off and 
there is no reaction if i press the menu-button and gphoto2 doesn't find
the camera anymore. If I press the release button or if I disconnect the 
lens for a moment everything turns normal for a couple of days.
The issue persists, no matter if I change options or not.

I closed the issue because of the fact that it must be another problem 
than the described one...

Kind Regards,
Stefan