gphoto/libgphoto2 #82: Capture delay

miks, 2016-08-29:
What's the minimum expected delay when doing only capturing?
With my Canon EOS100d I got something like 300ms delay when calling "capture" command. Instead pressing physical capture button on camera itself, capturing is happening immediately.

rlamarche, 2016-08-29:
Check if you have a property `burstnumber` and `burstinterval` in available widgets. This way you can ask your camera to take n photos.

miks, 2016-08-29:
Problem is not with burst capturing. I just tested capturing with OSX built in "Image Capture" which, I suppose, also use PTP to control camera, and there is no lag between pressing enter and capturing on camera.
With gphoto2 there is no "realtime" feel when pressing enter from my application (through python).
I wonder whether it's gphoto "fault" or PTP protocol itself.

msmeissn, 2016-08-29:
the initial startup might be a bit long.

are you using the ciommandline tool? can you capture debugtrace with it, then you will see the timestamps.

osx opens the camera as soon as you plug it in, so the startup code happens when its plugged in, while it is done by gphoto2 before that.

miks, 2016-08-29:
I'm opening connection to camera and capturing afterward, so I believe there is no "startup" issue.
From my understanding also doing capturing with gphoto2 "shell" mode makes capturing without any "startup" delays.
So, my question stays same: what are minimum expected capturing delay possible with Canon EOS cameras? It's not my concern how slow or fast image are downloaded, all I'm interested is time between capturing command sent from gphoto2 and actual capturing happening on camera.

Sija, 2016-09-06:
Might be related to #24, does it happen on version `2.5.8`?

miks, 2016-09-06:
I find out that ubuntu guest within osx virtualbox was to blame. At osx or ubuntu host there are no delay between capture command and real capture by camera. So, I'm closing this.

closed 2016-09-06 by miks