gphoto/libgphoto2 #83: Nikon D7200 Support

kanyck, 2016-09-02:
Darktable *almost* works with D7200 via libgphoto2 in PTP mode (all but Live View Close-up and Focusing) which makes it effectively unusable: shots are misfocused.
However D7100 is listed as supported. Most probably D7200 is close enough to try a shot. 
Is it possible to include support for D7200? I'm ready to test.  I'm on Gentoo thus have no problem to try out patches if any.

Thanx anyway,
//Nick

msmeissn, 2016-09-04:
The support is generic even if the ID is not known. So the D7200 should already work well.

The USB ids would be interesting for our database, and --summary and --list-all-config output.

Is the d7100 working better in the tasks you mention?

kanyck, 2016-09-04:
Hi msmeissn! 
$ LANG=C gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
USB PTP Class Camera           usb:005,003 

$ LANG=C gphoto2 --summary | pastebinit 
http://pastebin.com/RmAVxKQc

$ LANG=C gphoto2 --list-all-config | pastebinit 
http://pastebin.com/W5U8P5Lb

To put it clearly -- I don't have D7100 -- I've only seen in on the list of supported cameras and assumed the D7200 may be close enough to be added painlessly.
Unfortunately libgphoto2 doesn't support my D7200 in full:
LANG=C gphoto2 --port usb: --capture-tethered
Waiting for events from camera. Press Ctrl-C to abort.                         
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
...
and what's worse -- I cannot focus my camera from the software. I cannot do it manually, either, because Live View zoom doesn't work, too. In Darktable I see focus and zoom buttons grayed and not functional.
So it will be very nice of you to suggest some solution on this issue.

msmeissn, 2016-09-07:
can you also report the USB ids, just run "lsusb" with the camera attached.

are you trying to press the shutter during this --capture-tethered?

have to find time to debug the zooming during liveview issue

kanyck, 2016-09-07:
lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0566:3058 Monterey International Corp. 
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 008 Device 002: ID 05e3:0732 Genesys Logic, Inc. All-in-One Cardreader
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 05e3:0612 Genesys Logic, Inc. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 04b0:0439 Nikon Corp. 
Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

You don't need lsusb -v, do you?

LANG=C gphoto2 --port usb: --capture-tethered
Waiting for events from camera. Press Ctrl-C to abort.                         
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1f1 changed
UNKNOWN PTP Property d105 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property d1b4 changed
Saving file as _DSC1967.NEF                                                    
CAPTURECOMPLETE
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1f1 changed
^C
Cancelling...
Thus, it makes a shot but quite obviously -- misfocused. Also, it may be relevant -- I normally use AF-L/AE-L button to focus in AF-ON mode and it's what is set on the camera. However, this shouldn't influence Darktable focusing feature, should it?

kanyck, 2016-09-07:
Marcus,
I've applied a patch.
Now libgphoto2 recognizes the camera properly and I may shoot pics in focus via both --capture-tethered (focusing with AF-ON button) and --capture-image-and-download (autofocus ok).
However I tried out darktable (mode I really need) with no success -- nothing changed yet...

msmeissn, 2020-03-01:
should hopefully been better now I hope.

closed 2020-03-01 by msmeissn